UniversalRobots/Universal_Robots_ROS2_Driver #504: moveit2 ros2 universal robot grippers

enriLoniterp, 2022-10-07:
I am interest in the model of the gripper of universal robot, it is available ?  there is some guide i could you use to implent it in my application? thank you very much

fmauch, 2022-10-10:
Which gripper are you referring to? To my knowledge, not Universal Robot gripper exists.

enriLoniterp, 2022-10-10:
https://www.universal-robots.com/plus/products/onrobot/rg2-gripper/

Rg2 gripper is sold to customers as universal robots gripper

fmauch, 2022-10-10:
To my knowledge there doesn't exist a ROS2 driver for this, yet. Anyway, this would be out of scope for this driver, so I'll close this issue.

You'd need a ROS package for the gripper (description and driver), build a custom URDF containing the robot and the gripper, create a custom moveit config with that URDF and you would be good to go.

closed 2022-10-10 by fmauch