UniversalRobots/Universal_Robots_ROS2_Driver #507: Humble - ur_controllers build fail

g-argyropoulos, 2022-10-10:
### Description
I try to build the Universal_Robots_ROS2_Driver but it fails at the ur_controllers package due to **params_** not being declared.

My setup

    OS: Ubuntu 22.04.1
    ROS Distro: Humble

### Command

    colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release

### Result
```
--- stderr: ur_controllers                                    
/home/innovation/ros2_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In member function ‘virtual controller_interface::return_type ur_controllers::ScaledJointTrajectoryController::update(const rclcpp::Time&, const rclcpp::Duration&)’:
/home/innovation/ros2_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:109:26: error: ‘params_’ was not declared in this scope
  109 |   const auto joint_num = params_.joints.size();
      |                          ^~~~~~~
gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:76: CMakeFiles/ur_controllers.dir/src/scaled_joint_trajectory_controller.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/ur_controllers.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ur_controllers [9.64s, exited with code 2]

```

fmauch, 2022-10-10:
Have you pulled or is this a fresh clone? Either case, you'll have to make sure to run the 

```
vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver-not-released.${ROS_DISTRO}.repos
```

command before building due to some code changes upstream. This will be necessary until the next package sync. If you only want to use the driver and not take part in active development, I suggest to use the binary packages by installing `ros-humble-ur` instead of building it from source. Please note that the README regarding this is currently [being worked on](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/494).

g-argyropoulos, 2022-10-11:
It was a fresh clone, so I had run the suggested command. It turned out that the problem was caused because I had also previously built the package in the underlay by mistake (some kind of conflict with the fresh clone I guess). Everything works fine now that I allowed overriding.

In any case, thank you for your time, your assistance and for bringing the binary packages to my attention.