UniversalRobots/Universal_Robots_ROS2_Driver #508: Cannot find forward position controller in Foxy

ddliu365, 2022-10-10:
Hi, folks,

I cannot activate the forward position controller on UR5e.

I installed our driver via `foxy` branch.

I launch the joint trajectory controller and check the controller list:

```
dongdong@dell:~/workspace/ros_ur_driver/src$ ros2 control list_controllers
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active    
force_torque_sensor_broadcaster[ur_controllers/ForceTorqueStateBroadcaster] active    
joint_trajectory_controller[joint_trajectory_controller/JointTrajectoryController] active    
io_and_status_controller[ur_controllers/GPIOController] inactive  
speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active
```

I didn't find the forward joint position controller in Foxy. Can you help to check?

fmauch, 2022-10-11:
Looks like this never got integrated. To use it, you first will have to add it to https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/foxy/ur_bringup/config/ur_controllers.yaml.

Then, you'll either have to start it by hand or modify https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/foxy/ur_bringup/launch/ur_control.launch.py according to the main branch in https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/bcdcf9496cf61190e98e6a0fd7f35b5a4d220878/ur_robot_driver/launch/ur_control.launch.py#L306-L310 and https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/bcdcf9496cf61190e98e6a0fd7f35b5a4d220878/ur_robot_driver/launch/ur_control.launch.py#L338.

A PR with this changes would be appreciated, as otherwise this might land too low on our priority list.

ddliu365, 2022-10-17:
will work on it

VinDp, 2024-06-05:
Since Foxy is not supported anymore and the `forward_position_controller` is spawned for the latest distros, I will close the issue.

closed 2024-06-05 by VinDp