UniversalRobots/Universal_Robots_ROS2_Driver #513: Control gripper via ROS2

LuRe97, 2022-10-12:
Hello,

I am currently using this driver to receive the current joint trajectories of my robot and to send new goal trajectories to move the robot. Now I want to control the connected gripper as well, but I am currently struggling as to how the gripper could be controlled via ROS2 topics. Is there a way to activate the grippers vacuum suction via ROS2 topics?

These are my used versions in software/hardware:
**Ubuntu Version:** 20.04
**ROS2-Version:** Foxy
**Robot:** Ur5e (Software 5.12)
**Gripper:** Schmalz Vacuum Gripper ECBPi 

Thanks in advance for your help.

enriLoniterp, 2022-10-18:
hi, you are able to teach me how to add the gripper in the universal robots driver in order to visualize it in Rviz during my simulation? It would be awesome for me simulate a pick operation with a gripper, actually i still can't attach a gripper on my ur robot( any one it's ok for simulation purpose).

Thank you in advance,
Enrico

fmauch, 2022-10-18:
For attaching a gripper, you would have to extend the robot description with the gripper and either start a ROS driver for the gripper to control it / read its state or control it e.g. via I/O service calls. The latter won't give you state information, though, which would be required for visualizing it in RViz.

Please notice that controlling a gripper goes beyond the scope of this repo, though. I understand that this is a big concern and we might add a tutorial with an example gripper at some point, but we definitively cannot give support for every gripper out there. For a ROS-supported gripper (AFAIK there are not too many grippers supported by ROS2 atm), the process should always be similar, though.

enriLoniterp, 2022-10-19:
hi, i think i understand the immense work that should be. In my case i don't need everything done but a simple guide with an example for at least do a pick and place application with a sample gripper would be amazing. I've lost so much time struggling to understand every passage ( i'm doing all by myself from ROS2 scratch) and still i'm not sure to how use a gripper with this robot or any other actually. It's all confused and someone should set a guide for everyone.

Thank you, enrico

fmauch, 2024-04-10:
Closing this as it is out of scope for this repo.

closed 2024-04-10 by fmauch