UniversalRobots/Universal_Robots_ROS2_Driver #515: DANGER: Robot Teleportation

doshininad, 2022-10-17:
I am controlling UR10 with ROS2 driver given in this repository.
When I Estop the robot and start it back, I experience robot teleportation (the robot moves with a very high speed and then it stops because the robot controller detects internal collision)

The exact steps I followed are:
- Run ROS2 Driver and MoveIt2 Script on the PC
- Run Ext_Ctrl URCap on the UR robot
- Set a goal position in Rviz
- Plan the trajectory using 'Plan' in Rviz
- Execute the trajectory using 'Execute' in Rviz
- Estop the robot (on the Teach Pendent) before the trajectory is completed
- Remove the Estop and activate the robot from Teach Pendent
- Play the ext_ctrl URCaps on Teach Pendent
- Robot Teleportation 
- Robot stops as the robot controller detects internal collision and stops the robot

This is very dangerous move (more specifically for a collaborative robot). I don't know if it is because I am using it in wrong way or a bug. But, it needed to be fixed as it is dangerous.

https://user-images.githubusercontent.com/37081073/196284134-495079b3-17da-498e-920b-d52fa2efc6ed.mp4

urrsk, 2022-11-03:
With the information giving, I believe that your planner is keep sending targets to the robot without knowing that the robot is stopped and not moving.

May I ask which controller you are using?

I will strongly suggest that you are using the **scaled_joint_trajectory_controller** as this loops pack the speed-scale of the robot including pausing the trajectory execution if the robot program is Emergency or safeguard stopped as well as pause

fmauch, 2024-04-10:
Closing this as a solution was proposed quite some time ago and we never heard back.

closed 2024-04-10 by fmauch