UniversalRobots/Universal_Robots_ROS2_Driver #522: 2.2.4 in Humble fails to build

audrow, 2022-10-21:
It seems that 2.2.4 introduces a regression that causes the packages in this repo to fail to build. For example,
https://build.ros2.org/view/Hbin_uJ64/job/Hbin_uJ64__ur_controllers__ubuntu_jammy_amd64__binary/21/console

I've reverted 2.2.4's PR on rosdistro:
https://github.com/ros/rosdistro/pull/35009

fmauch, 2022-10-23:
I would expect, that the error is indeed not introduced by 2.2.4 but caused by a regression introduced into ros2_controllers, which has been fixed in https://github.com/ros-controls/ros2_controllers/pull/448.

Seeing the latest build with 2.2.3 fail supports that statement.

https://build.ros2.org/job/Hbin_uJ64__ur_controllers__ubuntu_jammy_amd64__binary/43/console

Since I am pretty sure the build errors originate from that upstream issue, I didn't take any further actions but wait for the next release of ros2_controllers instead.

audrow, 2022-10-24:
I suspect you're right. Thanks for pointing this out.

fmauch, 2022-11-15:
@audrow could https://github.com/ros/rosdistro/pull/35009 maybe be re-reverted before the next sync? As we've discussed, the failed build does not follow a regression from this repo, but upstream problems.

Since `ros2_controllers` has been reverted to 2.12.0 build for `ur_controllers` are succeeding again. They should also for 2.2.4.

audrow, 2022-11-15:
Done in https://github.com/ros/rosdistro/pull/35331. I'm going to close this issue. Thanks, @fmauch!

fmauch, 2022-11-17:
As per request, version 2.2.4 was re-released to humble, but I seem to have forgotten that `joint_trajectory_controller` also changed its API between 2.12.0 and 2.13.0 to which we [adapted](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/492) in our 2.2.4 version. So, despite my assumption that 2.2.4 would be working fine with `ros2_controllers `2.12.0`, it does not. @audrow I could provide a quick fix that reverts the API changes here, so we can build with ros2_controllers 2.12.0 again. I think, that would be preferable to reverting to 2.2.3 again.

audrow, 2022-11-17:
> I could provide a quick fix that reverts the API changes here, so we can build with ros2_controllers 2.12.0 again. I think, that would be preferable to reverting to 2.2.3 again.

That sounds good to me.

fmauch, 2022-11-24:
Closing this, as things got resolved with releasing 2.2.5.

closed 2022-11-24 by fmauch