UniversalRobots/Universal_Robots_ROS2_Driver #529: ros2 launch ur_robot_driver ur_control.launch.py Error

LucaBross, 2022-11-05:
I updated the System with `sudo apt update` and `sudo apt upgrade`. Over 200 packages got updated. Most of them are form ros2 humble and the ur package.

After that update i tried to connect again to my robot with ` ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e robot_ip:=192.168.56.101 launch_rviz:=false initial_joint_controller:=joint_trajectory_controller`

then i got the following error 

`[INFO] [launch]: All log files can be found below /home/luca/.ros/log/2022-11-05-17-01-50-796909-Lenovo-V15-G2-ALC-609185`
`[INFO] [launch]: Default logging verbosity is set to INFO`
`[ERROR] [launch]: Caught exception in launch (see debug for traceback): executed command showed stderr output. Command: /opt/ros/humble/bin/xacro /opt/ros/humble/share/ur_description/urdf/ur.urdf.xacro robot_ip:=192.168.56.101 joint_limit_params:=/opt/ros/humble/share/ur_description/config/ur3e/joint_limits.yaml kinematics_params:=/opt/ros/humble/share/ur_description/config/ur3e/default_kinematics.yaml physical_params:=/opt/ros/humble/share/ur_description/config/ur3e/physical_parameters.yaml visual_params:=/opt/ros/humble/share/ur_description/config/ur3e/visual_parameters.yaml safety_limits:=true safety_pos_margin:=0.15 safety_k_position:=20 name:=ur3e script_filename:=/home/luca/workspace/ros_ur_driver/install/ur_robot_driver/share/ur_robot_driver/resources/ros_control.urscript input_recipe_filename:=/home/luca/workspace/ros_ur_driver/install/ur_robot_driver/share/ur_robot_driver/resources/rtde_input_recipe.txt output_recipe_filename:=/home/luca/workspace/ros_ur_driver/install/ur_robot_driver/share/ur_robot_driver/resources/rtde_output_recipe.txt prefix:="" use_fake_hardware:=false fake_sensor_commands:=false headless_mode:=false use_tool_communication:=false tool_parity:=0 tool_baud_rate:=115200 tool_stop_bits:=1 tool_rx_idle_chars:=1.5 tool_tx_idle_chars:=3.5 tool_device_name:=/tmp/ttyUR tool_tcp_port:=54321 tool_voltage:=24 `
`Captured stderr output:`
`warning: Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.`
`warning: Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.`
`warning: Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.`
`warning: Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.`
`warning: Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.`

How can i solve this?

fmauch, 2022-11-08:
This is actually an issue of the description, not the driver (basically, the issue comes from xacro breaking the api during humble, but let's not be too nitpicking). Anyway, this has been addressed in https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/pull/43 which has been released today, so this will be gone for good with the next sync.

In the meantime you can build the description from source inside your workspace. I'll not close this until the next humble sync, since this might help other users, as well.

LucaBross, 2022-11-08:
Thanks for the information. 
I cloned https://github.com/UniversalRobots/Universal_Robots_ROS2_Description to my workspace. Then I build it. 
How can I choose now to launch this description? 

I launched the workspaces like this:
```
source /opt/ros/humble/setup.bash
source /usr/share/colcon_argcomplete/hook/colcon-argcomplete.bash
source ~/workspace/ros_ur_driver/install/setup.bash
source ~/workspace/ros_tutorial/install/setup.bash
source ~/workspace/ws_moveit2/install/setup.bash
source ~/workspace/ur_control/install/setup.bash
```
The description hast this path:
`/home/luca/workspace/ur_control/src/Universal_Robots_ROS2_Description`

fmauch, 2022-11-09:
Sourcing multiple workspaces like that seems a bit awkward to me. Usually, I only source one workspace's `setup.bash`. See the according [`colcon documentation`](https://colcon.readthedocs.io/en/released/developer/environment.html#workspace-level).

Apart from that this should automatically use the correct description repo. In case you're unsure, you can easily delete the apt version by `sudo apt remove ros-humble-ur-description` but to my understanding this **should** not be necessary.

You can easily check the package's prefix by calling

```
ros2 pkg prefix ur_description
```

This should yield  the local install folder of your `ur_description` package. On my dev workspace this yields for example

```
/home/ubuntu/robot_folders/checkout/ur_ros2_rolling/colcon_ws/install/ur_description
```

levywl, 2022-11-10:
I had the same problem, after i install the driver and try to test it with:

`ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true`

Then I got the error:

`[ERROR] [launch]: Caught exception in launch (see debug for traceback): executed command showed stderr output. Command: /opt/ros/humble/bin/xacro /opt/ros/humble/share/ur_description/urdf/ur.urdf.xacro robot_ip:=xxx.yyy.zzz.www joint_limit_params:=/opt/ros/humble/share/ur_description/config/ur5e/joint_limits.yaml kinematics_params:=/opt/ros/humble/share/ur_description/config/ur5e/default_kinematics.yaml physical_params:=/opt/ros/humble/share/ur_description/config/ur5e/physical_parameters.yaml visual_params:=/opt/ros/humble/share/ur_description/config/ur5e/visual_parameters.yaml safety_limits:=true safety_pos_margin:=0.15 safety_k_position:=20 name:=ur ur_type:=ur5e script_filename:=ros_control.urscript input_recipe_filename:=rtde_input_recipe.txt output_recipe_filename:=rtde_output_recipe.txt prefix:="" 
Captured stderr output:
warning: Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
warning: Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
warning: Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
warning: Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
warning: Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.`

LucaBross, 2022-11-11:
Building the Description in my Workspace solved it for me 
go to src
`git clone https://github.com/UniversalRobots/Universal_Robots_ROS2_Description.git`
go to the root of your workspace 
`colcon build --packages-select Universal_Robots_ROS2_Description`

levywl, 2022-11-12:
> 

Thanks for the method, I followed your steps but maybe I misunderstand something, I clone the file in workspace/ros_ur_drivre/src. 
then colcon build in /workspace but get the error 
`WARNING:colcon.colcon_core.package_selection:ignoring unknown package 'Universal_Robots_ROS2_Description' in --packages-select
                     
Summary: 0 packages finished [0.15s]`

levywl, 2022-11-12:
Thanks for the method, I followed your steps but maybe I misunderstand something, I clone the file in 

workspace/ros_ur_drivre/src. 

then colcon build in /workspace but get the error 

`WARNING:colcon.colcon_core.package_selection:ignoring unknown package 'Universal_Robots_ROS2_Description' in --packages-select   
Summary: 0 packages finished [0.15s]`

LucaBross, 2022-11-12:
Oh you are right. I think the package is called `ur_description` 

Try just `colcon build` 
Or try `colcon build --event-handlers desktop_notification- status- --executor sequential` if you have a device with little RAM

levywl, 2022-11-12:
> 

Oh it works, I just `colcon build` in the workspace then I can run the rviz2. So grateful. Hope `4 packages had stderr output` won't cause some bugs

enriLoniterp, 2022-11-14:
copying in the src folder URDescription package it doesn't work for me and it prints me the same output error mentioned before about load_yaml(). How can i do?

LucaBross, 2022-11-14:
@enriLoniterp did you build and source after copying the Description?

VinDp, 2024-06-05:
Since a solution has been proposed and the issue is stale, I will close the issue.

closed 2024-06-05 by VinDp