UniversalRobots/Universal_Robots_ROS2_Driver #530: CMake can't find controller_interface

Lingaprahasam, 2022-11-06:
`--- stderr: ur_controllers
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findcontroller_interface.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "controller_interface", but CMake did not find one.

  Could not find a package configuration file provided by
  "controller_interface" with any of the following names:

    controller_interfaceConfig.cmake
    controller_interface-config.cmake

  Add the installation prefix of "controller_interface" to CMAKE_PREFIX_PATH
  or set "controller_interface_DIR" to a directory containing one of the
  above files.  If "controller_interface" provides a separate development
  package or SDK, be sure it has been installed.

---
Failed   <<< ur_controllers [2.00s, exited with code 1]
Aborted  <<< myworkcell_support [8.75s]`

fmauch, 2022-11-08:
This looks like there are dependencies missing. Did you install all the dependencies using the `rosdep install` command as explained in the README? Did that show any errors?

Lingaprahasam, 2022-11-20:
Yes I was missing the ros dependancies. Thanks

closed 2022-11-20 by Lingaprahasam