UniversalRobots/Universal_Robots_ROS2_Driver #531: Trying to do a moveL and moveJ simulated

marioesteban11, 2022-11-07:
Hi everyone,

I'm trying to do a MoveL, MoveJ and MoveC simulated in rviz with Moveit and in Gazebo, but iḿ not able to do it.

I'm in:
Ubuntu 22
Ros2 Humble
Robot: Ur3e

In my c++ code I use the geometry PoseStamped to move my UR3e in severeal position, but my robot goes crazy, doing some loops before getting to the goal I try to publish.

Do you have any idea to do this moves with geometry pose or any other options?

Thanks in advance

stefanscherzinger, 2022-11-07:
@marioesteban11 

> Do you have any idea to do this moves with geometry pose or any other options?

Until you get a response from the package maintainers, let me point you to an alternative [control approach](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/ros2/cartesian_motion_controller). That's a controller that follows a `geometry_msgs/PoseStamped`. It's part of the [cartesian_controllers](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/ros2) suite. It will interpolate for you and command the robot with `MoveJ` down in the driver for the UR robots.

marioesteban11, 2022-11-07:
Trying to clone your repository [cartesian controllerss](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers) the terminal says:

´´´
Cloning into 'cartesian_controllers'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
´´´

And I dont know if my github is wrongly configurated or i need any permision to clone it

LucaBross, 2022-11-07:
@marioesteban11  did you manage to set your Tool orientation with 

> In my c++ code I use the geometry PoseStamped to move my UR3e in severeal position

stefanscherzinger, 2022-11-07:
@marioesteban11
Please open further issues on `cartesian_controllers` on their issue tracker. Let's keep this discussion on the ROS2 UR driver.

marioesteban11, 2022-11-07:
@LucaBross not really:
'''
msg.orientation.w = 1.0;
    msg.orientation.x = 0;
    msg.orientation.y = 1;
    msg.orientation.z = 2;
'''

when i try to run this, with this orientations or others my Ur3e doesn´t seem to move the orientation

LucaBross, 2022-11-08:
@marioesteban11 Thanks, now i may be able to figure out the right orientation for me. 

I also have a lot of unnecessary movement if i use `move_group.move();`.
I tried to fix this with using a cartesian path:

```
  std::vector<geometry_msgs::msg::Pose> waypoints;
  waypoints.push_back(start_pose2);

  geometry_msgs::msg::Pose target_pose3 = start_pose2;

  target_pose3.position.z += 0.2;
  waypoints.push_back(target_pose3);

  target_pose3.position.y -= 0.2;
  target_pose3.position.x -= 0.2;
  waypoints.push_back(target_pose3); 

  target_pose3.position.z -= 0.2; 
  waypoints.push_back(target_pose3);  

  moveit_msgs::msg::RobotTrajectory trajectory;
  const double jump_threshold = 0.0;
  const double eef_step = 0.01;
  double fraction = move_group.computeCartesianPath(waypoints, eef_step, jump_threshold, trajectory);
  // move_group.execute(trajectory);
```

marioesteban11, 2022-11-11:
what are you trying to do with that code @LucaBross ? Are you trying to fix those crazy moves that sometimes happens?

LucaBross, 2022-11-11:
Yes, but i observed, that still there is some crazy movement happening. Please let me know if you figure something out to fix it.

marioesteban11, 2022-11-11:
Are you using ROS2 humble right @LucaBross. If its true can you tell me which includes are you using? Iḿ in trouble with this `move_group.computeCartesianPath`.

LucaBross, 2022-11-11:
I included this in my script: 
```
#include <moveit/move_group_interface/move_group_interface.h>
#include <moveit/planning_scene_interface/planning_scene_interface.h>

#include <moveit_msgs/msg/display_robot_state.hpp>
#include <moveit_msgs/msg/display_trajectory.hpp>

#include <moveit_msgs/msg/attached_collision_object.hpp>
#include <moveit_msgs/msg/collision_object.hpp>
```
not sure which of them i need for the Cartesian Path.

marioesteban11, 2022-11-11:
Yeah, I've just find it. I found the (move_group)[https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning_interface/move_group_interface/include/moveit/move_group_interface/move_group_interface.h). Now I'm trying to figure out how it works.

marioesteban11, 2022-11-30:
Hi @LucaBross, have you finished your move_group problem? I've been busy with other stuff, and I really could not resolve it yet, do you?

shrutichakraborty, 2024-02-13:
Hello @stefanscherzinger! I am using universal robot UR10e and ros2 drivers. Currently I am using scaled joint Trajectory controller and controlling the robot motion by coding the waypoints with joint positions. However, I would also like to be able to control the robot in cartesian cordinates (position, orientation sent to the end effector) as I am using vision in my application. Is this possible with the cartesian motion controller you mention? 

Using UR's internal coding language on the teach pendant we can define linear movements (MoveL) and set offsets, I'd like to be abel to do the same via ros2, what would you suggest? I would also need to store different tools and be able to command the robot to switch between them (that is also switch the payload and TCP for different tools) from ros2 code. How can I do this? 

Thanks

closed 2024-06-19 by VinDp