UniversalRobots/Universal_Robots_ROS2_Driver #532: Question about ROS1 package Universal_Robots_ROS_passthrough_controllers

marioesteban11, 2022-11-08:
Hi everyone,

I was looking for a controller which help me to do the MoveJ and moveL Ur move, and I found that ROS1 has [this controller](https://github.com/UniversalRobots/Universal_Robots_ROS_passthrough_controllers). But I didn´t found this particular controller in ROS2 Humble. 

Do you know any controller or something that can help me?

Thanks in advance.

fmauch, 2022-11-08:
Currently, this is not ported to ROS2. May I ask about your use case why you want to use MoveJ and MoveL directly?

fmauch, 2022-11-08:
Do you plan to use ROS in your final application? If yes, I don't necessarily understand why you'll have to use MoveJ or MoveL commands rather than using the follow_joint_trajectory controller.

marioesteban11, 2022-11-10:
Im doing a c++ code that moves the UR from a pointA to a point B. In this case I'm using geometry Pose Stamped to create both points.

The problem begins when the robot stars moving and instead of moving doing one of those moves. It does crazy things like doing a 360º turn when the orientation of the robot doesnt change

fmauch, 2022-11-10:
Are you using MoveIt! to generate the motions?

marioesteban11, 2022-11-10:
Yes, Iḿ launching 3 different launchers. 

First of all i launch ur_control_launch, with fake hardware and ur3e, then I launch moveit with rviz with ur3e following ur_moveit.llaunch.py and finally I launch my movement program.

I post u my code of this last launcher
`
#include <memory>

#include <unistd.h>
#include <rclcpp/rclcpp.hpp>
#include <moveit/move_group_interface/move_group_interface.h>

#include <tf2/LinearMath/Quaternion.h>

static const rclcpp::Logger LOGGER = rclcpp::get_logger("ur3_pick");
//RCLCPP_INFO_STREAM("x: %d, y: %d, z: %d, w:%d", myQuaternion.getX(),myQuaternion.getY(), yQuaternion.getZ(), myQuaternion.getW());
int main(int argc, char* argv[])
{
  // Initialize ROS and create the Node
  rclcpp::init(argc, argv);
  auto const node = std::make_shared<rclcpp::Node>(
      "hello_moveit", rclcpp::NodeOptions().automatically_declare_parameters_from_overrides(true));

  // Create a ROS logger
  auto const logger = rclcpp::get_logger("hello_moveit");

  // Create the MoveIt MoveGroup Interface
  using moveit::planning_interface::MoveGroupInterface;
  auto move_group_interface = MoveGroupInterface(node, "ur_manipulator"); // ur_manipulator

  for (int i = 0; i < 4; i++){
    
    // Set a target Pose
    // auto const target_pose = [] {
    //   int aux = 1;
    //   if (i % 2 == 0){
    //     i = -i;
    //   }
    //   geometry_msgs::msg::Pose msg;
    //   msg.orientation.w = -0.0;
    //   msg.position.x = 0.2*aux;
    //   msg.position.y = 0.0;
    //   msg.position.z = 0.6;
    //   return msg;
    // }();
    int aux = 1;

    if (i % 2 == 0){
      aux = -aux;
    }
    geometry_msgs::msg::Pose msg;
    
    //Pose Orientation
    tf2::Quaternion myQuaternion;
    myQuaternion.setRPY( 3.14/2, 0, 0 );  // Create this quaternion from roll/pitch/yaw (in radians)
    
    // Print the quaternion components (0,0,0,1)
    RCLCPP_INFO_STREAM(LOGGER, "x: " << myQuaternion.getX() << " y: " << myQuaternion.getY() << 
                    " z: " << myQuaternion.getZ() << " w: " << myQuaternion.getW());

    msg.orientation.x = myQuaternion.getX();
    msg.orientation.y = myQuaternion.getY();
    msg.orientation.z = myQuaternion.getZ();
    msg.orientation.w = myQuaternion.getW();

    msg.position.x = 0.2*aux;
    msg.position.y = 0.0;
    msg.position.z = 0.6;

    move_group_interface.setMaxVelocityScalingFactor(1.0);
    RCLCPP_INFO(logger, "SET POSE TARGET %d ", i);
    move_group_interface.setPoseTarget(msg);

    
     // Create a plan to that target pose
    auto const [success, plan] = [&move_group_interface] {
      moveit::planning_interface::MoveGroupInterface::Plan msg;
      auto const ok = static_cast<bool>(move_group_interface.plan(msg));
      return std::make_pair(ok, msg);
    }();
    // Execute the plan
    if (success )
    {
      RCLCPP_INFO(logger, "EXECUTING PLAN %d ", i);
      move_group_interface.execute(plan);
    }
    else
    {
      RCLCPP_ERROR(logger, "Planing failed!");
      return 1 ;
    }
    sleep(1);
  }

  
  // Shutdown ROS
  rclcpp::shutdown();
  return 0;
}

`

fmauch, 2022-11-15:
The "Crazy motions" are a result of the planning process, they don't come from the controller. To my experience, the most effective way to get nicer motions using MoveIt is restricting the robot's joint capacities. For this you would create your description package and change your joint limits there. This is why you can pass the `description` package to the MoveIt! launchfile.

VinDp, 2024-06-12:
Since this is pretty old and there were no further replies, I will close the issue.

closed 2024-06-12 by VinDp