UniversalRobots/Universal_Robots_ROS2_Driver #533: Adding real time control feature for RT kernel on ROS2

ddliu365, 2022-11-09:
Hi, folks,

I am using `forward_position_control` for high frequency control application using ros2. I found out current driver doesn't work well on RT-Preempt kernel. 

The environment I am using is:

ROS2: foxy
Ubuntu: 20.04
RT-Preempt patched kernel: 5.15.76-rt53

Without the kernel, I observed the cyclic delay (about 100s) with 500Hz control frequency.

With the kernel, I observed the robot is shaking all the times with 500Hz control frequency.

What I did:

I followed the real-time procedure in Universal_Robots_ROS_Driver(ros1 driver) [link](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/17f4c63810a899fe99049ef3193f503319dae25d/ur_robot_driver/doc/real_time.md).

Then, I build the ros2 from source by following the [link](https://docs.ros.org/en/foxy/Tutorials/Demos/Real-Time-Programming.html). Afterwards, I build the `Univeral_Robots_ROS2_Driver` from source.

Do I need to use UR ros 1 driver to enable the real time control feature or did I miss some important points?

Thanks.

fmauch, 2022-11-09:
Thanks for reporting this. I usually don't test with a real-time kernel, but setup should not really be different to the ros1 setup. I'll have to take a look into that... What does 

> I observed the robot is shaking all the times with 500Hz control frequency

mean particularly? Is it really shaking in place? Does it have small hickups while moving?

ddliu365, 2022-11-09:
It is moving but I can hear very large virabtion noise. I can observer very remarkable noise as the robot sticks at some point, suddenly moved to another positon. The base where the robot mounts is vibrating during the process.

destogl, 2022-12-04:
> ROS2: foxy
> Ubuntu: 20.04
> RT-Preempt patched kernel: 5.15.76-rt53

ros2_control has added per-default support for RT-PREEMPT scheduler in Galactic (check [ros2_control_node](https://github.com/ros-controls/ros2_control/blob/galactic/controller_manager/src/ros2_control_node.cpp#L49)). You can apply the same for the node in foxy – node used in this repostory. At least for ros2_control, we are not planning to back port this functionality to Foxy because of breaking changes. 

@fmauch we have moved the support for RT_PREEMPT into ros-controls/realtime_tools. Maybe you are also interested to add this per default into node implementation in this repository? I can do a PR, but not test it well. @ddliu365 perhaps you are interesting into testing this, but on Galactic/Humble.

fmauch, 2022-12-05:
@destogl thanks for the suggestions! Since we are using RT_PREEMPT scheduling inside the RTDE communication thread from the library, as well, we won't be able to switch to the realtime_tools implementation directly, but for the driver this might definitively be interesting!

VinDp, 2024-06-19:
Everything should be solved now, since realtime support is available.
I will close this issue, but feel free to reopen it if is still relevant.

closed 2024-06-19 by VinDp