UniversalRobots/Universal_Robots_ROS2_Driver #535: Local planner gets stuck because of driver constraints

skaeringur97, 2022-11-10:
I'm implementing hybrid planning to the UR10e robot and the local planner gets stuck often, asking for a re-plan from the global planner causing this action:

https://user-images.githubusercontent.com/31903354/201063046-b55b7f32-2abc-4a6d-b02f-1e19ced651c7.mp4

After a lot of debugging I found that it's caused by these lines in ur_controllers.yaml:
```
shoulder_pan_joint: { trajectory: 0.2, goal: 0.1 }
shoulder_lift_joint: { trajectory: 0.2, goal: 0.1 }
elbow_joint: { trajectory: 0.2, goal: 0.1 }
wrist_1_joint: { trajectory: 0.2, goal: 0.1 }
wrist_2_joint: { trajectory: 0.2, goal: 0.1 }
wrist_3_joint: { trajectory: 0.2, goal: 0.1 }
```
If I increase this threshold or remove these lines, the problem goes away. Is it possible that these values are too low? Increasing them to 0.3 and 0.15 mostly solves this issue:

https://user-images.githubusercontent.com/31903354/201064915-2e6db542-3924-413c-b0e4-b219c1ab1fce.mp4

Removing these lines completely fixes this issue:

https://user-images.githubusercontent.com/31903354/201065216-7f3df691-10a2-4a69-8a0d-e4edde5783fa.mp4

skaeringur97, 2022-11-10:
Making the joint limits stricter also improves the performance, even with 0.2 and 0.1 as the tolerances

skaeringur97, 2022-11-10:
I'll close this issue since this seems to be solved by limiting joints and keeping the tolerances the same

closed 2022-11-10 by skaeringur97

marioesteban11, 2022-11-15:
Hi @skaeringur97, can u post here a video with the final result pls.

Thanks in advance

skaeringur97, 2022-11-21:
@marioesteban11 

https://user-images.githubusercontent.com/31903354/203079095-7e12c6c5-72f1-445d-9d20-1474dd5712bf.mp4

marioesteban11, 2022-11-22:
May I know, which joint_limits and ur_controllers parameters are you using? @skaeringur97

skaeringur97, 2022-11-22:
@marioesteban11 

Joint limits:
```
joint_limits:
  shoulder_pan_joint:
    # acceleration limits are not publicly available
    has_acceleration_limits: false
    has_effort_limits: true
    has_position_limits: true
    has_velocity_limits: true
    max_effort: 330.0
    max_position: !degrees  -50.0
    max_velocity: !degrees  120.0
    min_position: !degrees -200.0
  shoulder_lift_joint:
    # acceleration limits are not publicly available
    has_acceleration_limits: false
    has_effort_limits: true
    has_position_limits: true
    has_velocity_limits: true
    max_effort: 330.0
    max_position: !degrees  5.0
    max_velocity: !degrees  -90.0
    min_position: !degrees -200.0
  elbow_joint:
    # acceleration limits are not publicly available
    has_acceleration_limits: false
    has_effort_limits: true
    has_position_limits: true
    has_velocity_limits: true
    max_effort: 150.0
    # we artificially limit this joint to half its actual joint position limit
    # to avoid (MoveIt/OMPL) planning problems, as due to the physical
    # construction of the robot, it's impossible to rotate the 'elbow_joint'
    # over more than approx +- 1 pi (the shoulder lift joint gets in the way).
    #
    # This leads to planning problems as the search space will be divided into
    # two sections, with no connections from one to the other.
    #
    # Refer to https://github.com/ros-industrial/universal_robot/issues/265 for
    # more information.
    max_position: !degrees  -5.0
    max_velocity: !degrees  180.0
    min_position: !degrees -150.0
  wrist_1_joint:
    # acceleration limits are not publicly available
    has_acceleration_limits: false
    has_effort_limits: true
    has_position_limits: true
    has_velocity_limits: true
    max_effort: 56.0
    max_position: !degrees  220.0
    max_velocity: !degrees  180.0
    min_position: !degrees -220.0
  wrist_2_joint:
    # acceleration limits are not publicly available
    has_acceleration_limits: false
    has_effort_limits: true
    has_position_limits: true
    has_velocity_limits: true
    max_effort: 56.0
    max_position: !degrees  220.0
    max_velocity: !degrees  180.0
    min_position: !degrees -220.0
  wrist_3_joint:
    # acceleration limits are not publicly available
    has_acceleration_limits: false
    has_effort_limits: true
    has_position_limits: true
    has_velocity_limits: true
    max_effort: 56.0
    max_position: !degrees  220.0
    max_velocity: !degrees  180.0
    min_position: !degrees -220.0
```

Parameters I use:
```
Driver - robot_ip:=aaa.bbb.cc.d use_fake_hardware:=true initial_joint_controller:=joint_trajectory_controller
Moveit -  use_fake_hardware:=true ur_type:=ur10e use_sim_time:=true launch_servo:=false
```
Reasons explained in #503 except for launch_servo:=false, I added that because for some reason psi.getKnownObjectNames() doesn't work for objects published to a move_group node before the current node is started. If I have launch_servo:=false, it works as expected

marioesteban11, 2022-11-25:
Hi @skaeringur97 , one more question, are u using geometryPose or something different to move the UR?

I have several movement problems and I want to solve it. I`m using UR3e if it gives you more info

skaeringur97, 2022-11-25:
@marioesteban11 Yes, I make a geometry_msgs::msg::PoseStamped object and send it to the hybrid planner using this:
```
moveit_msgs::msg::Constraints goal_constraints =
kinematic_constraints::constructGoalConstraints(end_effector, target_goal, tolerance_pose, tolerance_angle);
goal_motion_request.goal_constraints.push_back(goal_constraints);
```

If you explain your problem better and maybe show a video I can maybe help more. Maybe it's even better to make a new issue

marioesteban11, 2022-11-25:
I'm gonna post how I launch all my system, a video, and my code to move the robot.

First of all I start controller:

> ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e robot_ip:=192.168.1.75 use_fake_hardware:=true launch_rviz:=false initial_joint_controller:=joint_trajectory_controller

After that I launch moveit files:
> ros2 launch ur_robot_driver prueba_grande.launch.py robot_ip:=192.168.1.75 use_fake_hardware:=true  use_sim_time:=true

This is not the original moveit file, but it just launch directly the ur3e robot.

Last step is the code launcher.

and this is the code I'm using:

    `
#include <memory>
    
    #include <unistd.h>
    #include <rclcpp/rclcpp.hpp>
    #include <moveit/move_group_interface/move_group_interface.h>
    
    #include <tf2/LinearMath/Quaternion.h>
    #include <vector>
    #include <moveit/planning_scene/planning_scene.h>
    #include <moveit/planning_scene_interface/planning_scene_interface.h>
    #include <moveit/task_constructor/task.h>
    #include <moveit/task_constructor/solvers.h>
    #include <moveit/task_constructor/stages.h>
    #if __has_include(<tf2_geometry_msgs/tf2_geometry_msgs.hpp>)
    #include <tf2_geometry_msgs/tf2_geometry_msgs.hpp>
    #else
    #include <tf2_geometry_msgs/tf2_geometry_msgs.h>
    #endif
    #if __has_include(<tf2_eigen/tf2_eigen.hpp>)
    #include <tf2_eigen/tf2_eigen.hpp>
    #else
    #include <tf2_eigen/tf2_eigen.h>
    #endif
    
    
    
    
    
    static const rclcpp::Logger LOGGER = rclcpp::get_logger("ur3_pick");
    namespace mtc = moveit::task_constructor;
    //RCLCPP_INFO_STREAM("x: %d, y: %d, z: %d, w:%d", myQuaternion.getX(),myQuaternion.getY(), yQuaternion.getZ(), myQuaternion.getW());
    int main(int argc, char* argv[])
    {
      // Initialize ROS and create the Node
      rclcpp::init(argc, argv);
      auto const node = std::make_shared<rclcpp::Node>(
          "hello_moveit", rclcpp::NodeOptions().automatically_declare_parameters_from_overrides(true));
    
      // Create a ROS logger
      auto const logger = rclcpp::get_logger("hello_moveit");
    
      // Create the MoveIt MoveGroup Interface
      using moveit::planning_interface::MoveGroupInterface;
      auto sampling_planner = std::make_shared<mtc::solvers::PipelinePlanner>(node);
      auto interpolation_planner = std::make_shared<mtc::solvers::JointInterpolationPlanner>();
      auto cartesian_planner = std::make_shared<mtc::solvers::CartesianPath>();
      auto move_group_interface = MoveGroupInterface(node, "ur_manipulator"); // ur_manipulator
      auto joints = move_group_interface.getJointNames();
      //auto frame_robot = move_group_interface.getPoseReferenceFrame();
      //auto eof = move_group_interface.getEndEffectorLink();
      //move_group_interface.setPlannerId("TrajOptDefault"); //TrajOptDefault
      auto planner_id = move_group_interface.getPlannerId();
      //auto orientation_tolerance = move_group_interface.getGoalOrientationTolerance();
      //auto joint_tolerance = move_group_interface.getGoalJointTolerance();
[... 78 lines cut ...]
        //RCLCPP_INFO_STREAM(LOGGER, "shoulder_pan_joint: " << joint_values[0] << " shoulder_lift_joint: " << joint_values[1] << 
        //                " elbow_joint: " << joint_values[2] << " wrist_1_joint: " << joint_values[3] <<
        //                " wrist_2_joint: " << joint_values[4] << " wrist_3_joint: " << joint_values[5]);
        //std::cout << "JOINT VALUES SIZE: " << joint_values.size() << std::endl;
        //for (int i = 0; i < joint_values.size();i++){
        //  std::cout << joint_values.at(i) << " ";
        //}
        //move_group_interface.getJointValueTarget(joint_values);
        //RCLCPP_INFO_STREAM(LOGGER, "\n\n Joint values: " << joint_values[0] << " " << joint_values[1] << " " << joint_values[2] << " " <<joint_values[3]  << " " <<joint_values[4]  << " "<<joint_values[5]  << " ");
        sleep(1);
      }
    
    
    
    
      
      // Shutdown ROS
      rclcpp::shutdown();
      return 0;
    }
`

And the final video:

https://user-images.githubusercontent.com/48151661/203959332-099d314d-e02b-446e-87e5-e8d8aac1555c.mp4

skaeringur97, 2022-11-25:
@marioesteban11 this seems fine, apart from the wild trajectories. What problems do you have?

marioesteban11, 2022-11-25:
@skaeringur97 exactly that, I want that all the moves are smooth, more or less like yours

skaeringur97, 2022-11-25:
@marioesteban11 You could try using different planners. If you play around in Rviz you will see that some planners take much more reasonable paths. I don't know how to change the planners myself tho. I think limiting the joint constraints is also good if that's possible. You don't necessarily lose any reachable space by doing so.