UniversalRobots/Universal_Robots_ROS2_Driver #537: Need a bit more information on how to start working with simulated robot

sampreets3, 2022-11-15:
Hi, 
I am trying to get acquainted with the universal robots in simulation. I have already succeeded in building a Docker image containing the repository and other dependencies. However, I am unable to make my robot move via MoveIt2. I think the reason is because I am having problems in connecting the URSim container with my container that has everything else. I would really appreciate if you guys can detail the steps needed to start working on a simulation _(preferably Gazebo)_.

__System Information__:
Docker image (mentioned in #487 )

__Steps Taken__:
1. build workspace
2. `ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true`
3. In a separate terminal, `docker run --rm -it -p 5900:5900 -p 6080:6080 --name ursim universalrobots/ursim_e-series`
4. Plan using the motionplanning widget of RViz

I guess I am missing a key point here, which deals with making the ursim container be able to talk to my custom container. If you can point me to resources or a tutorial which shows how I can set up my simulation, that would be much appreciated. Thanks in advance 😀

danzimmerman, 2022-11-15:
I think this is more of a Docker networking question than it is a UR question.

I am not at all an expert on this, but I think it might work if you connect both containers on the `ursim_net` Docker network that's set up in the `ur_robot_driver/scripts/start_ursim.sh` script:

https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_robot_driver/scripts/start_ursim.sh

Set up the network [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_robot_driver/scripts/start_ursim.sh#L104) (running the script will do this for you)

Start your other container on another IP address using the Docker network `ursim_net` similar to [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_robot_driver/scripts/start_ursim.sh#L108).

I believe this will allow transparent communication between the numerous ports the driver and "robot" need to open as suggested here:

https://docs.docker.com/network/bridge/#connect-a-container-to-a-user-defined-bridge

danzimmerman, 2022-11-15:
You can see all the ports that need to be opened in this forum post:

https://forum.universal-robots.com/t/running-ursim-in-windows-subsystem-for-linux/3557/13?u=dan2

On Windows/WSL2 I had to explicitly map the port ranges to get ROS and URSim to talk to each other, but that was a connection between physical computers on my network and the virtual URSim IP.

On Linux, the Docker bridge network works fine for that.

sampreets3, 2022-11-15:
Hi @danzimmerman, thanks for the quick response. You are right, it might be more of a docker thing than a UR-related question. I will try running the shell script you provided, and shall close the issue if everything goes well.

fmauch, 2022-11-16:
Specifically, please have a look inside the docs on how to start it: https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/usage.html#usage-with-official-ur-simulator

This should be sufficient to get you running.

Of course, you can do things differently such as exposing the ports from your docker container, but then you'll have to be a bit more careful on your IP setup. E.g. you'll have to pass your docker host's IP (localhost) to the driver. Unless you have a specific reason to start URSim differently, I suggest using ths `start_ursim.sh` script. The details of that script are explained [here](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/installation/ursim_docker.html#ursim-docker).

I'll go ahead and close this, since I think this is answered and not an issue with this repo. Referring to the documentation, please see #494.

If you think, that there are still unanswered points, please feel free to post and/or repoen.

closed 2022-11-16 by fmauch