UniversalRobots/Universal_Robots_ROS2_Driver #538: ur_controllers build failures in ROS 2 Rolling

nuclearsandwich, 2022-11-16:
The build for ur_controllers is failing in on the build farm for ROS 2 Rolling on Ubuntu 22.04

## Latest build failures

* https://build.ros2.org/view/Rbin_uJ64/job/Rbin_uJ64__ur_controllers__ubuntu_jammy_amd64__binary/80/
* https://build.ros2.org/view/Rbin_ujv8_uJv8/job/Rbin_ujv8_uJv8__ur_controllers__ubuntu_jammy_arm64__binary/82/

The following ur packages are also blocked by this build failure:

* ur
* ur_bringup
* ur_calibration
* ur_robot_driver

This excerpt from the build logs may also be helpful.

```
00:03:38.294 In file included from /opt/ros/rolling/include/joint_trajectory_controller/tolerances.hpp:39,
00:03:39.586                  from /opt/ros/rolling/include/joint_trajectory_controller/joint_trajectory_controller.hpp:31,
00:03:39.586                  from /tmp/binarydeb/ros-rolling-ur-controllers-2.2.4/include/ur_controllers/scaled_joint_trajectory_controller.hpp:41,
00:03:39.586                  from /tmp/binarydeb/ros-rolling-ur-controllers-2.2.4/src/scaled_joint_trajectory_controller.cpp:41:
00:03:39.586 /opt/ros/rolling/include/joint_trajectory_controller_parameters.hpp:21:10: fatal error: parameter_traits/parameter_traits.hpp: No such file or directory
00:03:39.586    21 | #include <parameter_traits/parameter_traits.hpp>
00:03:39.586       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
00:03:39.586 compilation terminated.
```

fmauch, 2022-11-16:
We've just closed #522.

TLDR: This is an issue with ros2_controllers exposing its dependencies and this is already fixed in the current ros2_controllers default branch waiting to be released.

nuclearsandwich, 2022-11-16:
> We've just closed #522.

Apologies for the crossed streams!

fmauch, 2022-11-17:
> > We've just closed #522.
> 
> Apologies for the crossed streams!

It was just funny timing :-)

fmauch, 2022-11-21:
Everything back to normal with ros2_controllers 2.14.0 released: https://build.ros2.org/job/Rbin_uJ64__ur__ubuntu_jammy_amd64__binary/

Therefore, closing this issue.

closed 2022-11-21 by fmauch