UniversalRobots/Universal_Robots_ROS2_Driver #540: Commands run and exit without clear error

wischci, 2022-11-17:
Hi all, I am trying to execute some of the instructions on a physical robot, but am encountering issues where the command runs, but exits without any clear error. To start off with, this is running on ROS2 Humble. 

Specifically, what I am seeing is that when I run the following command (after following the setup instructions in the readme):

```
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e launch_rviz:=true robot_ip:=XX.XX.XX.XX
```
Result:
```
[INFO] [launch]: All log files can be found below /.......
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executed command showed stderr output. Command: /opt/ros/humble/bin/xacro /opt/ros/humble/share/ur_description/urdf/ur.urdf.xacro robot_ip:=xxx.yyy.zzz.www joint_limit_params:=/opt/ros/humble/share/ur_description/config/ur3e/joint_limits.yaml kinematics_params:=/opt/ros/humble/share/ur_description/config/ur3e/default_kinematics.yaml physical_params:=/opt/ros/humble/share/ur_description/config/ur3e/physical_parameters.yaml visual_params:=/opt/ros/humble/share/ur_description/config/ur3e/visual_parameters.yaml safety_limits:=true safety_pos_margin:=0.15 safety_k_position:=20 name:=ur ur_type:=ur3e script_filename:=ros_control.urscript input_recipe_filename:=rtde_input_recipe.txt output_recipe_filename:=rtde_output_recipe.txt prefix:="" 
Captured stderr output:
warning: Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
warning: Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
warning: Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
warning: Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
warning: Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
```

After printing out this log, the command exits without any other info. Additionally, pinging the robot's IP is successful. Any ideas what might be happening here? This result also happens for the `ur_robot_driver ur_control.launch.py` command.

coalman321, 2022-11-19:
It looks like this issue can be caused by not having the Universal Robots ROS2 descriptions package. I encountered the same issue with a bit more verbosity. Make sure you have this package in your workspace https://github.com/UniversalRobots/Universal_Robots_ROS2_Description.

fmauch, 2022-11-24:
There was an issue with xacro being changed in humble quite drastically, which is why this came up. With the latest version of `ur_description` which has been synchronized to the repositories yesterday, things should be working fine. No need to build the description from source.

I'm going to close this, since this should be working find on an up-to-date system, but feel free to comment / re-open if the issue persists.

closed 2022-11-24 by fmauch