UniversalRobots/Universal_Robots_ROS2_Driver #541: Completed trajectory execution with status ABORTED

marioesteban11, 2022-11-18:
Hi everyone, 

I'm trying to move the ur3e robot with the URSIM, with moveit. but a have this error: 

> Completed trajectory execution with status ABORTED

I'm in:
ROS2 HUMBLE
UBUNTU 22.04

As I said before, Im trying to emulate some moves in my ur3e robot with moveit, rviz and URSIM. My problem comes when I try to move the robot with the moveit interface. Because the following error shows up:

```
[move_group-1] [INFO] [1668761715.451816518] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[rviz2-2] [INFO] [1668761715.451979040] [move_group_interface]: Plan and Execute request accepted
[move_group-1] [INFO] [1668761715.452000062] [moveit_move_group_default_capabilities.move_action_capability]: executing..
[move_group-1] [INFO] [1668761715.453266881] [moveit_move_group_default_capabilities.move_action_capability]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[move_group-1] [INFO] [1668761715.453406408] [moveit_ros.plan_execution]: Planning attempt 1 of at most 1
[move_group-1] [INFO] [1668761715.453418121] [moveit_move_group_capabilities_base.move_group_capability]: Using planning pipeline 'move_group'
[move_group-1] [INFO] [1668761715.454242965] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'ur_manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-1] [WARN] [1668761715.547564862] [moveit_trajectory_processing.time_optimal_trajectory_generation]: Joint acceleration limits are not defined. Using the default 1 rad/s^2. You can define acceleration limits in the URDF or joint_limits.yaml.
[move_group-1] [INFO] [1668761715.549095042] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1668761715.549114561] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1668761715.549182105] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[move_group-1] [INFO] [1668761715.549547939] [moveit_ros.trajectory_execution_manager]: Starting trajectory execution ...
[move_group-1] [INFO] [1668761715.549578320] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1668761715.549593870] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1668761715.549728407] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to scaled_joint_trajectory_controller
[move_group-1] [INFO] [1668761715.550150032] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: scaled_joint_trajectory_controller started execution
[move_group-1] [WARN] [1668761715.550168869] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request rejected
[move_group-1] [ERROR] [1668761715.550151505] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal was rejected by server
[move_group-1] [ERROR] [1668761715.550216303] [moveit_ros.trajectory_execution_manager]: Failed to send trajectory part 1 of 1 to controller scaled_joint_trajectory_controller
[move_group-1] [INFO] [1668761715.550224960] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
[move_group-1] [INFO] [1668761715.559667313] [moveit_move_group_default_capabilities.move_action_capability]: Solution found but controller failed during execution
[rviz2-2] [INFO] [1668761715.560031615] [move_group_interface]: Plan and Execute request aborted
[rviz2-2] [ERROR] [1668761715.651703694] [move_group_interface]: MoveGroupInterface::move() failed or timeout reached
```

And I dont really know what is the problem, if its something that I code wrongly or I just cannot move the robot with Moveit and URSIM.

PD: I'm using scaled_joint_trajectory_controller.

In another terminal whre I launch the ur_control_launch this error shows up also:

`[ur_ros2_control_node-1] [INFO] [1668761715.550004163] [scaled_joint_trajectory_controller]: Received new action goal
[ur_ros2_control_node-1] [ERROR] [1668761715.550047689] [scaled_joint_trajectory_controller]: Can't accept new action goals. Controller is not running.
`

But I dont really know how to solve it`

Thanks in advance

g-argyropoulos, 2022-11-29:
Hello, it seems to me that you have not started the external_control URCap program from the virtual pendant. I post some links from the documentation for reference:
[Start hardware simulator](https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/usage.html#start-hardware-simulator-or-mockup)
[URSim external control](https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/installation/ursim_docker.html#external-control)
[Install URCap](https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/installation/install_urcap_e_series.html)

Assuming that this is indeed your problem and since you are using the provided URSim script, the external_control URCap comes preinstalled, so you only have to run it from the pendant before you test MoveIt.

Hope this helps.

marioesteban11, 2022-11-29:
I have installed ursim, but not with the docker explained [here](https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/installation/ursim_docker.html#external-control). I have it in a virtual machine. I don´t know if this makes it works different or not

g-argyropoulos, 2022-11-29:
The real time kernel is not required. Could you please provide the full output of both the driver and the moveit node?

marioesteban11, 2022-11-29:
Sure, here you are:

`ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e robot_ip:=192.168.56.101 launch_rviz:=false`

[INFO] [launch]: All log files can be found below /home/marioest/.ros/log/2022-11-29-11-52-58-661882-marioest-170177
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [170183]
[INFO] [controller_stopper_node-3]: process started with pid [170185]
[INFO] [ur_ros2_control_node-1]: process started with pid [170181]
[INFO] [robot_state_publisher-4]: process started with pid [170187]
[INFO] [spawner-5]: process started with pid [170189]
[INFO] [spawner-6]: process started with pid [170191]
[INFO] [spawner-7]: process started with pid [170193]
[INFO] [spawner-8]: process started with pid [170195]
[INFO] [spawner-9]: process started with pid [170197]
[INFO] [spawner-10]: process started with pid [170199]
[dashboard_client-2] [INFO] [1669719179.556362397] [UR_Client_Library]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[controller_stopper_node-3] [INFO] [1669719179.557094628] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[ur_ros2_control_node-1] [INFO] [1669719179.574126797] [resource_manager]: Loading hardware 'ur3e' 
[robot_state_publisher-4] Link base_link had 2 children
[robot_state_publisher-4] Link base had 0 children
[robot_state_publisher-4] Link base_link_inertia had 1 children
[robot_state_publisher-4] Link shoulder_link had 1 children
[robot_state_publisher-4] Link upper_arm_link had 1 children
[robot_state_publisher-4] Link forearm_link had 1 children
[robot_state_publisher-4] Link wrist_1_link had 1 children
[robot_state_publisher-4] Link wrist_2_link had 1 children
[robot_state_publisher-4] Link wrist_3_link had 2 children
[robot_state_publisher-4] Link flange had 1 children
[robot_state_publisher-4] Link tool0 had 0 children
[robot_state_publisher-4] Link ft_frame had 0 children
[robot_state_publisher-4] [INFO] [1669719179.575011633] [robot_state_publisher]: got segment base
[robot_state_publisher-4] [INFO] [1669719179.575111968] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1669719179.575123069] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-4] [INFO] [1669719179.575130434] [robot_state_publisher]: got segment flange
[robot_state_publisher-4] [INFO] [1669719179.575137097] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-4] [INFO] [1669719179.575143760] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-4] [INFO] [1669719179.575150723] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-4] [INFO] [1669719179.575157817] [robot_state_publisher]: got segment tool0
[robot_state_publisher-4] [INFO] [1669719179.575164430] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-4] [INFO] [1669719179.575171644] [robot_state_publisher]: got segment world
[robot_state_publisher-4] [INFO] [1669719179.575178978] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-4] [INFO] [1669719179.575186152] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-4] [INFO] [1669719179.575192935] [robot_state_publisher]: got segment wrist_3_link
[ur_ros2_control_node-1] [INFO] [1669719179.576476675] [resource_manager]: Initialize hardware 'ur3e' 
[ur_ros2_control_node-1] [INFO] [1669719179.576563233] [resource_manager]: Successful initialization of hardware 'ur3e'
[ur_ros2_control_node-1] [INFO] [1669719179.576734355] [resource_manager]: 'configure' hardware 'ur3e' 
[ur_ros2_control_node-1] [INFO] [1669719179.576743102] [resource_manager]: Successful 'configure' of hardware 'ur3e'
[ur_ros2_control_node-1] [INFO] [1669719179.576771287] [resource_manager]: 'activate' hardware 'ur3e' 
[ur_ros2_control_node-1] [INFO] [1669719179.576781957] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1669719179.576799311] [URPositionHardwareInterface]: Initializing driver...
[ur_ros2_control_node-1] [WARN] [1669719179.577274252] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[spawner-8] [INFO] [1669719179.590691820] [spawner_force_torque_sensor_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-7] [INFO] [1669719179.590837783] [spawner_speed_scaling_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1669719179.592116303] [spawner_io_and_status_controller]: Waiting for '/controller_manager' node to exist
[spawner-10] [INFO] [1669719179.592728361] [spawner_scaled_joint_trajectory_controller]: Waiting for '/controller_manager' services to be available
[spawner-9] [INFO] [1669719179.602480558] [spawner_forward_position_controller]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1669719179.604687409] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available
[ur_ros2_control_node-1] [INFO] [1669719180.163524464] [UR_Client_Library]: Negotiated RTDE protocol version to 2.
[ur_ros2_control_node-1] [INFO] [1669719180.163677631] [UR_Client_Library]: Setting up RTDE communication with frequency 500.000000
[ur_ros2_control_node-1] [INFO] [1669719181.170036284] [URPositionHardwareInterface]: Calibration checksum: 'calib_16756443741236045476'.
[... cut ...]
[move_group-1] [INFO] [1669719226.025747610] [moveit_move_group_default_capabilities.move_action_capability]: Timeout reached
[rviz2-2] [INFO] [1669719226.025991142] [move_group_interface]: Plan and Execute request aborted
[rviz2-2] [ERROR] [1669719226.059742744] [move_group_interface]: MoveGroupInterface::move() failed or timeout reached
[move_group-1] [INFO] [1669719226.069069463] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'scaled_joint_trajectory_controller' successfully finished

marioesteban11, 2022-11-29:
I also post the `ros2 control list_controllers`

speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active    
force_torque_sensor_broadcaster[force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster] active    
io_and_status_controller[ur_controllers/GPIOController] active    
scaled_joint_trajectory_controller[ur_controllers/ScaledJointTrajectoryController] active    
forward_position_controller[position_controllers/JointGroupPositionController] inactive  
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active

marioesteban11, 2022-11-29:
@g-argyropoulos. I think I solve it, I just forget to activate this: 
[image]
And now, it seems to work. With my little program, it does crazy moves, but it works.

Thanks for all, you really help me :)

closed 2022-11-29 by marioesteban11