UniversalRobots/Universal_Robots_ROS2_Driver #549: How to build the package on Windows10 ROS2 Humble?

ZZWang21, 2022-11-27:
Thanks in advance. 

Experts, I am using ROS2 Humble on Windows 10. 

I tried to "colcon build" the package, but it keeps failing. 

Could any expert share some experience? 

Thank you very much!

RobertWilbrandt, 2022-11-28:
We currently don't support running the driver on Windows and don't have any immediate plans to change that. If you want to control a robot arm from a windows computer i suggest that you set up an intermediate control pc running e.g. an Ubuntu with RT kernel as documented [here](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/installation/real_time.html) and interact with it through the network.

I just realized that we don't document this anywhere, i'll add a note for this in the README.

fmauch, 2024-04-10:
@RobertWilbrandt Could you please take care about a notice in the README as you proposed?

VinDp, 2024-07-10:
#1053 solves this issue.