UniversalRobots/Universal_Robots_ROS2_Driver #550: Need information about how to move the UR3e robot with URsim

marioesteban11, 2022-11-29:
Hi everyone,

I'm using Ubuntu 22.04 and ROS2 Humble.

Im trying to move my UR3e robot with Ursim.

I`m launching this:

> ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e robot_ip:=192.168.1.75 use_fake_hardware:=false launch_rviz:=false initial_joint_controller:=joint_trajectory_controller

> ros2 launch ur_robot_driver prueba_grande.launch.py robot_ip:=192.168.1.75 use_fake_hardware:=false  use_sim_time:=true launch_rviz:=true 

THis last command is just the same as the ur_moveit_config launcher, launching ur3e.

I'm also launching a virtual machine with the URsim.

My problem comes when I try to move this robot with moveit that it doesn´t move. It even doen´t give an error. I post a video to see how it works.

https://user-images.githubusercontent.com/48151661/204481488-23c07959-16f5-42e6-80db-423b4d5d8ee8.mp4

I also tried to do it with scaled_joint_trajectory controller but this error shows up:

https://user-images.githubusercontent.com/48151661/204481965-efca8a62-b70c-404f-a654-eb25f938dec8.mp4

I post you the error:
[move_group-1] [ERROR] [1669711646.591221147] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Action client not connected to action server: joint_trajectory_controller/follow_joint_trajectory
[move_group-1] [ERROR] [1669711646.591225205] [moveit_ros.trajectory_execution_manager]: Failed to send trajectory part 1 of 1 to controller joint_trajectory_controller
[move_group-1] [INFO] [1669711646.591233031] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
[move_group-1] [INFO] [1669711646.601253542] [moveit_move_group_default_capabilities.move_action_capability]: Solution found but controller failed during execution
[rviz2-2] [INFO] [1669711646.601663187] [move_group_interface]: Plan and Execute request aborted
[rviz2-2] [ERROR] [1669711646.692802774] [move_group_interface]: MoveGroupInterface::move() failed or timeout reached

marioesteban11, 2022-11-29:
I.ve just resolved this issue, except for the smooth moves, in this issue #541

closed 2022-11-29 by marioesteban11

TZECHIN6, 2023-03-14:
Same here [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached

So far I can only move it in the URSim polyscope, however when i try to move the moveit interactive marker --> plan and execute, it immedately failed due to above reason...

Any one know what will be the cause?

One thing I want to pin out is that I havent installed URCap as it states its for external control... I am not sure using a URSim in a local host is categorized as external or not...

[image]

how can you solve it?