UniversalRobots/Universal_Robots_ROS2_Driver #554: Orientation problems while using URSIM with moveit

marioesteban11, 2022-12-05:
Hi everyone,

Im trying to do some programs while using ursim with moveit! and ur3e.

I have all the instalation well downloaded. And I run this following commands:

> ros2 run ur_robot_driver start_ursim.sh -m ur3e
To start the ursim vnc webpage

> ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e robot_ip:=192.168.56.101 launch_rviz:=false
To start ur3e controllers

And:
> ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e robot_ip:=192.168.56.1 use_fake_hardware:=false initial_joint_controller:=scaled_joint_trajectory_controller use_sim_time:=true
To start moveit:

After this, I also turn on the URCAPS extern controller in the vnc webpage, and with this after turning on program with urcaps in the ursim, i can move the robot using the ursim interface, and sending plans from RVIZ.

My problem comes when I try to move my robot with a c++ program. The position works perfectly, but the problem appears with the orientation that doesnt match where I want the robot to go. I will show you 2 photos, one of ursim orientation, and where i want the robot to go:

URSIM orientation:

[image]

And program execution orientation:
[image]

I also try to set the orientation like this other form but it doesnt work neither:
[image]

This orientations are all in radians.

Thanks in advance. Hope you can help me.

gavanderhoorn, 2022-12-05:
Isn't this the classical axis-angle <-> rpy 'problem'?

marioesteban11, 2022-12-05:
Sry, I don´t know what that is @gavanderhoorn ... :((

gavanderhoorn, 2022-12-05:
See [Explanation on robot orientation](https://www.universal-robots.com/articles/ur/application-installation/explanation-on-robot-orientation/) on the UR support site.

ROS defaults to using quaternions for orientations, which are often converted to RPY.

UR uses axis-angle by default. While you also see three numbers, you cannot (directly) compare them to each other.

It's been a while, but I believe you can configure the TP to display orientations in different formats. Either convert your script to display axis-angle values, or configure the TP to show you RPY.

marioesteban11, 2022-12-05:
So, I cannot set a "normal" orientation. I mean I cannot say to the robot to move to a exact orientation?

I don´t really understand how this works

fmauch, 2023-01-03:
I think I explained the problem in https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/570#issuecomment-1348195836.

marioesteban11, 2023-01-03:
Yes, that issue solved my problem, thanks

closed 2023-01-03 by marioesteban11