UniversalRobots/Universal_Robots_ROS2_Driver #569: Current Rolling binaries?

danzimmerman, 2022-12-09:
I'm not finding the driver with a fresh Rolling installation on Jammy.

Tried first `sudo apt install ros-rolling-ur`, but the metapackage is not found. Is it released?

Tried verbatim from the instructions:

```
dan@computer:~/ros/dz_ws$ sudo apt-get install ros-${ROS_DISTRO}-ur-robot-driver
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-rolling-ur-robot-driver
```

If I try 

```
apt-cache search ros-rolling | grep ur
```

I only find

```
ros-rolling-ur-client-library - Standalone C++ library for accessing Universal Robots interfaces.
ros-rolling-ur-client-library-dbgsym - debug symbols for ros-rolling-ur-client-library
ros-rolling-ur-dashboard-msgs - Messages around the UR Dashboard server.
ros-rolling-ur-dashboard-msgs-dbgsym - debug symbols for ros-rolling-ur-dashboard-msgs
ros-rolling-ur-description - URDF description for Universal Robots
ros-rolling-ur-moveit-config - An example package with MoveIt2 configurations for UR robots.
ros-rolling-ur-msgs - Message and service definitions for interacting with Universal Robots robot controllers.
ros-rolling-ur-msgs-dbgsym - debug symbols for ros-rolling-ur-msgs
```

Looks like calibration is also missing.

I'm pointing to the right repo according to the installation instructions as far as I can tell:

```
dan@computer:~/ros/dz_ws$ cat /etc/apt/sources.list.d/ros2.list 
deb [arch=amd64 signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu jammy main
```

Doesn't seem like it's there either (it's there for Humble)

http://packages.ros.org/ros2/ubuntu/dists/jammy/main/binary-amd64/Packages

Seems like for now, I can install things from the `ros2-testing` repo, but I assume this isn't normal? This is my first time trying the driver on Rolling and native Jammy.

fmauch, 2022-12-12:
The rolling build was broken a while due to upstream issues. The broken upstream repo got included in the last sync which is why ur_robot_driver got removed, see https://discourse.ros.org/t/new-packages-for-ros-2-rolling-ridley-2022-11-18/28405.

We were putting a lot of effort into humble so things stay stable there, in rolling the break did in fact happen, sorry for the inconvenience. With the next sync everything should be back to normal again. No, I don't know when this is going to happen.

danzimmerman, 2022-12-12:
Great, thank you! Just wanted to check I wasn't doing something wrong here.

closed 2024-04-10 by fmauch