UniversalRobots/Universal_Robots_ROS2_Driver #570: URSIM and RVIZ coordinates does not match

marioesteban11, 2022-12-12:
Hi everyone,

I'm in ROS2 Humble, Ubuntu 22.04 and using UR3e robot

Im using Moveit and URSIM to move an ur3e robot, but I see that coordinates in both simulators does not match.

[image]

[image]

Both of them are suposed to be in the same position: 
x = 0.12522;
y = 0.13587;
z = 0.51750;

IT seems that X and Y coordinates are denied and Z seems to be correctly.

But my problem comes with orientation. I Know that ROS works with quaternions, but UR works with euler angles, if I'm not wrong.
I was wondering, if there is a missmatch between position coordinate, how can I know if my orientation are well done or I have to do something else to fix orientations.  Thanks in advance.

fmauch, 2022-12-13:
Please note that there are two coordinate frames: `base` and `base_link`. According to https://gavanderhoorn.github.io/rep/rep-0199.html#coordinate-frames `base` is aligned with the robot_controller's base frame, so this is equivalent with what you see on the TP.

I'll add some more elaborate description of how and why in the description (see https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/issues/49), but for now this should explain the discrepancy.

When setting your reference frame in RViz to `base` instead of `world` the frame `tool0` should line up with the readings from the TP as long as your TCP has no further transformations from your flange. 

What you see on the TP is not euler angles, but a rotation vector representation. When you click on one of the fields in the "move" tab, you'll enter the a mask where you can specify a target point. Here, you can select the rotation representation. Set that to "RPY"
[image]

Now, when we look at the transform in ROS, you'll see that both, position and orientation, do match if we select `base` as reference (tf2_echo uses a cut precision).
```
ros2 run tf2_ros tf2_echo base tool0
[INFO] [1670929176.397155576] [tf2_echo]: Waiting for transform base ->  tool0: Invalid frame ID "base" passed to canTransform argument target_frame - frame does not exist
At time 1670929177.348169086
- Translation: [-0.146, -0.328, 0.797]
- Rotation: in Quaternion [-0.713, 0.478, 0.116, 0.500]
- Rotation: in RPY (radian) [-2.236, 0.699, -0.830]
- Rotation: in RPY (degree) [-128.106, 40.052, -47.565]
- Matrix:
  0.516 -0.797  0.313 -0.146
 -0.565 -0.043  0.824 -0.328
 -0.643 -0.602 -0.472  0.797
  0.000  0.000  0.000  1.000
```

TZECHIN6, 2022-12-26:
hi @marioesteban11, I am also working on URSim with Moveit2, I am using galactic and ubuntu20 for my environment.

I couldn't initial the "pointer" in Rviz to move the manipulator, is it automatically show up when you start Rviz with Moveit?

marioesteban11, 2023-01-02:
What do you mean when you say "pointer in RVIZ? Do you mean the "fixed frame" or anything else?

TZECHIN6, 2023-01-02:
sorry, the pointer I am referring to is about the interactive marker, when I follow the commands to launch the Moveit example, no interactive marker shows up, and a warning says that the scene cannot be initialised.

yinguoxiangyi, 2023-06-01:
@marioesteban11  I have a question about how did you get the robot position in Rviz like following?
```
Both of them are suposed to be in the same position:
x = 0.12522;
y = 0.13587;
z = 0.51750;

IT seems that X and Y coordinates are denied and Z seems to be correctly.
```
thank you in advance.

fmauch, 2024-04-10:
As the original question was answered in https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/570#issuecomment-1348195836 I'll close this issue.

closed 2024-04-10 by fmauch