UniversalRobots/Universal_Robots_ROS2_Driver #572: Force-Torque measurements should be attached to correct frame

fmauch, 2022-12-13:
Currently our FTS measurements are published in the tool0 frame. This is not necessarily true, which is why we [changed that on the ROS1 driver recently](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/pull/566).

Particularly, I think we should make sure they are published in the controller frame (which we currently don't have, see #571)

dpiet, 2023-10-16:
Hi @fmauch, is this still the current status of the reported issue? In ROS 2 Humble testing I've noticed that changing the active TCP on the TP affects the wrench readings, but the `frame_id` is still reporting `tool0`.

fmauch, 2023-10-18:
Yes, this is unfortunately still the state. I do have this on my current work plate though, so if nothing severely comes in between I am confident to have something in the next couple of weeks.

closed 2025-03-14 by urfeex