UniversalRobots/Universal_Robots_ROS2_Driver #575: MoveIt 2 interactive marker missing

TZECHIN6, 2022-12-28:
### Environment
- Ubuntu 20.04
- ROS2 Galactic desktop (osrf docker image)
- MoveIt 2
- ur_model: UR5e

### Issue
1. I first created a docker and mounted the ur_workspace, follow steps mentioned in the document and run `rosdep update` `rosdep install` `colcon build`.
2. The workspace works fine and all the example command can be run
3. Then I try to use **MoveIt** to remotely plan and execute the ur5e in a simulation environment (The real one is inside the lab, and I would like to do simulation to familiarize the control before working on the real one)
4. I listed the questions in below to descript the situation I am facing

### Question
1. For simulation, should I use URSim or fake_hardware? I have already setup the URSim docker and installed URCap, also I can ping it in the different docker shell
2. Assume URSim is the one I should go for, I follow the step to start MoveIt and URSim, however the problem kicks in
3. **The interactive marker doesnt appear after laucnh...** I have read the document carefully and ensure all the settings are the same. Seems that it expect to show the marker once launched, so I have no clue how to solve this issue... Am I setting the IP address wrong?

URSim docker IP: 192.168.56.101
External control host IP: 192.168.56.1 (actually should I change this to the workspace shell's IP?)
ur_workspace docker IP: 192.168.56.2

So far I could use URSim to manually move the ur5e when launch with fake_hardware

I could provide more information/picture if it is needed. Thanks for everyone help!

TZECHIN6, 2022-12-28:
Anyone could explain how should I start the Moveit with URSim setup?

So far I can only use URSim with `ur_control.launch.py` to test with `test_joint_trajectory_controller.launch.py` and manual control via URSim HMI...

TZECHIN6, 2022-12-29:
Might be anyone could successfully run MoveIt demo with URSim using ROS2 galactic can share their experience?

I am completely fine for rebuilding the project.

TZECHIN6, 2022-12-30:
Do I have to use setup assistance to setup the ur5e robot first?? I saw there is a ros-industrial package containing ur5 with moveit, but I don't see it support ROS2, what is the different between that one and this repo? This one is the official right?

fmauch, 2023-01-03:
yes, this is the official ROS2 driver for UR. The package you mentioned is part of the ROS1 support packages.

To answer a couple of your questions first:

> For simulation, should I use URSim or fake_hardware? 

Both would work. However, in my opinion using URSim is preferable, since this will be closer to your actual hardware setup. Especially, since fake hardware atm. doesn't support the scaled trajectory controller.

> External control host IP: 192.168.56.1 (actually should I change this to the workspace shell's IP?)

Yes, this should be the IP under which the URSim container can reach the machine running the ROS driver.

> The interactive marker doesnt appear after laucnh...

I assume, you followed the steps in the [README](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/galactic#using-moveit)? If everything started up correctly (Do you see "You can start planning now" in the MoveIt!-Shell?) you should be able to use the interactive marker (I assume you mean the sphere at the robot's TCP?) to create a plan target. If you are using a locale with `,` instead of `.` as a decimal separator, I've had problems in the past. To mitigate this, add a `LC_ALL=C` in front of the moveit launchfile:
```
LC_ALL=C ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true
```

I hope, this helps.

TZECHIN6, 2023-01-10:
Sorry I am a bit busy in recent days, I will update here once tested it out.

Great appreciate for your help @fmauch !!

henningkayser, 2023-01-23:
@fmauch since you are mentioning the decimals here. I've just opened https://github.com/ros-planning/moveit2/issues/1882 describing the problem and suggested fix using locale independent double parsing. This affects MoveIt but also other packages including ros2_control and the UR driver.

closed 2023-03-14 by TZECHIN6