UniversalRobots/Universal_Robots_ROS2_Driver #577: Python with UR3e robot and Moveit2

marioesteban11, 2023-01-02:
Hi everyone! 

I'm using Ros2Humble, in ubuntu 22.04, and I was wondering if any of you know how to code a launcher and a executable file to move the robot in python. I have this doubt because I only know how to code this in C++ and I didn't find any info about python yet.

Thanks in advance

fmauch, 2023-01-05:
Sorry, but this seems to be in no relation to any issue regarding this driver.

To ask general ROS implementation questions, please use https://robotics.stackexchange.com/

closed 2023-01-05 by fmauch

marioesteban11, 2023-01-05:
Why it is not related @fmauch ? I'm 100% that ROS2 can be programmed with python. And I also think, ( not 100% sure) that this driver implemented in ROS1 also can be done with python. I'm just asking if in this repo, with your controllers and all that stuff can be done with python. I see a very clear relation.

adnan-saood, 2023-01-05:
@marioesteban11 I do not know if you are referring to launch files or nodes. Probably `.launch.py` files. If so, see the bringup package for detailed examples and everything related to what you need.

launch files in ROS1 are not `.py` (only ROS2), so i do not think you are correct there either.

fmauch, 2023-01-05:
It's not related, since your question is about using this driver (even through MoveIt!). This driver provides common interfaces to be integrated into applications. Writing those applications is discoupled from the driver itself. This issue tracker is about issues with the driver. If anything from this driver is malfunctioning, please open an issue here. For questions regarding your ROS application, this is simply not the right place. I hope you understand, that we try to focus our work here on the driver itself and cannot give general ROS support.