UniversalRobots/Universal_Robots_ROS2_Driver #580: How to start using the driver as binary and in an own workspace

hagenfal, 2023-01-12:
Hello,

I'm new into programming robots and I have some problems to understand how I can use this driver.

I've downloaded the driver as binary on humble and now I have to do some changes for example in the ur_description.
As I know you should not do that directly in your global binarys.

So my question is, if I could copie these binarys into my workspace and build them there? 
(This could cause build problems, because they don't have CMakeLists.)

Or do I have to download the driver as source in every workspace? 

It would be great if you could give me a instruction of how you do it!

Thanks:D

fmauch, 2023-01-13:
Hi,

as a first hint: This seems to be no driver-specific question, but rather a ROS2 question. Such questions are generally better asked on https://robotics.stackexchange.com/ There is a larger community of people that can help you out.

To answer a part of your question: Whenever you want to change a package that you've installed as a binary package, you should download the package's sources into your workspace and build it there. Be careful to actually use the correct version of the package. For example, this repo defaults to the `main` branch, which is supposed to be used with ROS2 humble.

That being said, your question raises the question why you want to change something in the `ur_description` package. Descriptions are usually built to be included in more complex scenarios. For example, if you want to embed the robot into a larger environment URDF, you would create a new description package containing your environment and then include the UR's description in that package. I currently don't have an example at hand, though. Same goes for launchfiles. At a first glance it seems obvious to change launchfiles, e.g. for changing the default values in the original package, but often all you need is create a launchfile in your own package, include the original launchfile there and parametrize it for your use case.

I hope, this answer helps, of so feel free to close this issue.

hagenfal, 2023-01-16:
Thank you very much, this will help me out! 

I'll try it!

closed 2023-01-16 by hagenfal