UniversalRobots/Universal_Robots_ROS2_Driver #583: Error while running Example Commands for Testing the Driver

winminnn, 2023-01-19:
Execute the following code to actually connect with the robot.

`ros2 launch ur_robot_driver ur_3e.launch.py robot_ip:=<my_ROBOT_IP> launch_rviz:=false headless_mode:=true`

and turn on another terminal and run the following code to test the robot.

`ros2 launch ur_robot_driver test_scaled_joint_trajectory_controller.launch.py`

Then I get an error like below.
`[ur_ros2_control_node-1] [ERROR] [1674107641.464314633] [tolerances]: Path state tolerances failed:
[ur_ros2_control_node-1] [ERROR] [1674107641.464320565] [tolerances]: Position Error: 0.534627, Position Tolerance: 0.200000
[ur_ros2_control_node-1] [ERROR] [1674107641.464328719] [tolerances]: Path state tolerances failed:
[ur_ros2_control_node-1] [ERROR] [1674107641.464334827] [tolerances]: Position Error: 0.532808, Position Tolerance: 0.200000
[ur_ros2_control_node-1] [ERROR] [1674107641.466676397] [tolerances]: Path state tolerances failed:
[ur_ros2_control_node-1] [ERROR] [1674107641.466735490] [tolerances]: Position Error: 0.537418, Position Tolerance: 0.200000
[ur_ros2_control_node-1] [ERROR] [1674107641.466759511] [tolerances]: Path state tolerances failed:
[ur_ros2_control_node-1] [ERROR] [1674107641.466765561] [tolerances]: Position Error: 0.538860, Position Tolerance: 0.200000
[ur_ros2_control_node-1] [ERROR] [1674107641.466772647] [tolerances]: Path state tolerances failed:
[ur_ros2_control_node-1] [ERROR] [1674107641.466840133] [tolerances]: Position Error: 1.616921, Position Tolerance: 0.200000
[ur_ros2_control_node-1] [ERROR] [1674107641.466852610] [tolerances]: Path state tolerances failed:
[ur_ros2_control_node-1] [ERROR] [1674107641.466858237] [tolerances]: Position Error: 0.535095, Position Tolerance: 0.200000
[ur_ros2_control_node-1] [ERROR] [1674107641.466865344] [tolerances]: Path state tolerances failed:
[ur_ros2_control_node-1] [ERROR] [1674107641.466870953] [tolerances]: Position Error: 0.533227, Position Tolerance: 0.200000
[ur_ros2_control_node-1] [ERROR] [1674107641.468588510] [tolerances]: Path state tolerances failed:
[ur_ros2_control_node-1] [ERROR] [1674107641.468617300] [tolerances]: Position Error: 0.537799, Position Tolerance: 0.200000
[ur_ros2_control_node-1] [ERROR] [1674107641.468631604] [tolerances]: Path state tolerances failed:
[ur_ros2_control_node-1] [ERROR] [1674107641.468638055] [tolerances]: Position Error: 0.539218, Position Tolerance: 0.200000
[ur_ros2_control_node-1] [ERROR] [1674107641.468646334] [tolerances]: Path state tolerances failed:
[ur_ros2_control_node-1] [ERROR] [1674107641.468652470] [tolerances]: Position Error: 1.618054, Position Tolerance: 0.200000
[ur_ros2_control_node-1] [ERROR] [1674107641.468660915] [tolerances]: Path state tolerances failed:
[ur_ros2_control_node-1] [ERROR] [1674107641.468666941] [tolerances]: Position Error: 0.535474, Position Tolerance: 0.200000
`

Why is this error occurring? And how can we actually move the robot?

fmauch, 2023-01-27:
Hi @winminnn thanks for reporting this. Could you please tell us a little bit about your system?

- Which ROS version are you using?
- Which branch/version of this driver are you using?
- Do you use a physical robot or URSim?
- If you use a physical robot: How are you connected to the robot?

winminnn, 2023-01-30:
> Hi @winminnn thanks for reporting this. Could you please tell us a little bit about your system?
> 
> * Which ROS version are you using?
   I used the humble version and Ubuntu version is 22.04

> * Which branch/version of this driver are you using?
   I used the humble branch

> * Do you use a physical robot or URSim?
   I used physical ur3e robot 

> * If you use a physical robot: How are you connected to the robot?
   I used the command below to connect the robot in remote control mode.
   `ros2 launch ur_robot_driver ur_3e.launch.py robot_ip:=<my_ROBOT_IP> launch_rviz:=false headless_mode:=true`
   I thought headless_mode meant remote control mode, but doesn't it??

fmauch, 2023-01-30:
Thanks for coming back on this. My last question was related to the physical connection. Do you use a direct cable between the ROS pc and the robot?

winminnn, 2023-01-31:
Thanks for the answer
Yes, I am using direct connection.
But the above problem was a firewall problem and it was solved. thanks

But another problem arose.
It runs in remote control mode, but most of it doesn't run normally, the connection is automatically disconnected, and sometimes the ros pc and robot are connected normally.
So I looked for logs when I connected normally and when I couldn't connect.
When it runs normally, it looks like this

`seungmin@seungmin-Alpha:~$ ros2 launch ur_robot_driver ur3e.launch.py robot_ip:=192.168.20.35 launch_rviz:=false headless_mode:=true
[INFO] [launch]: All log files can be found below /home/seungmin/.ros/log/2023-01-30-16-52-35-688724-seungmin-Alpha-11236
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [11241]
[INFO] [controller_stopper_node-3]: process started with pid [11243]
[INFO] [ur_ros2_control_node-1]: process started with pid [11239]
[INFO] [robot_state_publisher-4]: process started with pid [11245]
[INFO] [spawner-5]: process started with pid [11247]
[INFO] [spawner-6]: process started with pid [11260]
[INFO] [spawner-7]: process started with pid [11279]
[INFO] [spawner-8]: process started with pid [11293]
[INFO] [spawner-9]: process started with pid [11296]
[INFO] [spawner-10]: process started with pid [11299]
[ur_ros2_control_node-1] [INFO] [1675065156.598621464] [resource_manager]: Loading hardware 'ur3e' 
[ur_ros2_control_node-1] [INFO] [1675065156.602745425] [resource_manager]: Initialize hardware 'ur3e' 
[ur_ros2_control_node-1] [INFO] [1675065156.602874461] [resource_manager]: Successful initialization of hardware 'ur3e'
[ur_ros2_control_node-1] [INFO] [1675065156.603187775] [resource_manager]: 'configure' hardware 'ur3e' 
[ur_ros2_control_node-1] [INFO] [1675065156.603205478] [resource_manager]: Successful 'configure' of hardware 'ur3e'
[ur_ros2_control_node-1] [INFO] [1675065156.603254350] [resource_manager]: 'activate' hardware 'ur3e' 
[ur_ros2_control_node-1] [INFO] [1675065156.603272286] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1675065156.603298554] [URPositionHardwareInterface]: Initializing driver...
[ur_ros2_control_node-1] [WARN] [1675065156.619932660] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[robot_state_publisher-4] Link base_link had 2 children
[robot_state_publisher-4] Link base had 0 children
[robot_state_publisher-4] Link base_link_inertia had 1 children
[robot_state_publisher-4] Link shoulder_link had 1 children
[robot_state_publisher-4] Link upper_arm_link had 1 children
[robot_state_publisher-4] Link forearm_link had 1 children
[robot_state_publisher-4] Link wrist_1_link had 1 children
[robot_state_publisher-4] Link wrist_2_link had 1 children
[robot_state_publisher-4] Link wrist_3_link had 2 children
[robot_state_publisher-4] Link flange had 1 children
[robot_state_publisher-4] Link tool0 had 0 children
[robot_state_publisher-4] Link ft_frame had 0 children
[robot_state_publisher-4] [INFO] [1675065156.595118313] [robot_state_publisher]: got segment base
[robot_state_publisher-4] [INFO] [1675065156.595325204] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1675065156.595345680] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-4] [INFO] [1675065156.595356921] [robot_state_publisher]: got segment flange
[robot_state_publisher-4] [INFO] [1675065156.595367195] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-4] [INFO] [1675065156.595378065] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-4] [INFO] [1675065156.595388306] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-4] [INFO] [1675065156.595398316] [robot_state_publisher]: got segment tool0
[robot_state_publisher-4] [INFO] [1675065156.595408370] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-4] [INFO] [1675065156.595418627] [robot_state_publisher]: got segment world
[robot_state_publisher-4] [INFO] [1675065156.595428770] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-4] [INFO] [1675065156.595438482] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-4] [INFO] [1675065156.595448653] [robot_state_publisher]: got segment wrist_3_link
[dashboard_client-2] [INFO] [1675065156.550394937] [UR_Client_Library]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[controller_stopper_node-3] [INFO] [1675065156.583047596] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[ur_ros2_control_node-1] [INFO] [1675065156.815830895] [UR_Client_Library]: Negotiated RTDE protocol version to 2.
[ur_ros2_control_node-1] [INFO] [1675065156.816169603] [UR_Client_Library]: Setting up RTDE communication with frequency 500.000000
[spawner-9] [INFO] [1675065157.158115354] [spawner_forward_position_controller]: Waiting for '/controller_manager' services to be available
[spawner-5] [INFO] [1675065157.308876822] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-8] [INFO] [1675065157.344210994] [spawner_force_torque_sensor_broadcaster]: Waiting for '/controller_manager' services to be available
[... cut ...] seungmin_workspace/install/ur_robot_driver/share/ur_robot_driver/config/ur_controllers.yaml'].`

When I searched for the difference, when I press ctrl-c when connected normally
`[ur_ros2_control_node-1] [INFO] [1675066050.835796164] [UR_Client_Library]: Connection to reverse interface dropped.`
While the connection is disconnected with the code, when the connection is not established, the connection is automatically disconnected with the code even without pressing ctrl-c. 

What is the problem??

fmauch, 2023-01-31:
`Connection to reverse interface dropped.` basically means that the program running on the robot (which is interpreting the commands sent from ROS through the reverse interface) stopped. This could be to multiple reasons:

* The program on the robot was manually stopped
* Another program was started on the robot ("Another program" can also be a manual motion on the TP)
* A timeout in the communication happened. The script expects a motion command to be sent every 2 ms. When you aren't using a real-time capable system, it is very well possible that this fails. This is why we recommend using a PREEMPT_RT kernel (see also https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/60f08359f6484d1671e12bab6e01fc49eb0fadc8/ur_robot_driver/doc/real_time_benchmarking.md)

I am a bit surprised that you use `headless_mode=True` on local control mode, from my understanding this should not work. I assume, that you still manually start a program on the TP? (Which is not what `headless_mode` is for).

If that's what you do, then I think this is what happens:

When using `headless_mode` the interpreting script is sent to the robot for execution right after the driver has been started. If the robot is running in local control mode, this script will simply be ignored by the robot controller. When the robot is in remote control mode, the scripts starts and connects to the reverse interface (In your output that happens at timestamp `1675066425.791171146`). Now, when you start the program on the TP as you did before, the running script gets stopped as explained above, hence the `Connection to reverse interface dropped.`
With the following statement I am not 100% sure, but I think when you interrupt our program with another program, the other program doesn't start rightaway (at least that's what I experienced when manually moving the robot while the program was running), hence you end up with no program running.

I hope this all makes sense. As the main takeaway: When using `headless_mode` in combination with remote control mode, don't manually start a program on the robot. When you are not in remote control mode, don't use `headless_mode`.

VinDp, 2024-07-03:
Since the issue is pretty old, I will close it.
Feel free to reopen if still relevant.

closed 2024-07-03 by VinDp