UniversalRobots/Universal_Robots_ROS2_Driver #588: No Controller Detected with UR simulator

Shawn-Armstrong, 2023-02-03:
I'm attempting to use the ROS2 driver in conjunction with a UR simulator using WSL2 running Ubuntu 22.04. 

I modified the docker run command in `start_ursim.sh` to support port forwarding as such:
  
```bash
docker run --rm -d -p 6080:6080 --net ursim_net --ip 192.168.56.101\`
```

Afterwards, I ran the following command using the README.md hyperlinked [instructions](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/usage.html#usage-with-official-ur-simulator). 

```Console
~$ ros2 run ur_robot_driver start_ursim.sh -m ur3e
```

PolyScope loads but no controller is detected.
  
<kbd>[image]</kbd>

- I've tried using other models with the same results. 
- I've tried adjusting PolyScope safety settings with no resolve.

fmauch, 2023-02-03:
Could you please share the full docker command that you  use to start URSim?

Shawn-Armstrong, 2023-02-03:
> Could you please share the full docker command that you use to start URSim?

This is the entire docker command from my `start_ursim.sh`:
  
```Bash
# run docker container
  docker run --rm -d -p 6080:6080 --net ursim_net --ip 192.168.56.101\
  -v "${URCAP_STORAGE}":/urcaps \
  -v "${PROGRAM_STORAGE}":/ursim/programs \
  -e ROBOT_MODEL="${ROBOT_MODEL}" \
  --name ursim \
  universalrobots/ursim_${ROBOT_SERIES} || exit
```

Shawn-Armstrong, 2023-02-09:
This issue was resolved by updating Docker Desktop. 

My version was behind by a patch number.

TZECHIN6, 2023-10-08:
how do you make the WSL2 work with the VNC application? I follow the step to start the urim in my Ubuntu 22 WSL2 environment, it direct me to a 192.168.56.101:6080 URL for the application (which use docker net), I can ping this address in my WSL2, but not in windows powershell.

Shawn-Armstrong, 2023-10-08:
> how do you make the WSL2 work with the VNC application? I follow the step to start the urim in my Ubuntu 22 WSL2 environment, it direct me to a 192.168.56.101:6080 URL for the application (which use docker net), I can ping this address in my WSL2, but not in windows powershell.

Hello, try http://localhost:6080/vnc_auto.html.

TZECHIN6, 2023-10-08:
well, i used another command to start the docker and works. The trick is to use --net host.
```
docker run --rm -it --net host -v "${HOME}/Project/ur5e_ws/resources/urcaps:/urcaps" universalrobots/ursim_e-series
```

@Shawn-Armstrong can i use the docker net then passing localhost in windows browser to access VNC server?

Shawn-Armstrong, 2023-10-08:
@TZECHIN6 Apologies, I asked this several months ago and just realized the instructions provided in the question are slightly different than what I thought they were. I drafted this [document ](https://gist.github.com/Shawn-Armstrong/65da7cbaddcfbda8489b843f0697d1a8) to specifically address your question.

My initial question is more related to using the ROS2 driver in conjunction with WSL2 / UR3e container. I was able to accomplish for anyone who is interest. These document explain my setup and usage. 
[Setup](https://gist.github.com/Shawn-Armstrong/2451cc2405b0577c4469c8b187e19ebb)
[Usage](https://gist.github.com/Shawn-Armstrong/2451cc2405b0577c4469c8b187e19ebb)

TZECHIN6, 2023-10-08:
Thanks for the document @Shawn-Armstrong , one thing i am not so clear in here is that, do your docker environment separate from the Ubuntu WSL, which means your URsim setup do not require ROS2 no matter what. Am I right?

Shawn-Armstrong, 2023-10-09:
> ...your URsim setup do not require ROS2
Correct, this setup can be used without the ROS2 driver. 

> One thing i am not so clear in here is that, do your docker environment separate from the Ubuntu WSL...
This question seems to be asking about the low level details related to Docker's deep integration between WSL2 and Windows. I'm not an expert in this area and I know it's fairly complex. 

Your questions don't really seem related to the initial issue. Out of respect for the developers in this project, I'm going to close this issue. If you have further questions, feel free to comment on the gist and I may respond when I have some time.

closed 2023-10-09 by Shawn-Armstrong