UniversalRobots/Universal_Robots_ROS2_Driver #597: SetIO service not working with fake hardware

ipa-cmh, 2023-02-20:
**Release: Humble**
Using the ``io_and_status_controller/set_io`` service blocks with fake hardware.
This is slightly annoying when trying to test an application with fake hardware.

**Steps to reproduce:**
```bash
ros2 launch ur_robot_driver ur5e.launch.py use_fake_hardware:=true robot_ip:="xxx.yyy.zzz.vvv" fake_sensor_commands:=true
```

```bash
ros2 service call /io_and_status_controller/set_io ur_msgs/srv/SetIO "{fun: 1, pin: 0, state: 1}"
```

**Things tried**
Adding ``fake_gpio_commands`` to ur_description's ur.ros_control.xacro.

```xml
        <xacro:if value="${use_fake_hardware}">
          <plugin>mock_components/GenericSystem</plugin>
          <param name="fake_sensor_commands">${fake_sensor_commands}</param>
          <param name="state_following_offset">0.0</param>
          <param name="fake_gpio_commands">true</param>
        </xacro:if>
```
This seems to not have an effect, so I suppose this might be an issue with the io_and_status_controller?

firesurfer, 2023-02-28:
I fixed this in the PRs for multiarm support. 
https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/596

firesurfer, 2023-05-24:
@fmauch  I guess we can close this now. Fix is merged now.

closed 2024-07-10 by VinDp