UniversalRobots/Universal_Robots_ROS2_Driver #601: When will ur_driver available on ROS2 humble (Ubuntu 22.04)

TZECHIN6, 2023-03-07:
The building status is fail right now... I have tried to install it  following the guideline from README, however the build is not success so far

RobertWilbrandt, 2023-03-07:
The problems in CI right now are not build failures, but errors in our execution test. We are looking into that, but that is not relevant to you being able to build or use the driver. The driver also **is** available on humble, you should be able to install it using `sudo apt install ros-humble-ur-robot-driver`.

What exactly are you trying to do and what error messages are you getting?

TZECHIN6, 2023-03-08:
I will try this again and get back in here. Thanks for letting me know about the meaning of CI status.

According to the README, I should use vcstool to import the project repos, but it tells me to import `not.release.repo` is it correct? as its actually empty.

RobertWilbrandt, 2023-03-09:
That should be correct, we add packages to that list if they perform changes that require adjustments on our side before they release them. This is currently not the case for any of our dependencies. We always keep the file around because we don't want to change the building instructions everytime we need to add a package there.

RobertWilbrandt, 2023-03-23:
With no more info on this i'll close this for now. Feel free to open a new issue if you experience problems while building.

closed 2023-03-23 by RobertWilbrandt