UniversalRobots/Universal_Robots_ROS2_Driver #602: Connection to reverse interface drops when using Moveit

alin742, 2023-03-09:
#### Description
Hello, I'm trying to test the Moveit example stated [here](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/usage.html#using-moveit). I am testing this on a ur10e robot actual hardware. 

I tired running the controller test found [here](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/usage.html#example-commands-for-testing-the-driver) and it works with no problem.

output of ur10e.launch.py
```
[ur_ros2_control_node-1] [INFO] [1678382358.439182340] [UR_Client_Library]: Robot requested program
[ur_ros2_control_node-1] [INFO] [1678382358.439456319] [UR_Client_Library]: Sent program to robot
[ur_ros2_control_node-1] [INFO] [1678382358.487151610] [UR_Client_Library]: Robot connected to reverse interface. Ready to receive control commands.
[ur_ros2_control_node-1] [INFO] [1678382391.964929835] [UR_Client_Library]: Connection to reverse interface dropped.
[ur_ros2_control_node-1] [INFO] [1678382391.965078908] [UR_Client_Library]: Robot connected to reverse interface. Ready to receive control commands.
[ur_ros2_control_node-1] [WARN] [1678382394.389241947] [UR_Client_Library]: Connection attempt on port 50001 while maximum number of clients (1) is already connected. Closing connection.
[ur_ros2_control_node-1] [INFO] [1678382394.389503611] [UR_Client_Library]: Connection to reverse interface dropped.
[ur_ros2_control_node-1] [INFO] [1678382401.189589887] [scaled_joint_trajectory_controller]: Received new action goal
[ur_ros2_control_node-1] [ERROR] [1678382401.189816606] [scaled_joint_trajectory_controller]: Can't accept new action goals. Controller is not running.
```

output of ur_moveit-launch.py
```
[move_group-1] [INFO] [1678382393.397069335] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[move_group-1] [INFO] [1678382393.397937252] [moveit_move_group_default_capabilities.move_action_capability]: executing..
[rviz2-2] [INFO] [1678382393.399366422] [move_group_interface]: Planning request accepted
[move_group-1] [INFO] [1678382393.406630866] [moveit_move_group_default_capabilities.move_action_capability]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[move_group-1] [INFO] [1678382393.406797235] [moveit_move_group_capabilities_base.move_group_capability]: Using planning pipeline 'move_group'
[move_group-1] [INFO] [1678382393.412087421] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'ur_manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-1] [WARN] [1678382394.008784156] [moveit_trajectory_processing.time_optimal_trajectory_generation]: Joint acceleration limits are not defined. Using the default 1 rad/s^2. You can define acceleration limits in the URDF or joint_limits.yaml.
[move_group-1] [INFO] [1678382394.018199170] [moveit_move_group_default_capabilities.move_action_capability]: Motion plan was computed successfully.
[rviz2-2] [INFO] [1678382394.022024057] [move_group_interface]: Planning request complete!
[rviz2-2] [INFO] [1678382394.096908176] [move_group_interface]: time taken to generate plan: 0.094288 seconds
[move_group-1] [INFO] [1678382401.174710410] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Received goal request
[move_group-1] [INFO] [1678382401.175590475] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution request received
[move_group-1] [INFO] [1678382401.175761844] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1678382401.175897676] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1678382401.176359356] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[rviz2-2] [INFO] [1678382401.175945323] [move_group_interface]: Execute request accepted
[move_group-1] [INFO] [1678382401.187071912] [moveit_ros.trajectory_execution_manager]: Starting trajectory execution ...
[move_group-1] [INFO] [1678382401.187308429] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1678382401.187421890] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1678382401.187937163] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to scaled_joint_trajectory_controller
[move_group-1] [ERROR] [1678382401.195567640] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal was rejected by server
[move_group-1] [INFO] [1678382401.195571177] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: scaled_joint_trajectory_controller started execution
[move_group-1] [WARN] [1678382401.195928840] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request rejected
[move_group-1] [ERROR] [1678382401.196237263] [moveit_ros.trajectory_execution_manager]: Failed to send trajectory part 1 of 1 to controller scaled_joint_trajectory_controller
[move_group-1] [INFO] [1678382401.196311706] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
[move_group-1] [INFO] [1678382401.196914089] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: ABORTED
[rviz2-2] [INFO] [1678382401.200963251] [move_group_interface]: Execute request aborted
[rviz2-2] [ERROR] [1678382401.274380810] [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached
```

#### Enviroment
- ROS Distro: Humble
- OS Version: Ubuntu 22.04 server (arm64)
- Binary Build

TZECHIN6, 2023-03-14:
Same here [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached

So far I can only move it in the URSim polyscope, however when i try to move the moveit interactive marker --> plan and execute, it immedately failed due to above reason...

Any one know what will be the cause?

One thing I want to pin out is that I havent installed URCap as it states its for external control... I am not sure using a URSim in a local host is categorized as external or not...

[image]

TZECHIN6, 2023-03-14:
@alin742 I just found there were some closed issue talking abnout this, they mentioned that URCap is needed for Moveit to send command to move the URSim urarm... I will try it out now

gavanderhoorn, 2023-03-14:
> I am not sure using a URSim in a local host is categorized as external or not...

URSim runs the "exact same" controller software as your real robot, it just doesn't have a physical controller/robot attached.

So, yes, you should treat URSim as a real robot and install the urcap.

TZECHIN6, 2023-03-14:
Woohooo!!! @gavanderhoorn thanks for that, I am able to do so after installing URCap and setup the external control in the polyscope. Plz let me know if you need help on this @alin742

alin742, 2023-03-14:
@TZECHIN6 Thank you for checking, I had the URCap external control installed on the actual robot when this error happened so I don't think thats the issue here.

alin742, 2023-03-14:
Do I have to have an RT Kernel on the ROS machine for this to work?

TZECHIN6, 2023-03-14:
As I will test it on real robot (ur5e) in the future, i will keep an eye on this issue loop. So far I have no idea what cause your problem, sorry~ might be a screen capture would be better to analyse it.

alin742, 2023-03-14:
I will try this again this thursday and see if i can provide a screen capture.

RobertWilbrandt, 2023-03-14:
I agree that the original issue has little to do with the ursim external_control problem.

From your console output, the following line seem relevant to me:
```
[ur_ros2_control_node-1] [WARN] [1678382394.389241947] [UR_Client_Library]: Connection attempt on port 50001 while maximum number of clients (1) is already connected. Closing connection.
```

Could you describe the network setup you are using? Is your control pc directly connected to the robot or are there other network participants? Maybe you could also take a look at #583 and see if any of the listed problems apply to you.

The problem here is not related to MoveIt, but to the connection loss from the robot to your reverse interface. This leads to our controllers getting stopped and ultimately MoveIt (correctly) complaining that the controller cannot execute its trajectory.

alin742, 2023-03-14:
The robot and the control pc are connected to the same switch (they are both on the same LAN). The steps to generate this output are as follows:
1- Starţ the ur10e.launch.py script with the robot_ip parameter set to the robots IP
2- Run the external control URCap script on the teachpendant
3- make sure that the URCap script is connected to the ur10e.launch script
4- start the ur_moveit.launch.py script
5- plan a path and execute

It is worth to note that the controller test scripts work and the robot follows the sent trajectories. It's only when I try to run the ur_moveit trajectories that the connection to the reverse interface drops

RobertWilbrandt, 2023-03-14:
The only connection between moveit running and the connection dropping i can imagine is that the planning leads to increased system load, which results in the driver missing its 2ms deadline to send a new motion goal to the robot. Can you check if you already drop the reverse interface connection when only planning? In that case, you might want to take a look at [this page](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/60f08359f6484d1671e12bab6e01fc49eb0fadc8/ur_robot_driver/doc/real_time_benchmarking.md) where we investigate different options for improving the drivers real-time performance.

alin742, 2023-03-24:
It was exactly that, I was trying it on a raspberry pi at first. I switched to a more capable pc and now it works.

> The only connection between moveit running and the connection dropping i can imagine is that the planning leads to increased system load, which results in the driver missing its 2ms deadline to send [...]

closed 2023-03-24 by alin742