UniversalRobots/Universal_Robots_ROS2_Driver #611: MoveIt Config does not plan around pinch point

MarqRazz, 2023-03-14:
On the UR hardware there exists a "pinch" volume between the `forearm_link` and `wrist2_link` where the robot controller will fault if the robot enters it. Currently MoveIt is unaware of this constraint and will plan into this volume causing the arm to fault which requires manually rescuing the robot to move it out of the area before motion can be re-enabled.
[image]

Would it be possible for Universal Robots to release documentation defining this keep out volume for each arm model?

One resolution to fix this issue would be to create an additional `pinch_link` to fill this volume that is attached to the `forearm_link` and is only collision checked with `wrist2_link`. Before starting this work I wanted to get input from the community to hear if there are other ideas on how we can fix this issue.

RobertWilbrandt, 2023-03-23:
Thank you for bringing up this issue. This seems like it was discussed for ROS1 quite some time ago in https://github.com/ros-industrial/universal_robot/issues/241. In principal i agree with their point that the meshes should faithfully model the robot geometry, but we'll have to look into options to consider this during planning.

olufskovnielsen, 2023-03-28:
Hi,
I understand this is an issue for collision checkers and path planners like MoveIt when using UR eSeries robots (this check does not exist on CB3 robots). The robot software checks if the distance from the tool flange to the lower arm is less than 2.5cm -  and if the distance from the wrist3 lid to the lower arm is less than 2.5 cm. This is done to avoid clamping fingers when moving the robot, in particular when freedriving.
An option to resolve this would be to add an extra collision geometry to the collision model for the areas where this collision can occur -  the lower arm, the tool flange, and the lid of the wrist 3 joint. I think it will resolve this issue, and not restrict workspace for other movements. It may also be possible to define a clearance for particular collisions in collision checking?
We could also disclose an algorithm to perform the check, but it would not fit very well into how path planners work...

I hope this is helpful

gavanderhoorn, 2023-03-28:
> We could also disclose an algorithm to perform the check, but it would not fit very well into how path planners work...

MoveIt does (did?) have support for custom 'constraint sampler' plugins, which might be able to model what you describe.