UniversalRobots/Universal_Robots_ROS2_Driver #617: Velocity control

felipebapa, 2023-03-27:
Hi! 
We are working in the adaptation of an UR5 robot for 3D Printing. At this moment we have developed a way for sending different points, and with an external IK program we calculate the different joint positions. 
Now, we have 2 different issues:

1. How can we control the velocity of each movement from one configuration to another?

We are not using moveII2!, only the different controllers of the driver. We know that with the `scaled_joint_controller` we can control the velocity, but it could be possible to do this from a node taking as a parameter an external variable like the flow rate of PLA... or temperature...?

2. How can we add a gripper to the robot description?

We know that it is something with tue URDF file, but I am not sure. 
Also, we are aware that this is an issue that is not so related with this repository, but someone has an answer? We would really appreciate it.

Thanks!

`DISTRIBUTION: foxy.`
`Ubuntu 20.04`

closed 2023-04-10 by felipebapa

adnan-saood, 2023-04-10:
The gripper can be added to a file with the same layout as `ur.urdf.xacro` in `UniversalRobots/Universal_Robots_ROS2_Description`

The readme of that package has the description files' structure. Once that is configured everything works great with this driver package. I can provide a sample implementation if you are still working on it.

If you want to have a nice file structure, you can add the collision and visual stl files to the yaml files mentioned in the readme.

felipebapa, 2023-04-10:
Hi! I actually solved the issue. 
Now, I am trying to do the described in [#625](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/625).

Do you know the answer?

Thank you so much!