UniversalRobots/Universal_Robots_ROS2_Driver #619: CI Failure: Missing command interfaces

RobertWilbrandt, 2023-04-06:
Our humble CI currently fails. During execution tests, the hardware interface dies due to the following exception:
```console
[ur_ros2_control_node-3] [INFO] [1680755341.288273949] [resource_manager]: Loading hardware 'ur5e' 
[ur_ros2_control_node-3] [INFO] [1680755341.295206915] [resource_manager]: Initialize hardware 'ur5e' 
[ur_ros2_control_node-3] [INFO] [1680755341.295287715] [resource_manager]: Successful initialization of hardware 'ur5e'
[ur_ros2_control_node-3] terminate called after throwing an instance of 'std::runtime_error'
[ur_ros2_control_node-3]   what():  Wrong state or command interface configuration.
[ur_ros2_control_node-3] missing state interfaces:
[ur_ros2_control_node-3] 
[ur_ros2_control_node-3] missing command interfaces:
[ur_ros2_control_node-3] ' gpio/tool_voltage_cmd '	' zero_ftsensor/zero_ftsensor_cmd '	' zero_ftsensor/zero_ftsensor_async_success '	
[dashboard_client-4] [0m[INFO] [1680755341.260516185] [UR_Client_Library]: Connected: Universal Robots Dashboard Server
[dashboard_client-4] [0m
[controller_stopper_node-5] [0m[INFO] [1680755341.303615095] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller[0m
[ERROR] [ur_ros2_control_node-3]: process has died [pid 18199, exit code -6, cmd '/root/target_ws/install/ur_robot_driver/lib/ur_robot_driver/ur_ros2_control_node --ros-args --params-file /tmp/launch_params_djvvmchi --params-file /root/target_ws/install/ur_robot_driver/share/ur_robot_driver/config/ur5e_update_rate.yaml --params-file /root/target_ws/install/ur_robot_driver/share/ur_robot_driver/config/ur_controllers.yaml'].
```

This only occurs on testing, not on main. This seems related to https://github.com/ros-controls/ros2_control/pull/953, which is now part of `ros2_control` 2.0.25.

fmauch, 2023-04-11:
See https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/582

fmauch, 2023-04-11:
This should be fixed now. The README doesn't reflect this, yet, since we show scheduled builds there, only. However, head of the humble branch seems to build (and test) fine atm, hence closing this.