UniversalRobots/Universal_Robots_ROS2_Driver #625: Using Foxy build from Source

felipebapa, 2023-04-08:
Hi! 
I am new in ros and ros2. At this moment I am trying to use the driver with [Foxy build from source](http://docs.ros.org/en/foxy/Installation/Alternatives/Ubuntu-Development-Setup.html#get-ros-2-code). I have some different errors. I don't know if this driver support this build from foxy, or if I must do something more. Thanks!

felipebapa, 2023-04-10:
Update: I am working now in with Humble in Ubuntu 22.04, but I have a problem when I try to run a node, from the README and my own. 

First, I start the driver with `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5 robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=true`

Second, I launch the publisher with `ros2 launch ur_robot_driver test_joint_trajectory_controller.launch.py`.

I have output from the terminal, it seems to be ok. Some joint angles are showed.  But I don't see changes in the robot angles.

What should I do? Thanks!

adnan-saood, 2023-04-11:
Don't use fake hardware, use the docker ursim instead. 

Here: https://hub.docker.com/r/universalrobots/ursim_cb3

After successfully setting up the ursim, write its IP address with the `robot_ip:=<>` argument.

If everything is successful, it should work right away.

fmauch, 2023-04-11:
Seconding what @adnan-saood wrote. Docker with URSim is the better option to simulate the robot. However, did you look at the test node's output? There's a security check that only moves the robot if it is close to the starting position. This is to prevent users of real robots that blindly execute this script to potentially damage their robot / environment if it is mounted in a way it would collide when executing the example trajectory.

Either bring the robot close to the starting pose manually or [remove the check](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/be751913bf5b870214dcf9648a20fffa015e8687/ur_robot_driver/config/test_goal_publishers_config.yaml#L21) in the config file (if you are using a simulated robot or understand what you're doing).

We could improve in the documentation, though, hence keeping this open with the respecting label.

adnan-saood, 2023-04-11:
I agree with you about the documentation label, when I configured the URSim last year, it was not very friendly to read from both the docker hub and the ur_driver readme.
I have to do basically the same documentation at work right now, if no one is assigned to do it I would be happy to provide it. 

best,

fmauch, 2023-04-12:
@felipebapa Since last year we restructured a lot of the documentation, where most of the stuff is moved out of the README.md. Are you aware of our [getting started guide](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/usage.html#usage-with-official-ur-simulator)? I was more referencing towards clearly stating that for simulation purposes URSim should be best used. If you feel the guide would need some extension, feel free to contribute on it. It lives [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_robot_driver/doc/usage.rst).

@felipebapa The getting started guide should help you, as well. I assume, it might be a "program on the robot not started" issue, or similar. If the guide doesn't help you, please re-post all terminal output here (from the driver and the test script).

felipebapa, 2023-04-13:
Thank you so much for your help!

The commands and the output that I have are these:

For starting the controller:

`ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy reverse_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=true`

```

[INFO] [launch]: All log files can be found below /home/felipe/.ros/log/2023-04-13-18-35-26-155649-felipe-OMEN-31934
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [31937]
[INFO] [robot_state_publisher-2]: process started with pid [31939]
[INFO] [rviz2-3]: process started with pid [31941]
[INFO] [spawner-4]: process started with pid [31943]
[INFO] [spawner-5]: process started with pid [31945]
[INFO] [spawner-6]: process started with pid [31947]
[INFO] [spawner-7]: process started with pid [31949]
[INFO] [spawner-8]: process started with pid [31951]
[INFO] [spawner-9]: process started with pid [31953]
[robot_state_publisher-2] [INFO] [1681428926.918257628] [robot_state_publisher]: got segment base
[robot_state_publisher-2] [INFO] [1681428926.918518606] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1681428926.918546155] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-2] [INFO] [1681428926.918561929] [robot_state_publisher]: got segment flange
[robot_state_publisher-2] [INFO] [1681428926.918575246] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-2] [INFO] [1681428926.918588461] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-2] [INFO] [1681428926.918601011] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-2] [INFO] [1681428926.918613599] [robot_state_publisher]: got segment tool0
[robot_state_publisher-2] [INFO] [1681428926.918625650] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-2] [INFO] [1681428926.918640915] [robot_state_publisher]: got segment world
[robot_state_publisher-2] [INFO] [1681428926.918653551] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-2] [INFO] [1681428926.918668525] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-2] [INFO] [1681428926.918681481] [robot_state_publisher]: got segment wrist_3_link
[ros2_control_node-1] [INFO] [1681428926.926759970] [resource_manager]: Loading hardware 'ur5e' 
[ros2_control_node-1] [INFO] [1681428926.928472532] [resource_manager]: Initialize hardware 'ur5e' 
[ros2_control_node-1] [WARN] [1681428926.928564290] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-1] <state_interface name="velocity"> 
[ros2_control_node-1]   <param name="initial_value">0.0</param> 
[ros2_control_node-1] </state_interface>
[ros2_control_node-1] [INFO] [1681428926.928644611] [resource_manager]: Successful initialization of hardware 'ur5e'
[ros2_control_node-1] [INFO] [1681428926.928937826] [resource_manager]: 'configure' hardware 'ur5e' 
[ros2_control_node-1] [INFO] [1681428926.928953964] [resource_manager]: Successful 'configure' of hardware 'ur5e'
[ros2_control_node-1] [INFO] [1681428926.929002580] [resource_manager]: 'activate' hardware 'ur5e' 
[ros2_control_node-1] [INFO] [1681428926.929012248] [resource_manager]: Successful 'activate' of hardware 'ur5e'
[ros2_control_node-1] [INFO] [1681428926.937732077] [controller_manager]: update rate is 500 Hz
[ros2_control_node-1] [INFO] [1681428926.941828486] [controller_manager]: RT kernel is recommended for better performance
[ros2_control_node-1] [INFO] [1681428927.567302208] [controller_manager]: Loading controller 'force_torque_sensor_broadcaster'
[ros2_control_node-1] [INFO] [1681428927.588266108] [controller_manager]: Loading controller 'joint_state_broadcaster'
[ros2_control_node-1] [INFO] [1681428927.609494180] [controller_manager]: Loading controller 'forward_position_controller'
[spawner-8] [INFO] [1681428927.620037146] [spawner_force_torque_sensor_broadcaster]: Loaded force_torque_sensor_broadcaster
[ros2_control_node-1] [INFO] [1681428927.627149815] [controller_manager]: Loading controller 'scaled_joint_trajectory_controller'
[spawner-5] [INFO] [1681428927.644345801] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-1] [INFO] [1681428927.647859038] [controller_manager]: Loading controller 'io_and_status_controller'
[spawner-9] [INFO] [1681428927.658720656] [spawner_forward_position_controller]: Loaded forward_position_controller
[ros2_control_node-1] [INFO] [1681428927.660069308] [controller_manager]: Loading controller 'speed_scaling_state_broadcaster'
[ros2_control_node-1] [INFO] [1681428927.677720879] [controller_manager]: Configuring controller 'force_torque_sensor_broadcaster'
[ros2_control_node-1] [INFO] [1681428927.679850056] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ros2_control_node-1] [INFO] [1681428927.679911828] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[spawner-4] [INFO] [1681428927.683393771] [spawner_scaled_joint_trajectory_controller]: Loaded scaled_joint_trajectory_controller
[ros2_control_node-1] [INFO] [1681428927.685486873] [controller_manager]: Configuring controller 'forward_position_controller'
[... cut ...] _joint_trajectory_controller --params-file /home/felipe/workspace/ros_ur_driver/install/ur_robot_driver/share/ur_robot_driver/config/test_goal_publishers_config.yaml'].
```

These are the steps that I used when I was working with Foxy, and the robot response was the expected. Different joint angles configurations.

Now, working with Humble, the robot model in RVIZ2, real robot and URSIM are not responding.

 I followed the steps of the guide and the Humble Branch README, with a succesfull build.

destogl, 2023-04-21:
@felipebapa you should use, because per default scaled controller is started.

```
ros2 launch ur_robot_driver test_scaled_joint_trajectory_controller.launch.py
```.

VinDp, 2024-06-19:
I will close this issue, since it's rather old and a solution was proposed.
Feel free to reopen in case it's still relevant.

closed 2024-06-19 by VinDp