UniversalRobots/Universal_Robots_ROS2_Driver #630: Controlling UR10e with Moveit Servo

robertokcanale, 2023-04-14:
Hi, I am trying to set up a UR robot and control it with MoveIt2 servo. 
I am currently using ubuntu 22.04, Ros2 Humble. I am trying to control a UR10e. So far, I am connected to the robot and can control it via Moveit Simply. Here are the 2 terminals commands needed for me. 

"
os2 launch ur_robot_driver ur_control.launch.py robot_ip:=192.168.0.100 ur_type:=ur10e  use_fake_hardware:=false launch_rviz:=false initial_joint_controller:=scaled_joint_trajectory_controller

ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur10e launch_rviz:=true use_fake_hardware:=false
"

How do I control it with Moveit2 Servo? So far, moveit launches a defalt servo node, but how do I activate it, what do i do with it, how to actually control the robot with moveit servo? Is there any guide? the basic page doesnt really work.

RobertWilbrandt, 2023-04-26:
Hey, yeah we should probably provide more documentation on that. In order to start moving with servo, you need to do two things:
1. Switch controllers such that `forward_position_controller` is running, e.g. by using `ros2 control switch_controllers --deactivate scaled_joint_trajectory_controller --activate forward_position_controller` or changing your robot launch parameters.
2. Starting servo: `ros2 service call /servo_node/start_servo std_srvs/srv/Trigger`

You should now be able to send commands to it and see the robot move, for example:
`ros2 topic pub /servo_node/delta_twist_cmds geometry_msgs/msg/TwistStamped "{ header: { stamp: 'now' },  twist: {linear: {x: -0.1}, angular: {  }}}" -r 10`

Can you verify if this works for you? For general information on how to interact with servo further, see the [MoveIt tutorial](https://moveit.picknik.ai/humble/doc/examples/realtime_servo/realtime_servo_tutorial.html) on the topic.

robertokcanale, 2023-04-28:
Hello. thanks for the reply. I followed your instructions and did a few tests and trials. It seems that everything works up to the service call to the servo node. 

When I start the servo node, it crashes or nothing happens. I try to send the twist command but nothing happens (I tried sending the twist command before and after calling the servo start. Nothing works). At the same time, I also tried to call other services, like _/servo_node/stop_servo_ exactly as you mentioned above, and this other services work. The _/start_servo/_  unfortunately doesn't work.

Controller side it looks good: 
```
io_and_status_controller[ur_controllers/GPIOController] active    
force_torque_sensor_broadcaster[force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster] active    
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active    
joint_trajectory_controller[joint_trajectory_controller/JointTrajectoryController] inactive  
speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active    
forward_position_controller[position_controllers/JointGroupPositionController] active 
```

The Servo node seems to starts correctly:
`[servo_node_main-2] [WARN] [1682655944.392024681] [moveit_servo.servo_node]: Intra-process communication is disabled, consider enabling it by adding: 
[servo_node_main-2] extra_arguments=[{'use_intra_process_comms' : True}]
[servo_node_main-2] to the Servo composable node in the launch file
[servo_node_main-2] [INFO] [1682655944.399326047] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00371013 seconds
[servo_node_main-2] [INFO] [1682655944.399364162] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[servo_node_main-2] [INFO] [1682655944.461372368] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/joint_states'
[servo_node_main-2] [INFO] [1682655944.475557574] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[servo_node_main-2] [INFO] [1682655944.475653924] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[servo_node_main-2] [INFO] [1682655944.478181060] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[servo_node_main-2] [INFO] [1682655944.478651199] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on '/servo_node/publish_planning_scene'
`

Pomorondza, 2023-05-03:
`[servo_node_main-2]   what():  Invalid argument`
If you ask me I would check the parameters you are passing to the servo node.
```
# Servo node for realtime control
 servo_yaml = load_yaml("ur_moveit_config", "config/ur_servo.yaml")
 servo_params = {"moveit_servo": servo_yaml}
 servo_node = Node(
      package="moveit_servo",
      condition=IfCondition(launch_servo),
      executable="servo_node_main",
      parameters=[
          **servo_params,
          robot_description,
          robot_description_semantic,**
      ],
      output="screen",
  )
```

robertokcanale, 2023-05-09:
Yea I managed to get that to work I believe, but the servo node is still not working...
You can have a look at my RQT graph here.... i compared it with the ones from the tutorials. First of all, my servo node is taking in twist commands but it is not throwing anything out: it says it publishes something, but it doesnt.. so it feels like the UR robot is not reading controls form anywhere... should I maye try with another controller?
[image]

Pomorondza, 2023-05-09:
Since you want to use forward_position_controller with MoveIt servo, it seems like you didn't load and activate this controller yet. On the other hand, joint_trajectory_controller is active, and this one you will need to deactivate so only forward_position_controller is active.

robertokcanale, 2023-05-10:
Yea I made the controller switch but still no , nothing happens. 

Would you mind sharing your RQT_Graph to have a look at how it works?

fornaroale, 2023-09-28:
> Hey, yeah we should probably provide more documentation on that. In order to start moving with servo, you need to do two things:
>
> 1. Switch controllers such that `forward_position_controller` is [...]

Hello, I really apologize if I reopen this rather old issue, but I'm having issues trying to setup my "digital" UR5e robot (RViz and MoveIt Humble, no real robot) on ROS2. I currently followed the following steps (given in the documentation):
- `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=false`
- `ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true`
- `ros2 control switch_controllers --deactivate scaled_joint_trajectory_controller --activate forward_position_controller`
- `ros2 service call /servo_node/start_servo std_srvs/srv/Trigger`

.. and after these, I'm publishing the following data, as you advised:
`ros2 topic pub /servo_node/delta_twist_cmds geometry_msgs/msg/TwistStamped "{ header: { stamp: 'now' },  twist: {linear: {x: -0.1}, angular: {  }}}" -r 10`

Unfortunately, the robot is not moving in RViz.
Could I kindly ask you to help me with this problem? Thank you.

eladpar, 2024-01-17:
manged to move the robot a bit:

- there is no service /servo_node/start_servo
- changing to forward position controller using 
```ros2 control switch_controllers --deactivate scaled_joint_trajectory_controller --activate forward_position_controller```

before:
```
ros2 control list_controllers 
force_torque_sensor_broadcaster[force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster] active    
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active    
forward_position_controller[position_controllers/JointGroupPositionController] inactive  
io_and_status_controller[ur_controllers/GPIOController] active    
scaled_joint_trajectory_controller[ur_controllers/ScaledJointTrajectoryController] active    
speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active
```
after:
```
ros2 control list_controllers 
force_torque_sensor_broadcaster[force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster] active    
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active    
forward_position_controller[position_controllers/JointGroupPositionController] active    
io_and_status_controller[ur_controllers/GPIOController] active    
scaled_joint_trajectory_controller[ur_controllers/ScaledJointTrajectoryController] inactive  
speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active
```

- iv'e updated the command type using : ```ros2 service call /servo_node/switch_command_type moveit_msgs/srv/ServoCommandType "{command_type: 1}"```
- published ``` ros2 topic pub /servo_node/delta_twist_cmds geometry_msgs/msg/TwistStamped "{ header: { stamp: 'now', frame_id: 'tool0'},  twist: {linear: {x: -1.3, y: 0.5}, angular: {  }}}``

the whole process:
#change type to twist:
ros2 service call /servo_node/switch_command_type moveit_msgs/srv/ServoCommandType "{command_type: 1}"
#check cforawrd position controller is active:
ros2 control list_controllers
#if not , change it by
ros2 control switch_controllers --deactivate scaled_joint_trajectory_controller --activate forward_position_controller
#check where the tool 0 is located by:
ros2 run tf2_ros tf2_echo world tool0
# look at the Translation and decide where to move it : - Translation: [-0.588, -0.040, 0.832]
#pub command
ros2 topic pub /servo_node/delta_twist_cmds geometry_msgs/msg/TwistStamped "{ header: { stamp: 'now', frame_id: 'tool0'},  twist: {linear: {x: -1.3, y: 0.5}, angular: {  }}}"

I think this should be a tutorial

shrutichakraborty, 2024-04-10:
> Hey, yeah we should probably provide more documentation on that. In order to start moving with servo, you need to do two things:
>
> 1. Switch controllers such that `forward_position_controller` is [...]

Hello @RobertWilbrandt, I am working on the same setup, I have a UR10e with ROS2 Humble. I have followed the installation on moveit2 tutorials and I have installed it in a `ws_moveit2`. I am able to launch the robot and swithc controllers but the `ros2 service call /servo_node/switch_command_type moveit_msgs/srv/ServoCommandType "{command_type: 1}"`  service does not show up for me at all. 

Any ideas?

Thanks!

IrvingF7, 2024-08-04:
> > Hey, yeah we should probably provide more documentation on that. In order to start moving with servo, you need to do two things:
> > 
> > 1. Switch controllers such that `forward_position_controller` is running, e.g. by using `ros2 control switch_controllers --deactivate scaled_joint_trajectory_controller --activate forward_position_controller` or changing your robot launch parameters.
> > 2. Starting servo: `ros2 service call /servo_node/start_servo std_srvs/srv/Trigger`
> > 
> > You should now be able to send commands to it and see the robot move, for example: `ros2 topic pub /servo_node/delta_twist_cmds geometry_msgs/msg/TwistStamped "{ header: { stamp: 'now' }, twist: {linear: {x: -0.1}, angular: { }}}" -r 10`
> > Can you verify if this works for you? For general information on how to interact with servo further, see the [MoveIt tutorial](https://moveit.picknik.ai/humble/doc/examples/realtime_servo/realtime_servo_tutorial.html) on the topic.
> 
> Hello @RobertWilbrandt, I am working on the same setup, I have a UR10e with ROS2 Humble. I have followed the installation on moveit2 tutorials and I have installed it in a `ws_moveit2`. I am able to launch the robot and swithc controllers. On calling the servo_start server `ros2 service call /servo_node/switch_command_type moveit_msgs/srv/ServoCommandType "{command_type: 1}"` and publishing commands to `/delta_twist_commands` my robot does not move as there is nothing published to "`forwrd_position_controller/commands` topic. Any ideas?
> 
> Thanks!

Hi!

If you don't have any specific reason that you have to use `forwrd_position_controller` to do the teleoperation, then I have a [working package](https://github.com/ai4ce/ur_ros2) (at least in RViz) that may be helpful. I directly publish velocities to `scaled_joint_trajectory_controller`. I will test this on the hardware in a day or two and update it properly as well.

minku1219, 2024-09-18:
Hello everyone.
Following the discussion i am able to use servo with the forward_position_control with twist_stamp msgs, but my goal is to use velocity control on UR5.
So on trying to send the velocity commands to the robot, the robot moves but in wrong direction.
Any suggestions or any example for the same is highly appreciated.