UniversalRobots/Universal_Robots_ROS2_Driver #638: Failed to extract calibration_correction from robot

johannesWen, 2023-04-24:
Hi, 
I tried to extract the calibration file from our robot, but the node dies with the following:

> [INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [calibration_correction-1]: process started with pid [44405]
[calibration_correction-1] [WARN] [1682326948.337488006] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[calibration_correction-1] [WARN] [1682326948.337516093] [UR_Client_Library]: Failed to read from stream, reconnecting in 1 seconds...
[calibration_correction-1] [INFO] [1682326949.453246809] [ur_calibration]: checksum: [2435027159 2435087802 291696670 2434238810 2435294084 2435582923 ]
[calibration_correction-1] dh_theta: [4.70162132143459e-08 -0.355966740506671 0.131086476454506 0.22487962654294 -3.12863399230812e-06 2.79138398645573e-08 ]
[calibration_correction-1] dh_a: [-8.04920862353862e-05 -0.574050656914237 -0.557095252188037 -5.53592843207487e-05 -3.36652393803452e-05 0 ]
[calibration_correction-1] dh_d: [0.181247653865116 58.3851848043911 -38.794200261633 -19.4173541597492 0.119865183659164 0.115719879264749 ]
[calibration_correction-1] dh_alpha: [1.57105582188688 -0.00365562283886373 -0.00650505651936458 1.57075868479094 -1.57137703797895 0 ]
[calibration_correction-1] calibration_status: 2
[calibration_correction-1] 
[calibration_correction-1] free(): invalid pointer
[ERROR] [calibration_correction-1]: process has died [pid 44405, exit code -6, cmd '/home/demo/colcon_ws/install/ur_calibration/lib/ur_calibration/calibration_correction --ros-args --params-file /tmp/launch_params_y1m7skyg --params-file /tmp/launch_params_ju74tac2'].

Do you have any idea what's going wrong? 

BR

fmauch, 2023-04-24:
From a first glance, everything looks alright. You seem to be connecting to the robot fine and extract the data successfully.

Since you seem to be building from source: Could you please try getting a traceback, e.g. using [backward_ros](https://github.com/pal-robotics/backward_ros/tree/foxy-devel) for the `ur_calibration` package?

johannesWen, 2023-04-24:
Yes of course. Here is the traceback:

> [INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [calibration_correction-1]: process started with pid [49244]
[calibration_correction-1] [WARN] [1682344999.512421530] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[calibration_correction-1] [WARN] [1682344999.512454472] [UR_Client_Library]: Failed to read from stream, reconnecting in 1 seconds...
[calibration_correction-1] [INFO] [1682345000.642865474] [ur_calibration]: checksum: [2435027159 2435087802 291696670 2434238810 2435294084 2435582923 ]
[calibration_correction-1] dh_theta: [4.70162132143459e-08 -0.355966740506671 0.131086476454506 0.22487962654294 -3.12863399230812e-06 2.79138398645573e-08 ]
[calibration_correction-1] dh_a: [-8.04920862353862e-05 -0.574050656914237 -0.557095252188037 -5.53592843207487e-05 -3.36652393803452e-05 0 ]
[calibration_correction-1] dh_d: [0.181247653865116 58.3851848043911 -38.794200261633 -19.4173541597492 0.119865183659164 0.115719879264749 ]
[calibration_correction-1] dh_alpha: [1.57105582188688 -0.00365562283886373 -0.00650505651936458 1.57075868479094 -1.57137703797895 0 ]
[calibration_correction-1] calibration_status: 2
[calibration_correction-1] 
[calibration_correction-1] free(): invalid pointer
[calibration_correction-1] Stack trace (most recent call last):
[calibration_correction-1] #12   Object "", at 0xffffffffffffffff, in 
[calibration_correction-1] #11   Object "/home/demo/colcon_ws/install/ur_calibration/lib/ur_calibration/calibration_correction", at 0x55c3ec418164, in _start
[calibration_correction-1] #10   Source "../csu/libc-start.c", line 392, in __libc_start_main_impl [0x7fe804e29e3f]
[calibration_correction-1] #9    Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7fe804e29d8f]
[calibration_correction-1] #8    Source "/home/demo/colcon_ws/src/digital-twin-ros2/src/Universal_Robots_ROS2_Driver/ur_calibration/src/calibration_correction.cpp", line 162, in main [0x55c3ec417caf]
[calibration_correction-1]         160:   try {
[calibration_correction-1]         161:     auto calib_node = std::make_shared<CalibrationCorrection>();
[calibration_correction-1]       > 162:     calib_node->run();
[calibration_correction-1]         163:     if (!calib_node->writeCalibrationData()) {
[calibration_correction-1]         164:       RCLCPP_ERROR_STREAM(calib_node->get_logger(), "Failed writing calibration data. See errors above for details.");
[calibration_correction-1]         165:       return -1;
[calibration_correction-1] #7  | Source "/home/demo/colcon_ws/src/digital-twin-ros2/src/Universal_Robots_ROS2_Driver/ur_calibration/src/calibration_correction.cpp", line 116, in ~URStream
[calibration_correction-1]     |   114:     calibration_data_.reset(new YAML::Node);
[calibration_correction-1]     |   115:     *calibration_data_ = consumer.getCalibrationParameters();
[calibration_correction-1]     | > 116:   }
[calibration_correction-1]     |   117: 
[calibration_correction-1]     |   118:   bool writeCalibrationData()
[calibration_correction-1]       Source "/opt/ros/humble/include/ur_client_library/comm/stream.h", line 42, in run [0x55c3ec4201f5]
[calibration_correction-1]          39:  * header type.
[calibration_correction-1]          40:  */
[calibration_correction-1]          41: template <typename T>
[calibration_correction-1]       >  42: class URStream : public TCPSocket
[calibration_correction-1]          43: {
[calibration_correction-1]          44: public:
[calibration_correction-1]          45:   /*!
[calibration_correction-1] #6    Source "./malloc/malloc.c", line 3391, in __libc_free [0x7fe804ea54d2]
[calibration_correction-1] #5    Source "./malloc/malloc.c", line 4439, in _int_free [0x7fe804ea2ac3]
[calibration_correction-1] #4    Source "./malloc/malloc.c", line 5664, in malloc_printerr [0x7fe804ea0d7b]
[calibration_correction-1] #3    Source "../sysdeps/posix/libc_fatal.c", line 155, in __stack_chk_fail [0x7fe804e896f5]
[calibration_correction-1] #2    Source "./stdlib/abort.c", line 79, in abort [0x7fe804e287f2]
[calibration_correction-1] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7fe804e42475]
[calibration_correction-1] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[calibration_correction-1]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[calibration_correction-1]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7fe804e96a7c]
[calibration_correction-1] Aborted (Signal sent by tkill() 49244 1003)
[ERROR] [calibration_correction-1]: process has died [pid 49244, exit code -6, cmd '/home/demo/colcon_ws/install/ur_calibration/lib/ur_calibration/calibration_correction --ros-args --params-file /tmp/launch_params_cjiojx_u --params-file /tmp/launch_params_o3mfnn_6'].

fmauch, 2024-07-15:
@johannesWen I will go ahead and close this due to inactivity and many positive experiences with the calibration extraction process since this issue was opened. If the issue still persists, please feel free to comment / reopen.

closed 2024-07-15 by fmauch