UniversalRobots/Universal_Robots_ROS2_Driver #662: Controlling the robot arm through /scaled_joint_trajectory_controller/joint_trajectory topic

francescosemeraro, 2023-05-02:
Good afternoon!

I would like to send joint configurations for the robot to assume through the /scaled_joint_trajectory_controller/joint_trajectory
 topic. I built a publisher that sends the exact positions of the test_scaled_joint_trajectory_controller.launch.py launch file. When I try to do that, the robot won't move. The message is received correctly by the topic, but for some reason the information is not processed. The launch file works okay. What am I missing?

santoshbalaji, 2023-11-05:
There are few things you can check. 

1. First for JointTrajectoryPoint for all fields (velocities, accelerations, effort) give same size of data given for positions. Just array with value 0 for joint names will do.
2. Not sure whether you are trying work on real robot or simulation. In general you have to provide header timestamp which is part of joint trajectory message. This timestamp will be used for filtering messages of wrong timeline.
3. Worst case still have issues then you can try actions instead of using topics.

:)

fmauch, 2023-11-06:
Just to add: Setting `time_from_start` might also be a good thing to do.

fmauch, 2024-04-10:
I assume this can be closed. If there is still anything missing please comment and / or reopen.

closed 2024-04-10 by fmauch