UniversalRobots/Universal_Robots_ROS2_Driver #674: scaled JTC does not build against humble version of ros2_controllers

firesurfer, 2023-05-15:
```
scaled_joint_trajectory_controller.cpp:274:16: error: no matching function for call to ‘ur_controllers::ScaledJointTrajectoryController::publish_state(const rclcpp::Time&, joint_trajectory_controller::JointTrajectoryController::JointTrajectoryPoint&, joint_trajectory_controller::JointTrajectoryController::JointTrajectoryPoint&, joint_trajectory_controller::JointTrajectoryController::JointTrajectoryPoint&)’
  274 |   publish_state(time, state_desired, state_current, state_error);
      |   ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```

Apparently the definition of publish_state in the JTC has changed.

WaReStu, 2023-05-23:
Okay, this seems to be exactly the same problem I encounter during a build from source as per the main page ([main-page](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver#build-from-source)).

Just to save others with really fresh installs from searching (like I did): Right now, neither the binary install of the Driver, nor the build from source work successfully for **me** on two seperate ubuntu 22.04 desktops.

Now keeping to the source build: This happens to me during Step 4:
`colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release`

> --- stderr: ur_controllers
> /home/tallarean/ws_ur/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In member function ‘virtual controller_interface::return_type ur_controllers::ScaledJointTrajectoryController::update(const rclcpp::Time&, const rclcpp::Duration&)’:
> /home/tallarean/ws_ur/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:274:16: error: no matching function for call to ‘ur_controllers::ScaledJointTrajectoryController::publish_state(const rclcpp::Time&, joint_trajectory_controller::JointTrajectoryController::JointTrajectoryPoint&, joint_trajectory_controller::JointTrajectoryController::JointTrajectoryPoint&, joint_trajectory_controller::JointTrajectoryController::JointTrajectoryPoint&)’
>   274 |   publish_state(time, state_desired, state_current, state_error);
>       |   ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> In file included from /home/tallarean/ws_ur/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/scaled_joint_trajectory_controller.hpp:41,
>                  from /home/tallarean/ws_ur/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:41:
> /opt/ros/humble/include/joint_trajectory_controller/joint_trajectory_controller.hpp:249:8: note: candidate: ‘void joint_trajectory_controller::JointTrajectoryController::publish_state(const JointTrajectoryPoint&, const JointTrajectoryPoint&, const JointTrajectoryPoint&)’
>   249 |   void publish_state(
>       |        ^~~~~~~~~~~~~
> /opt/ros/humble/include/joint_trajectory_controller/joint_trajectory_controller.hpp:249:8: note:   candidate expects 3 arguments, 4 provided
> gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:76: CMakeFiles/ur_controllers.dir/src/scaled_joint_trajectory_controller.cpp.o] Error 1

As step 5 mentions to update before filing an issue, I of course did that too, to no avail with the output:
`#All required rosdeps installed successfully`

I guess I could either try to fix it in the source code myself, hoping that it does not break anywhere else
or I could try a different version of ROS itself, as there seems to be the change that broke this build?

First I'll try my luck again with the binaries, maybe there is a way to stop the access to the gpio and zero_ftsensor that stops that from working ( see #677 ). Otherwise I'll play around with different software versions.

fmauch, 2023-05-23:
Just for reference: Are you trying to build the main branch against humble? If so, please don't we have a humble branch for a reason. We don't plan to keep the main branch compatible to "older" releases.

firesurfer, 2023-05-24:
#682 fixes it

WaReStu, 2023-05-24:
> Just for reference: Are you trying to build the main branch against humble? If so, please don't we have a humble branch for a reason. We don't plan to keep the main branch compatible to "older" releases.

This time the build from source (yes, from humble branch with git clone -b humble ...) went through. 
I can only guess that I might have missed to clone the humble branch and tried to build main (as fmauch reminded me to).
Sorry for the error and confusion. 

Sadly, it does not fix my problem, I now run into a problem at the same point where the binary instal fails for me.
@fmauch Should I open a new issue with detailed history of my installation atempts?

fmauch, 2023-08-02:
As written above it is not intended to build the main branch on humble, at all. The main branch is for the most recent changes that are meant to run with rolling. We try to port back as much as possible to humble and iron. If there is something missing in your opinion, please open a separate issue about this.

@WaReStu yes, please. If there is an issue that persists when trying to use the binary installation, please open a separate issue.