UniversalRobots/Universal_Robots_ROS2_Driver #677: Missing state interfaces / command interfaces

leander2189, 2023-05-18:
I have a setup with MoveIt and a robot_description an UR10e, which is set with UR Simulator in a docker image. This setup was working correctly until I updated to the latest binaries (Humble distro, version 2.2.6-1 from May 11th).

From that moment I am getting these errors, and the controller manager node dies:
```bash
[ur_ros2_control_node-4] [INFO] [resource_manager]: Successful initialization of hardware 'bmt_1' (initialize_hardware() at ./src/resource_manager.cpp:118)
[ur_ros2_control_node-4] terminate called after throwing an instance of 'std::runtime_error'
[ur_ros2_control_node-4]   what():  Wrong state or command interface configuration.
[ur_ros2_control_node-4] missing state interfaces:
[ur_ros2_control_node-4] ' ur_speed_scaling/speed_scaling_factor '	' ur_gpio/digital_output_0 '	' ur_gpio/digital_output_1 '	' ur_gpio/digital_output_2 '	' ur_gpio/digital_output_3 '	' ur_gpio/digital_output_4 '	' ur_gpio/digital_output_5 '	' ur_gpio/digital_output_6 '	' ur_gpio/digital_output_7 '	' ur_gpio/digital_output_8 '	' ur_gpio/digital_output_9 '	' ur_gpio/digital_output_10 '	' ur_gpio/digital_output_11 '	' ur_gpio/digital_output_12 '	' ur_gpio/digital_output_13 '	' ur_gpio/digital_output_14 '	' ur_gpio/digital_output_15 '	' ur_gpio/digital_output_16 '	' ur_gpio/digital_output_17 '	' ur_gpio/digital_input_0 '	' ur_gpio/digital_input_1 '	' ur_gpio/digital_input_2 '	' ur_gpio/digital_input_3 '	' ur_gpio/digital_input_4 '	' ur_gpio/digital_input_5 '	' ur_gpio/digital_input_6 '	' ur_gpio/digital_input_7 '	' ur_gpio/digital_input_8 '	' ur_gpio/digital_input_9 '	' ur_gpio/digital_input_10 '	' ur_gpio/digital_input_11 '	' ur_gpio/digital_input_12 '	' ur_gpio/digital_input_13 '	' ur_gpio/digital_input_14 '	' ur_gpio/digital_input_15 '	' ur_gpio/digital_input_16 '	' ur_gpio/digital_input_17 '	' ur_gpio/standard_analog_output_0 '	' ur_gpio/standard_analog_output_1 '	' ur_gpio/standard_analog_input_0 '	' ur_gpio/standard_analog_input_1 '	' ur_gpio/analog_io_type_0 '	' ur_gpio/analog_io_type_1 '	' ur_gpio/analog_io_type_2 '	' ur_gpio/analog_io_type_3 '	' ur_gpio/tool_mode '	' ur_gpio/tool_output_voltage '	' ur_gpio/tool_output_current '	' ur_gpio/tool_temperature '	' ur_gpio/tool_analog_input_0 '	' ur_gpio/tool_analog_input_1 '	' ur_gpio/tool_analog_input_type_0 '	' ur_gpio/tool_analog_input_type_1 '	' ur_gpio/robot_mode '	' ur_gpio/robot_status_bit_0 '	' ur_gpio/robot_status_bit_1 '	' ur_gpio/robot_status_bit_2 '	' ur_gpio/robot_status_bit_3 '	' ur_gpio/safety_mode '	' ur_gpio/safety_status_bit_0 '	' ur_gpio/safety_status_bit_1 '	' ur_gpio/safety_status_bit_2 '	' ur_gpio/safety_status_bit_3 '	' ur_gpio/safety_status_bit_4 '	' ur_gpio/safety_status_bit_5 '	' ur_gpio/safety_status_bit_6 '	' ur_gpio/safety_status_bit_7 '	' ur_gpio/safety_status_bit_8 '	' ur_gpio/safety_status_bit_9 '	' ur_gpio/safety_status_bit_10 '	' ur_gpio/program_running '	' ur_system_interface/initialized '	
[ur_ros2_control_node-4] missing command interfaces:
[ur_ros2_control_node-4] ' ur_speed_scaling/target_speed_fraction_cmd '	' ur_speed_scaling/target_speed_fraction_async_success '	' ur_gpio/standard_digital_output_cmd_0 '	' ur_gpio/standard_digital_output_cmd_1 '	' ur_gpio/standard_digital_output_cmd_2 '	' ur_gpio/standard_digital_output_cmd_3 '	' ur_gpio/standard_digital_output_cmd_4 '	' ur_gpio/standard_digital_output_cmd_5 '	' ur_gpio/standard_digital_output_cmd_6 '	' ur_gpio/standard_digital_output_cmd_7 '	' ur_gpio/standard_digital_output_cmd_8 '	' ur_gpio/standard_digital_output_cmd_9 '	' ur_gpio/standard_digital_output_cmd_10 '	' ur_gpio/standard_digital_output_cmd_11 '	' ur_gpio/standard_digital_output_cmd_12 '	' ur_gpio/standard_digital_output_cmd_13 '	' ur_gpio/standard_digital_output_cmd_14 '	' ur_gpio/standard_digital_output_cmd_15 '	' ur_gpio/standard_digital_output_cmd_16 '	' ur_gpio/standard_digital_output_cmd_17 '	' ur_gpio/standard_analog_output_cmd_0 '	' ur_gpio/standard_analog_output_cmd_1 '	' ur_gpio/tool_voltage_cmd '	' ur_gpio/io_async_success '	' ur_payload/mass '	' ur_payload/cog.x '	' ur_payload/cog.y '	' ur_payload/cog.z '	' ur_payload/payload_async_success '	' ur_resend_robot_program/resend_robot_program_cmd '	' ur_resend_robot_program/resend_robot_program_async_success '	' ur_zero_ftsensor/zero_ftsensor_cmd '	' ur_zero_ftsensor/zero_ftsensor_async_success '	
```

This is quite strange, as I only have enabled joint_state_broadcaster and joint_trajectory_controller in the ur_ros2_control_configuration.yaml file. Moreover, launcher file has all the unneeded controllers disabled:
``` python
    nodes_to_start = [
        # control_node,
        ur_control_node,
        dashboard_client_node,
        # tool_communication_node,
        # controller_stopper_node,
        joint_state_broadcaster_spawner,
        joint_trajectory_controller_spawner,
        # io_and_status_controller_spawner,
        # speed_scaling_state_broadcaster_spawner,
        # force_torque_sensor_broadcaster_spawner,
        # forward_position_controller_spawner_stopped,
        # initial_joint_controller_spawner_stopped,
        # initial_joint_controller_spawner_started,
    ]
```

I've been looking for that error message through the source code (both in ur_robot_driver and ros2_control) with the hope that I could understand the origin of the issue.

Can anyone point me where to look to better understand this problem?

firesurfer, 2023-05-22:
Hi @leander2189 

@fmauch I think you got the same issue when we tested #594 ? If I remember correctly it was due to some configuration issue?

JHeuverRiwo, 2023-05-22:
We are seeing similar issues on native Ubuntu 22.04, using Ros 2 Humble with latest install from binaries, using the real robot in our case; the controller crashes because it cannot find state interfaces such as the gpio and ftsensor.

WaReStu, 2023-05-22:
Same issue here, we have 
binary installed ROS2, 
binary installed MoveIt2,
binary installed ur_driver.

We are trying to get a UR3E to run, but get the same error regarding an "gpio/tool_voltage_cmd".
This in turn gives a fatal error. Afterwards, the program waits forever for controller_manager services to become available:

> [controller_stopper_node-3] [INFO] [1676529372.435627526] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
> [spawner-7] [INFO] [1676529372.975880602] [spawner_speed_scaling_state_broadcaster]: Waiting for '/controller_manager' services to be available
> [spawner-6] [INFO] [1676529373.002446651] [spawner_io_and_status_controller]: Waiting for '/controller_manager' services to be available
> [spawner-5] [INFO] [1676529373.024326283] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available
> [spawner-9] [INFO] [1676529373.032714184] [spawner_forward_position_controller]: Waiting for '/controller_manager' services to be available

...and so on.
We then purged binary ur_driver and tried to build from source, where we encounter a fatal build error:

> /localhome/demo/ws_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In member function ‘virtual controller_interface::return_type ur_controllers::ScaledJointTrajectoryController::update(const rclcpp::Time&, const rclcpp::Duration&)’:
> /localhome/demo/ws_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:274:16: error: no matching function for call to ‘ur_controllers::ScaledJointTrajectoryController::publish_state(const rclcpp::Time&, joint_trajectory_controller::JointTrajectoryController::JointTrajectoryPoint&, joint_trajectory_controller::JointTrajectoryController::JointTrajectoryPoint&, joint_trajectory_controller::JointTrajectoryController::JointTrajectoryPoint&)’
>   274 |   publish_state(time, state_desired, state_current, state_error);
>       |   ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> In file included from /localhome/demo/ws_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/scaled_joint_trajectory_controller.hpp:41,
>                  from /localhome/demo/ws_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:41:
> /opt/ros/humble/include/joint_trajectory_controller/joint_trajectory_controller.hpp:249:8: note: candidate: ‘void joint_trajectory_controller::JointTrajectoryController::publish_state(const JointTrajectoryPoint&, const JointTrajectoryPoint&, const JointTrajectoryPoint&)’
>   249 |   void publish_state(
>       |        ^~~~~~~~~~~~~
> /opt/ros/humble/include/joint_trajectory_controller/joint_trajectory_controller.hpp:249:8: note:   candidate expects 3 arguments, 4 provided
> gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:76: CMakeFiles/ur_controllers.dir/src/scaled_joint_trajectory_controller.cpp.o] Error 1
> gmake[1]: *** [CMakeFiles/Makefile2:223: CMakeFiles/ur_controllers.dir/all] Error 2
> gmake: *** [Makefile:146: all] Error 2

I'll keep an eye on this issue, any input or ideas would be great.

firesurfer, 2023-05-23:
Just to give a bit of background information on this issue: In #594 there is an additional `tf_prefix` field introduced which allows having multiple UR instances with different prefixes. Due to a bug in ros2_control which didn't allow empty strings to be passed to hardware_interface (and the controllers?) we had to include workaround. My  guess is what you see is that somehow an additional whitespace is inserted. As said before we saw this while testing but were able to quickly fix it. Unfortunately I do not remember the actual solution. 

Also if I see it correctly @leander2189 has an issue where the hardware_interface does not startup correctly  whereas @WaReStu has an issue with the gpio controller

In order to further diagnose this could you provide the following information:

1. result of: `ros2 control list_hardware_interface` ?  (prob. not possible for @leander2189 as the controller manager does not start)
2. The part of your xacro file where you declare the UR (if setup in your own system)
3. If you use a custom launch file where you override any parameters also that 
4. Perhaps also try if you can start the default launch file sucessfully

leander2189, 2023-05-23:
I think there are 2 different problems here:
* In my URDF I am setting `tf_prefix="ur_"`, and therefore the controller manager is throwing an error, even if I am defining that prefix also in the controllers.yaml file:
```yaml
speed_scaling_state_broadcaster:
  ros__parameters:
    state_publish_rate: 100.0
    tf_prefix: ur_

io_and_status_controller:
  ros__parameters:
    tf_prefix: ur_
```

* On the other hand, if I set `tf_prefix=""` in the URDF it still complains about:
    - gpio/tool_voltage_cmd
    - zero_ftsensor/zero_ftsensor_cmd
    - zero_ft_sensor/zero_ftsensor_async_succes

What I don't fully understand is why controller manager is complaining about all these when I am not launching I/O controllers in my setup (see the launch file in my first message)

As @firesurfer says, I can't execute `ros2 control list_hardware_interface` as it dies instantly.

By the way, I am launching controller manager with this code:
```python
    ur_control_node = Node(
        package="ur_robot_driver",
        executable="ur_ros2_control_node",
        parameters=[robot_description,
                    update_rate_config_file,
                    initial_joint_controllers,
                    {'use_sim_time': sim_lc_eval}],
        output={'full'},
        condition=UnlessCondition(use_fake_hardware),
    )
```

Where robot_description contains full URDF, update_rate_config_file and initial_joint_controllers are the paths to those files, which are copied from upstream

WaReStu, 2023-05-23:
@firesurfer My issue seems to be closely related to leander2189 after his `tf_prefix="ur_"` see specifically:

> On the other hand, if I set tf_prefix="" in the URDF it still complains about:
>     gpio/tool_voltage_cmd
>     zero_ftsensor/zero_ftsensor_cmd
>     zero_ft_sensor/zero_ftsensor_async_succes

These 3 points correlate to my console output, see:
> [ur_ros2_control_node-1] missing command interfaces:
> [ur_ros2_control_node-1] ' **gpio/tool_voltage_cmd** ' ' **zero_ftsensor/zero_ftsensor_cmd** ' ' **zero_ftsensor/zero_ftsensor_async_success** '
> [ERROR] [ur_ros2_control_node-1]: process has died [pid 9993, exit code -6, cmd '/opt/ros/humble/lib/ur_robot_driver/ur_ros2_control_node --ros-args --params-file /tmp/launch_params_pcb71xnt --params-file /opt/ros/humble/share/ur_robot_driver/config/ur3e_update_rate.yaml --params-file /opt/ros/humble/share/ur_robot_driver/config/ur_controllers.yaml'].

Which also seems to be an issue with @JHeuverRiwo and their clean install, which leads me to believe that this is an issue with the current binary builds and would happen to everyone, regardless of @leander2189 specific use case.

WaReStu, 2023-05-23:
> First of all please note that I am not the maintainer, just the guy who wrote #594 and the other multiarm support patches. Perhaps @fmauch (who is the maintainer) has some more input about this issue.

Right, but you seem to be quite active in this project, so you probably know more about how stuff like this is handled. But thanks for clarifying. I'm quite new to writing issues in github repos, sorry.

> @WaReStu I think you are right about this being an issue with the current binary build. I just tested it on my own setup where I run humble with two UR16e and built the UR driver from scratch. There everything worked fine.

Hm, I guess I'll try building the driver from source one more time. 
I'll create a new issue with reference to this should that not work. 
That way we can distinguish between the actual problem better.

firesurfer, 2023-05-25:
I have the same issue now in a semi binary build (ros2control from the repo - humble and building the driver from source)

As @leander2189 says the issues also occurs when setting a tf_prefix.

What is interesting is that it works fine against the mocked interfaces. Also there doesn't seem to be an additional whitespace inserted:

Excerpt from ros2_control list_hardware_interfaces

```
	ur_bot/zero_ftsensor/zero_ftsensor_async_success [available] [unclaimed]
	ur_bot/zero_ftsensor/zero_ftsensor_cmd [available] [unclaimed]
	ur_top/elbow_joint/position [available] [unclaimed]
	ur_top/elbow_joint/velocity [available] [unclaimed]
	ur_top/gpio/io_async_success [available] [unclaimed]
	ur_top/gpio/standard_analog_output_cmd_0 [available] [unclaimed]
	ur_top/gpio/standard_analog_output_cmd_1 [available] [unclaimed]
	ur_top/gpio/standard_digital_output_cmd_0 [available] [unclaimed]
``

leander2189, 2023-05-25:
Thanks for the technical support @firesurfer and @fmauch!

firesurfer, 2023-05-25:
Ok I found the issue. In #688 @fmauch forgot to remove the workaround for the command interfaces. I fixed it in #692

fmauch, 2023-05-26:
**Regarding the source build:**

I'll propagate the necessary changes to the humble branch over the weekend. However, the humble branch will fail as long as https://github.com/ros-controls/ros2_control/pull/1030 isn't merged. I expect that to happen soon, though.

**Regarding the binary build:** 

Things got broken in the binary build which should not happen. We are very sorry for that. Our tests should have spotted that, but they didn't, we'll have to improve on that.

VegaDeg, 2023-06-02:
Thanks for your helo, and I am wondering if this issue is supposed to be fixed by now. I just built the main branch on Ubuntu 22.04 and Ros2 Iron and met the same problem with @leander2189...

fmauch, 2023-06-02:
Things should be back to working normally right now. I'll do some manual tests using humble and rolling and then trigger binary releases.

fmauch, 2023-06-02:
Releases are triggered. Since the last humble sync was just last week it will roughly take a month to fix the binary packages. Builds from source should work, however. Keeping this open until next sync.

WaReStu, 2023-06-04:
> ... Builds from source should work, however.

After trying both I can confirm that building from source works, but only if one uses the repo for the semi-binary builds, i.e:
> vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_**Driver.humble.repos**

This kind of makes sense to me, as this seems to build some packages outside of the ur-driver itself and overwrites those?
So do I understand this correctly that there was some sort of change in the humble branch of ROS that lead to the driver not working? Because if I try to build from source as described in the readme.md of this humble branch (this ur-driver) it would do a binary build which has the following errors still:
(I'm going to write this out so if someone searches for this error they might be lead here:)

If you run:
> ros2 launch ur_robot_driver ur_control.launch.py ur_type:=<ur_type> robot_ip:=192.168.56.101 launch_rviz:=true

And get the following errors:
> [INFO] [ur_ros2_control_node-1]: process started with pid [15414]
> [ERROR] [ur_ros2_control_node-1]: process has died [pid 15414, exit code -6, cmd '/home/tallarean/ws_ur/ros_ur_driver/install/ur_robot_driver/lib/ur_robot_driver/ur_ros2_control_node --ros-args --params-file /tmp/launch_params_1snf_0q_ --params-file /home/tallarean/ws_ur/ros_ur_driver/install/ur_robot_driver/share/ur_robot_driver/config/ur3e_update_rate.yaml --params-file /home/tallarean/ws_ur/ros_ur_driver/install/ur_robot_driver/share/ur_robot_driver/config/ur_controllers.yaml'].
> [INFO] [robot_state_publisher-4]: process started with pid [15420]

followed later by:
> [ur_ros2_control_node-1] terminate called after throwing an instance of 'std::runtime_error'
> [ur_ros2_control_node-1]   what():  text not specified in the tf_prefix tag

Read above and issue carefully.

Anyway, we now have a somewhat working system to write code for so I am happy. I'll try the set up process using binary packages once the sync has propagated. @fmauch and everyone else: thanks for the help!

panagelak, 2023-06-05:
fyi this snippet in the hardware_interface.cpp (in the latest humble brach) throws an error for empty tf_prefix and i had to pass a dummy one char string

also why remove the first char? i think this was missed when reverting the ''_" workaround

  // Obtain the tf_prefix from the urdf so that we can have the general interface multiple times
  // NOTE using the tf_prefix at this point is some kind of workaround. One should actually go through the list of gpio
  // state interface in info_ and match them accordingly
  std::string tf_prefix = info_.hardware_parameters.at("tf_prefix");
  tf_prefix.erase(0, 1);

leander2189, 2023-06-08:
> Additionally, we usually give the UR a tf_prefix string `ur5/`, so that all links, joints etc. are prefixed by that. I had to _remove any namespacing_ in our URDF (removing all the `ur5/` prefixes) to get the full-source build to work with our current workspace.
> 

I keep having this exact same issue. I built from source following packages (in humble branch):
- Universal_Robots_ROS2_Driver
- Universal_Robots_ROS2_Description
- ros2_control

When having a tf_prefix `ur_`, controllers are not loaded correctly.

destogl, 2023-06-08:
You have to manually adjust joints names in controller setup when using prefix. There is no good way to do this dynamically in yaml files

leander2189, 2023-06-08:
> You have to manually adjust joints names in controller setup when using prefix. There is no good way to do this dynamically in yaml files

Joint names are manually adjusted and not giving any problem.

The problem is with io_and_status_controller. I have set `tf_prefix: ur_` in the controller configuration but it doesn't work.

The workaround that we are using for this situation is adding the line `tf_prefix.erase(0, 1);` that was removed [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/692/files)

Howie-Ye, 2023-06-16:
Yesterday I accepted the Ubuntu system update(cause I've to cancel it after booting) ,then I can't link and operate the UR robot. 
I found this issue and semi-binary built the latest packages(Humble , version 2.2.7) in a new workspace to test if it would work. Unfortunately, it doesn't  work and give the error as before and similar to the ones mentioned above and in #716.

`[ERROR] [ur_ros2_control_node-1]: process has died [pid 16080, exit code -6, cmd '/home/ye/ws/install/ur_robot_driver/lib/ur_robot_driver/ur_ros2_control_node --ros-args --params-file /tmp/launch_params_xtzh10fh --params-file /home/ye/ws/install/ur_robot_driver/share/ur_robot_driver/config/ur5_update_rate.yaml --params-file /home/ye/ws/install/ur_robot_driver/share/ur_robot_driver/config/ur_controllers.yaml'].`

Has this bug fixed or not? Is there something I've missed? Thanks for any help !!!

fmauch, 2023-06-19:
I've just re-checked locally on Humble and Rolling and with a from-source build everything seems to be working fine for me. The corresponding ros2_control changes have been released 2 days ago and should enter the testing repo soon so it will be included into the next sync.

leander2189, 2023-06-22:
I've just checked and it keeps crashing. I am building the following packages from source:
- ros2_control - humble branch
- Universal_Robots_ROS2_Driver - humble branch
- Universal_Robots_ROS2_Description - ros2 branch

As stated in previous comments, in my URDF I am setting prefix for UR
```xml
<xacro:ur_robot
    name="$(arg name)"
    tf_prefix="ur_"
```

Then, in the ros2 control configuration:
```yaml
speed_scaling_state_broadcaster:
  ros__parameters:
    state_publish_rate: 100.0
    tf_prefix: ur_

io_and_status_controller:
  ros__parameters:
    tf_prefix: ur_
```

I keep getting the following error:
```bash
[ur_ros2_control_node-4] terminate called after throwing an instance of 'std::runtime_error'
[ur_ros2_control_node-4]   what():  Wrong state or command interface configuration.
[ur_ros2_control_node-4] missing state interfaces:
[ur_ros2_control_node-4] ' ur_speed_scaling/speed_scaling_factor '	' ur_gpio/digital_output_0 '	' ur_gpio/digital_output_1 '	' ur_gpio/digital_output_2 '	' ur_gpio/digital_output_3 '	' ur_gpio/digital_output_4 '	' ur_gpio/digital_output_5 '	' ur_gpio/digital_output_6 '	' ur_gpio/digital_output_7 '	' ur_gpio/digital_output_8 '	' ur_gpio/digital_output_9 '	' ur_gpio/digital_output_10 '	' ur_gpio/digital_output_11 '	' ur_gpio/digital_output_12 '	' ur_gpio/digital_output_13 '	' ur_gpio/digital_output_14 '	' ur_gpio/digital_output_15 '	' ur_gpio/digital_output_16 '	' ur_gpio/digital_output_17 '	' ur_gpio/digital_input_0 '	' ur_gpio/digital_input_1 '	' ur_gpio/digital_input_2 '	' ur_gpio/digital_input_3 '	' ur_gpio/digital_input_4 '	' ur_gpio/digital_input_5 '	' ur_gpio/digital_input_6 '	' ur_gpio/digital_input_7 '	' ur_gpio/digital_input_8 '	' ur_gpio/digital_input_9 '	' ur_gpio/digital_input_10 '	' ur_gpio/digital_input_11 '	' ur_gpio/digital_input_12 '	' ur_gpio/digital_input_13 '	' ur_gpio/digital_input_14 '	' ur_gpio/digital_input_15 '	' ur_gpio/digital_input_16 '	' ur_gpio/digital_input_17 '	' ur_gpio/standard_analog_output_0 '	' ur_gpio/standard_analog_output_1 '	' ur_gpio/standard_analog_input_0 '	' ur_gpio/standard_analog_input_1 '	' ur_gpio/analog_io_type_0 '	' ur_gpio/analog_io_type_1 '	' ur_gpio/analog_io_type_2 '	' ur_gpio/analog_io_type_3 '	' ur_gpio/tool_mode '	' ur_gpio/tool_output_voltage '	' ur_gpio/tool_output_current '	' ur_gpio/tool_temperature '	' ur_gpio/tool_analog_input_0 '	' ur_gpio/tool_analog_input_1 '	' ur_gpio/tool_analog_input_type_0 '	' ur_gpio/tool_analog_input_type_1 '	' ur_gpio/robot_mode '	' ur_gpio/robot_status_bit_0 '	' ur_gpio/robot_status_bit_1 '	' ur_gpio/robot_status_bit_2 '	' ur_gpio/robot_status_bit_3 '	' ur_gpio/safety_mode '	' ur_gpio/safety_status_bit_0 '	' ur_gpio/safety_status_bit_1 '	' ur_gpio/safety_status_bit_2 '	' ur_gpio/safety_status_bit_3 '	' ur_gpio/safety_status_bit_4 '	' ur_gpio/safety_status_bit_5 '	' ur_gpio/safety_status_bit_6 '	' ur_gpio/safety_status_bit_7 '	' ur_gpio/safety_status_bit_8 '	' ur_gpio/safety_status_bit_9 '	' ur_gpio/safety_status_bit_10 '	' ur_gpio/program_running '	' ur_system_interface/initialized '	
[ur_ros2_control_node-4] missing command interfaces:
[ur_ros2_control_node-4] ' ur_speed_scaling/target_speed_fraction_cmd '	' ur_speed_scaling/target_speed_fraction_async_success '	' ur_gpio/standard_digital_output_cmd_0 '	' ur_gpio/standard_digital_output_cmd_1 '	' ur_gpio/standard_digital_output_cmd_2 '	' ur_gpio/standard_digital_output_cmd_3 '	' ur_gpio/standard_digital_output_cmd_4 '	' ur_gpio/standard_digital_output_cmd_5 '	' ur_gpio/standard_digital_output_cmd_6 '	' ur_gpio/standard_digital_output_cmd_7 '	' ur_gpio/standard_digital_output_cmd_8 '	' ur_gpio/standard_digital_output_cmd_9 '	' ur_gpio/standard_digital_output_cmd_10 '	' ur_gpio/standard_digital_output_cmd_11 '	' ur_gpio/standard_digital_output_cmd_12 '	' ur_gpio/standard_digital_output_cmd_13 '	' ur_gpio/standard_digital_output_cmd_14 '	' ur_gpio/standard_digital_output_cmd_15 '	' ur_gpio/standard_digital_output_cmd_16 '	' ur_gpio/standard_digital_output_cmd_17 '	' ur_gpio/standard_analog_output_cmd_0 '	' ur_gpio/standard_analog_output_cmd_1 '	' ur_gpio/tool_voltage_cmd '	' ur_gpio/io_async_success '	' ur_payload/mass '	' ur_payload/cog.x '	' ur_payload/cog.y '	' ur_payload/cog.z '	' ur_payload/payload_async_success '	' ur_resend_robot_program/resend_robot_program_cmd '	' ur_resend_robot_program/resend_robot_program_async_success '	' ur_hand_back_control/hand_back_control_cmd '	' ur_hand_back_control/hand_back_control_async_success '	' ur_zero_ftsensor/zero_ftsensor_cmd '	' ur_zero_ftsensor/zero_ftsensor_async_success '	
```

gavanderhoorn, 2023-06-22:
> `' ur_speed_scaling/speed_scaling_factor `

are the spaces before and after the name supposed to be there?

leander2189, 2023-06-22:
> are the spaces before and after the name supposed to be there?

No, they shouldn't be there. I think that's the source of the issue, somewhere in the code that space is being added.

>Does it work without the prefix?

Yes, it works without the prefix

fmauch, 2023-06-22:
> > `' ur_speed_scaling/speed_scaling_factor `
> 
> are the spaces before and after the name supposed to be there?
> 
> Does it work without the prefix?

The spaces come from `resource_manager`:

https://github.com/ros-controls/ros2_control/blob/0dfe01a875079d53e02a9e16138b1bc01b524823/hardware_interface/src/resource_manager.cpp#LL1092C25-L1092C25

I'll investigate further.

fmauch, 2023-06-23:
@leander2189 thanks for sticking with this. We were indeed missing a backport of #688 and #692 on Humble. I tested with #725 and it seems to be working fine.

TZECHIN6, 2023-06-25:
https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/724
I am also having the same kind of problem. I have gone through all the discussions above, is the Humble branch currently stable to do development? I am still not quite sure how to fix this, is it just adding the line https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/692/files shown in here is good to go?

fmauch, 2023-06-28:
Just to add: The humble branch is fine now given a recent enough ros2_control. To achieve this either build ros2_control from sources, use the testing repo or wait until after the next sync, which should be happening tomorrow.

leander2189, 2023-06-29:
Is it already corrected? I just pulled latest sources and keeps giving the same error

fmauch, 2023-06-29:
This works for me:

```
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.56.101 launch_rviz:=false tf_prefix:=my_ur_ controllers_file:="$(pwd)/ur_robot_driver/config/ur_controllers_prefixed.yaml"
```

with 
<details>
<summary>ur_controllers_prefixed.yaml</summary>

```yaml
controller_manager:
  ros__parameters:
    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    io_and_status_controller:
      type: ur_controllers/GPIOController

    speed_scaling_state_broadcaster:
      type: ur_controllers/SpeedScalingStateBroadcaster

    force_torque_sensor_broadcaster:
      type: force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster

    joint_trajectory_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    scaled_joint_trajectory_controller:
      type: ur_controllers/ScaledJointTrajectoryController

    forward_velocity_controller:
      type: velocity_controllers/JointGroupVelocityController

    forward_position_controller:
      type: position_controllers/JointGroupPositionController

speed_scaling_state_broadcaster:
  ros__parameters:
    state_publish_rate: 100.0
    tf_prefix: "my_ur_"

io_and_status_controller:
  ros__parameters:
    tf_prefix: "my_ur_"

force_torque_sensor_broadcaster:
  ros__parameters:
    sensor_name: my_ur_tcp_fts_sensor
    state_interface_names:
      - force.x
      - force.y
      - force.z
      - torque.x
      - torque.y
      - torque.z
    frame_id: my_ur_tool0
    topic_name: ft_data

joint_trajectory_controller:
  ros__parameters:
    joints:
      - my_ur_shoulder_pan_joint
      - my_ur_shoulder_lift_joint
      - my_ur_elbow_joint
      - my_ur_wrist_1_joint
      - my_ur_wrist_2_joint
      - my_ur_wrist_3_joint
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
    state_publish_rate: 100.0
    action_monitor_rate: 20.0
    allow_partial_joints_goal: false
    constraints:
      stopped_velocity_tolerance: 0.2
      goal_time: 0.0
      shoulder_pan_joint: { trajectory: 0.2, goal: 0.1 }
      shoulder_lift_joint: { trajectory: 0.2, goal: 0.1 }
      elbow_joint: { trajectory: 0.2, goal: 0.1 }
      wrist_1_joint: { trajectory: 0.2, goal: 0.1 }
      wrist_2_joint: { trajectory: 0.2, goal: 0.1 }
      wrist_3_joint: { trajectory: 0.2, goal: 0.1 }

scaled_joint_trajectory_controller:
  ros__parameters:
    joints:
      - my_ur_shoulder_pan_joint
      - my_ur_shoulder_lift_joint
      - my_ur_elbow_joint
      - my_ur_wrist_1_joint
      - my_ur_wrist_2_joint
      - my_ur_wrist_3_joint
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
    state_publish_rate: 100.0
    action_monitor_rate: 20.0
    allow_partial_joints_goal: false
    constraints:
      stopped_velocity_tolerance: 0.2
      goal_time: 0.0
      shoulder_pan_joint: { trajectory: 0.2, goal: 0.1 }
      shoulder_lift_joint: { trajectory: 0.2, goal: 0.1 }
      elbow_joint: { trajectory: 0.2, goal: 0.1 }
      wrist_1_joint: { trajectory: 0.2, goal: 0.1 }
      wrist_2_joint: { trajectory: 0.2, goal: 0.1 }
      wrist_3_joint: { trajectory: 0.2, goal: 0.1 }
    speed_scaling_interface_name: my_ur_speed_scaling/speed_scaling_factor

forward_velocity_controller:
  ros__parameters:
    joints:
      - my_ur_shoulder_pan_joint
      - my_ur_shoulder_lift_joint
      - my_ur_elbow_joint
      - my_ur_wrist_1_joint
      - my_ur_wrist_2_joint
      - my_ur_wrist_3_joint
    interface_name: velocity

forward_position_controller:
  ros__parameters:
    joints:
      - my_ur_shoulder_pan_joint
      - my_ur_shoulder_lift_joint
      - my_ur_elbow_joint
      - my_ur_wrist_1_joint
      - my_ur_wrist_2_joint
      - my_ur_wrist_3_joint
```
</details>

with versions
* `ur_robot_driver`: 2.2.8
* `ros2_control`: 2.28.0

TZECHIN6, 2023-06-30:
@fmauch Thanks for the information, it would be nice if you could notice us once it passes the integration test.

fmauch, 2023-06-30:
With Humble being synchronized and the packages available in mainline with the versions [noted above](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/677#issuecomment-1612567087) I'll go ahead and close this as completed. A full binary installation should work again now.

closed 2023-06-30 by fmauch