UniversalRobots/Universal_Robots_ROS2_Driver #686: Zeroing ftsensor seems to be broken

fmauch, 2023-05-24:
I just tested this on an e-Series ur3e and zeroing did not work (using the custom script interface). This should be investigated further.

firesurfer, 2023-05-25:
I found the issue. Seems to be due to the handback interface (which stops the program) 
When removing these lines: https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/643eada22fea3f31b5318160c57ad58bf4c90a94/ur_robot_driver/src/hardware_interface.cpp#L685

Zeroing works

firesurfer, 2023-05-25:
Ok I fixed the issue in the linked PR. 

This was a rather nasty bug as we didn't export enough command interfaces. I would suggest using sorted interfaces in the gpio controller in order to avoid such bugs in the future.

PR #693 (2023-06-01): added missing command interfaces into gpio controller
Fixes #686