UniversalRobots/Universal_Robots_ROS2_Driver #694: Velocity command interface

fafux, 2023-05-25:
Hi,
I am trying to control an UR robot both in simulation and with the real hardware with the command interface in velocity with ROS2 foxy. The values of the joint velocities states, in simulation, are changing but the robot is not moving. While if I try to move the robot using the position command interface I can do it.
Is it possible to move the robot only using the velocity command interface? If so, is there any example on how to do it?
Thanks.

fmauch, 2023-06-06:
For Foxy, velocity commands have not been implemented. We suggest to use ROS 2 humble, since Foxy is end of life by now.

For humble, a forwarding velocity controller is built in and simply has to be activated using the [`ros2 control switch_controllers` cli tool](https://control.ros.org/master/doc/ros2_control/ros2controlcli/doc/userdoc.html). To use it, there is a test motion script `ros2 launch ur_robot_driver test_forward_velocity_controller.launch.py` which runs [this script](https://github.com/ros-controls/ros2_controllers/blob/humble/ros2_controllers_test_nodes/ros2_controllers_test_nodes/publisher_forward_position_controller.py)

tldr:
While the robot is running, do
```bash 
ros2 control load_controller forward_velocity_controller
ros2 control set_controller_state forward_velocity_controller inactive
ros2 control switch_controllers --deactivate scaled_joint_trajectory_controller --activate forward_velocity_controller

# start motion
ros2 topic pub --once /forward_velocity_controller/commands std_msgs/msg/Float64MultiArray "{data:[0,0,0,0,0,0.1]}"

# stop motion
ros2 topic pub --once /forward_velocity_controller/commands std_msgs/msg/Float64MultiArray "{data:[0,0,0,0,0,0.0]}"
```

fmauch, 2023-07-03:
Confirmed that with fake hardware this does not work. With URSim / a real robot this works, however.

closed 2023-07-12 by fafux

XHT-debug, 2025-09-15:
> Hi,你好，
> 
> I am using ROS2 humble and I follow your instructions:我使用ROS2谦虚，我按照你的指示：
> 
> ```
> ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur10e robot_ip:=yyy.yyy.yyy.yyy initial_joint_controller:=scaled_joint_trajectory_controller use_fake_hardware:=true launch_rviz:=true
> ros2 control load_controller forward_velocity_controller
> ros2 control set_controller_state forward_velocity_controller inactive
> ros2 control switch_controllers --deactivate scaled_joint_trajectory_controller --activate forward_velocity_controller
> ros2 topic pub --once /forward_velocity_controller/commands std_msgs/msg/Float64MultiArray "{data:[0,0,0,0,0,0.1]}"
> ```
> 
> but the robot is not moving.但机器人不动

When the ur driver starts the servo node, it will load the ur_servo.yaml parameter file. The content of the command_out_topic in the parameter file needs to be modified to /forwards_velocity_controller/commands, otherwise the servo node will only process instructions for the topic /forwards_position_controller/commands