UniversalRobots/Universal_Robots_ROS2_Driver #697: Apply fix for trajectory execution race condition to ScaledJointTrajectoryController

schornakj, 2023-05-26:
The JointTrajectoryController in ros2_controllers had [a race condition](https://github.com/ros-controls/ros2_controllers/pull/565) in which trajectory execution could erroneously succeed while the robot was still in motion, which caused undesirable behavior when performing multiple motion plans and trajectories in series.

Since this repo's ScaledJointTrajectoryController is based on JointTrajectoryController, it probably also has the same problem.

To resolve this, the changes from https://github.com/ros-controls/ros2_controllers/pull/565 need to be applied to ScaledJointTrajectoryController.

panagelak, 2023-06-05:
@schornakj  @fmauch 

I create a pr for this for humble branch https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/710

fmauch, 2023-09-21:
Should be resolved by #810, #811, #813

closed 2023-09-21 by fmauch