UniversalRobots/Universal_Robots_ROS2_Driver #698: Usage of Driver, Description, Gazebo Simulation on ROS2

segapss, 2023-05-29:
Hey,
I have installed Universal_Robots_ROS2_Driver using binary builds. 

1. I checked that [ursim](https://hub.docker.com/r/universalrobots/ursim_e-series) works based on docker. But unable to run [Usage with official UR simulator](https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/usage.html#usage-with-official-ur-simulator) on ros2 commands. Could I get explanation what's wrong on this?
2. For full simulation on Gazebo(I don't have real robot), should I install [Description](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description) [Gazebo Simulation](https://github.com/UniversalRobots/Universal_Robots_ROS2_Gazebo_Simulation) also or just Driver is enough?
Thanks in advance.

segapss, 2023-05-31:
@firesurfer Thanks. Are you referring to [this](https://github.com/ros-controls/ros2_control)? Then apart from Gazebo, do you think it is wise to rely on Universal_Robots_ROS2_Driver and URSIM for it? I think [Articulated Robotics](https://www.youtube.com/watch?v=J02jEKawE5U) is giving a good solution. Then you sure that ROS2 control can be operated without real robot?

firesurfer, 2023-06-01:
@segapss 

So either way you are doing this. You will / want to always use ros2control somewhere in between. If you specify `use_fake_hardware=true`  ros2control will initiate a simple mockinterface (which is enough if you want to test position controlled commands).

We tried some time ago to setup everything with gazebo in order to test force controlled algorithms but had a hard time to get everything running stable. We decided in the end not to use it. 

Regarding point 3. You can use the robotiq grippers fairly easy on the URs.

segapss, 2023-06-01:
@firesurfer Thanks. Then I will need to rely on ros2control more.

sillkjc, 2023-06-05:
My team has written a Gazebo plugin that watches ./joint_states and sets the gazebo robot joint angles accordingly. We run ur_sim in the loop with use_fake_hardware false, so the simulated code does not know it is simulated.

We will be publishing this on our github, but if there is interest to accelerate this let me know.

PietroDrd, 2023-11-02:
@sillkjc Hi!
Is there any update about your Gazebo plugin? Thanks!

brta-jc, 2023-11-10:
> @sillkjc Hi! Is there any update about your Gazebo plugin? Thanks!
Hi PietroDrd, I've pushed it here, there is also a Noetic branch.
https://github.com/Boeing/gazebo_set_joint_positions_plugin/tree/humble

We are currently in queue to get ros-humble-boeing-* packages added to apt so they can easily be pulled as debs.  As we get the flow down we'll be releasing more packages.

VinDp, 2024-07-03:
Since the issue is pretty old and everything seems to be solved, I will close it.
It could be helpful to add that Gazebo simulation depends only on the [description](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description) package and the [moveit_config](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/main/ur_moveit_config) (part of the driver repo). Also we now have an [updated documentation](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_simulation_gz/ur_simulation_gz/doc/index.html) for GZ for Rolling and Jazzy.

closed 2024-07-03 by VinDp