UniversalRobots/Universal_Robots_ROS2_Driver #711: How to use the calibration file extracted by ur_calibration

yinguoxiangyi, 2023-06-08:
I am using galactic branch.
Is it the right way of using calibration file by specify the arg `kinematic_config` with `ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e use_fake_hardware:=false launch_rviz:=false initial_joint_controller:=joint_trajectory_controller robot_ip:=<robot_ip> kinematics_config:=${HOME}/my_robot_calibration.yaml`

destogl, 2023-06-11:
This would be the official way. Or even better, you create your description package and recreate the structure of description package and add parameters for your scenario

VinDp, 2024-07-03:
Since the issue is pretty old, I will close it. 
Also Galactic is EOL, but for Humble and on [this issue](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/918) could be useful for the question.

closed 2024-07-03 by VinDp