UniversalRobots/Universal_Robots_ROS2_Driver #712: Launch file "ur_control.launch.py" doesn't work after update

Luc90, 2023-06-08:
Hello,

My installation is:
- Ubuntu 22.04
- ROS Humble
- UR5e

I installed Universal_Robots_ROS2_Driver three months ago and it worked perfectly (Binary installation). 
Earlier this week, I did an installation on another PC. Unfortunately, after running ur_sim, when I run the following command for the driver:

- ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.56.101 launch_rviz:=false

I get the following error:

`[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [271144]
[INFO] [controller_stopper_node-3]: process started with pid [271146]
[INFO] [ur_ros2_control_node-1]: process started with pid [271142]
[INFO] [robot_state_publisher-4]: process started with pid [271148]
[INFO] [spawner-5]: process started with pid [271150]
[INFO] [spawner-6]: process started with pid [271152]
[INFO] [spawner-7]: process started with pid [271154]
[INFO] [spawner-8]: process started with pid [271156]
[INFO] [spawner-9]: process started with pid [271167]
[INFO] [spawner-10]: process started with pid [271197]
[robot_state_publisher-4] [INFO] [1686149494.412737731] [robot_state_publisher]: got segment base
[robot_state_publisher-4] [INFO] [1686149494.412935030] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1686149494.412959424] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-4] [INFO] [1686149494.412972711] [robot_state_publisher]: got segment flange
[robot_state_publisher-4] [INFO] [1686149494.412983927] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-4] [INFO] [1686149494.412995106] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-4] [INFO] [1686149494.413006491] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-4] [INFO] [1686149494.413018046] [robot_state_publisher]: got segment tool0
[robot_state_publisher-4] [INFO] [1686149494.413029807] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-4] [INFO] [1686149494.413040482] [robot_state_publisher]: got segment world
[robot_state_publisher-4] [INFO] [1686149494.413050835] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-4] [INFO] [1686149494.413062439] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-4] [INFO] [1686149494.413073940] [robot_state_publisher]: got segment wrist_3_link
[ur_ros2_control_node-1] [INFO] [1686149494.414389619] [resource_manager]: Loading hardware 'ur5e' 
[ur_ros2_control_node-1] [INFO] [1686149494.417200661] [resource_manager]: Initialize hardware 'ur5e' 
[ur_ros2_control_node-1] [INFO] [1686149494.417313177] [resource_manager]: Successful initialization of hardware 'ur5e'
[dashboard_client-2] [INFO] [1686149494.418829260] [UR_Client_Library]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[ur_ros2_control_node-1] terminate called after throwing an instance of 'std::runtime_error'
[ur_ros2_control_node-1]   what():  Wrong state or command interface configuration.
[ur_ros2_control_node-1] missing state interfaces:
[ur_ros2_control_node-1] 
[ur_ros2_control_node-1] missing command interfaces:
[ur_ros2_control_node-1] ' gpio/tool_voltage_cmd '    ' zero_ftsensor/zero_ftsensor_cmd '    ' zero_ftsensor/zero_ftsensor_async_success '    
[controller_stopper_node-3] [INFO] [1686149494.423530573] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[ERROR] [ur_ros2_control_node-1]: process has died [pid 271142, exit code -6, cmd '/opt/ros/humble/lib/ur_robot_driver/ur_ros2_control_node --ros-args --params-file /tmp/launch_params_yrk5rp7o --params-file /opt/ros/humble/share/ur_robot_driver/config/ur5e_update_rate.yaml --params-file /opt/ros/humble/share/ur_robot_driver/config/ur_controllers.yaml'].
[spawner-6] [INFO] [1686149497.183844053] [spawner_io_and_status_controller]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1686149497.195356052] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-9] [INFO] [1686149497.231017369] [spawner_forward_position_controller]: Waiting for '/controller_manager' node to exist
[spawner-7] [INFO] [1686149497.246468733] [spawner_speed_scaling_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-8] [INFO] [1686149497.265553259] [spawner_force_torque_sensor_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-10] [INFO] [1686149497.294490525] [spawner_scaled_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1686149499.205192816] [spawner_io_and_status_controller]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1686149499.216240649] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-9] [INFO] [1686149499.253499594] [spawner_forward_position_controller]: Waiting for '/controller_manager' node to exist
[spawner-7] [INFO] [1686149499.267065358] [spawner_speed_scaling_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-8] [INFO] [1686149499.287230726] [spawner_force_torque_sensor_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-10] [INFO] [1686149499.316739022] [spawner_scaled_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1686149501.225837894] [spawner_io_and_status_controller]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1686149501.237086897] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-9] [INFO] [1686149501.275209504] [spawner_forward_position_controller]: Waiting for '/controller_manager' node to exist
[spawner-7] [INFO] [1686149501.290882269] [spawner_speed_scaling_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-8] [INFO] [1686149501.308571132] [spawner_force_torque_sensor_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-10] [INFO] [1686149501.339052663] [spawner_scaled_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1686149503.248131680] [spawner_io_and_status_controller]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1686149503.258707353] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-9] [INFO] [1686149503.295886932] [spawner_forward_position_controller]: Waiting for '/controller_manager' node to exist
[spawner-7] [INFO] [1686149503.313064618] [spawner_speed_scaling_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-8] [INFO] [1686149503.329784375] [spawner_force_torque_sensor_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-10] [INFO] [1686149503.359706548] [spawner_scaled_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [ERROR] [1686149505.269203092] [spawner_io_and_status_controller]: Controller manager not available
[spawner-5] [ERROR] [1686149505.280398788] [spawner_joint_state_broadcaster]: Controller manager not available
[spawner-9] [ERROR] [1686149505.315949768] [spawner_forward_position_controller]: Controller manager not available
[spawner-7] [ERROR] [1686149505.334229573] [spawner_speed_scaling_state_broadcaster]: Controller manager not available
[spawner-8] [ERROR] [1686149505.350795915] [spawner_force_torque_sensor_broadcaster]: Controller manager not available
[spawner-10] [ERROR] [1686149505.383068781] [spawner_scaled_joint_trajectory_controller]: Controller manager not available
[ERROR] [spawner-6]: process has died [pid 271152, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner io_and_status_controller -c /controller_manager --ros-args'].
[ERROR] [spawner-5]: process has died [pid 271150, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_state_broadcaster --controller-manager /controller_manager --ros-args'].
[ERROR] [spawner-7]: process has died [pid 271154, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner speed_scaling_state_broadcaster --controller-manager /controller_manager --ros-args'].
[ERROR] [spawner-9]: process has died [pid 271167, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner forward_position_controller -c /controller_manager --inactive --ros-args'].
[... cut ...] os-args'].
[ERROR] [spawner-10]: process has died [pid 271197, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner scaled_joint_trajectory_controller -c /controller_manager --ros-args'].
`

Looking on the internet, I saw that others had had the same problem:[Universal Robot forum:](https://forum.universal-robots.com/t/ros2-humble-ur5-connection-throws-an-error/28268)
I tried to build from source as recommended, but I have the same error.

Do you know what the problem is ?
Thank you.

destogl, 2023-06-08:
Are you using correct version of the source branches? Humble branch for humble?

We need more info on your setup to have better idea where the issue might be.

Luc90, 2023-06-08:
Yes i use the correct branch.

my setup is as follows:

- Ubuntu 22.04 in WMware Workstation 16 with ROS Humble (binary installation).

I have installed Universal_Robots_ROS2_Driver with binary installation. I also tried to Build from source with humble branch.
I have setup URSim with Docker and I can launch it and it works.

In both cases (binary installation or build from source), I get the following error when I run the command "ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.56.101 launch_rviz:=true" :

> luc@luc-virtual-machine:~$ ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.56.101 launch_rviz:=true
> [INFO] [launch]: All log files can be found below /home/luc/.ros/log/2023-06-08-15-55-53-640366-luc-virtual-machine-4160
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [dashboard_client-2]: process started with pid [4166]
> [INFO] [controller_stopper_node-3]: process started with pid [4168]
> [INFO] [ur_ros2_control_node-1]: process started with pid [4164]
> [INFO] [robot_state_publisher-4]: process started with pid [4170]
> [INFO] [rviz2-5]: process started with pid [4172]
> [INFO] [spawner-6]: process started with pid [4174]
> [INFO] [spawner-7]: process started with pid [4176]
> [INFO] [spawner-8]: process started with pid [4178]
> [INFO] [spawner-9]: process started with pid [4180]
> [INFO] [spawner-10]: process started with pid [4182]
> [INFO] [spawner-11]: process started with pid [4185]
> [rviz2-5] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
> [dashboard_client-2] [INFO] [1686232554.869734818] [UR_Client_Library]: Connected: Universal Robots Dashboard Server
> [dashboard_client-2] 
> [controller_stopper_node-3] [INFO] [1686232554.872555851] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
> [ur_ros2_control_node-1] [INFO] [1686232554.888273912] [resource_manager]: Loading hardware 'ur5e' 
> [ur_ros2_control_node-1] [INFO] [1686232554.913835261] [resource_manager]: Initialize hardware 'ur5e' 
> [ur_ros2_control_node-1] [INFO] [1686232554.914075811] [resource_manager]: Successful initialization of hardware 'ur5e'
> [ur_ros2_control_node-1] terminate called after throwing an instance of 'std::runtime_error'
> [ur_ros2_control_node-1]   what():  Wrong state or command interface configuration.
> [ur_ros2_control_node-1] missing state interfaces:
> [ur_ros2_control_node-1] 
> [ur_ros2_control_node-1] missing command interfaces:
> [ur_ros2_control_node-1] ' gpio/tool_voltage_cmd '	' zero_ftsensor/zero_ftsensor_cmd '	' zero_ftsensor/zero_ftsensor_async_success '	
> [robot_state_publisher-4] [INFO] [1686232555.097928890] [robot_state_publisher]: got segment base
> [robot_state_publisher-4] [INFO] [1686232555.098935611] [robot_state_publisher]: got segment base_link
> [robot_state_publisher-4] [INFO] [1686232555.099525110] [robot_state_publisher]: got segment base_link_inertia
> [robot_state_publisher-4] [INFO] [1686232555.100001379] [robot_state_publisher]: got segment flange
> [robot_state_publisher-4] [INFO] [1686232555.100761466] [robot_state_publisher]: got segment forearm_link
> [robot_state_publisher-4] [INFO] [1686232555.101548451] [robot_state_publisher]: got segment ft_frame
> [robot_state_publisher-4] [INFO] [1686232555.103249330] [robot_state_publisher]: got segment shoulder_link
> [robot_state_publisher-4] [INFO] [1686232555.104005802] [robot_state_publisher]: got segment tool0
> [robot_state_publisher-4] [INFO] [1686232555.104364192] [robot_state_publisher]: got segment upper_arm_link
> [robot_state_publisher-4] [INFO] [1686232555.104673685] [robot_state_publisher]: got segment world
> [robot_state_publisher-4] [INFO] [1686232555.108953925] [robot_state_publisher]: got segment wrist_1_link
> [robot_state_publisher-4] [INFO] [1686232555.109537313] [robot_state_publisher]: got segment wrist_2_link
> [robot_state_publisher-4] [INFO] [1686232555.109549185] [robot_state_publisher]: got segment wrist_3_link
> [rviz2-5] [INFO] [1686232555.866927328] [rviz2]: Stereo is NOT SUPPORTED
> [rviz2-5] [INFO] [1686232555.869443144] [rviz2]: OpenGl version: 4.1 (GLSL 4.1)
> [rviz2-5] [INFO] [1686232556.032776636] [rviz2]: Stereo is NOT SUPPORTED
> [ERROR] [ur_ros2_control_node-1]: process has died [pid 4164, exit code -6, cmd '/opt/ros/humble/lib/ur_robot_driver/ur_ros2_control_node --ros-args --params-file /tmp/launch_params_ukaly6ns --params-file /opt/ros/humble/share/ur_robot_driver/config/ur5e_update_rate.yaml --params-file /opt/ros/humble/share/ur_robot_driver/config/ur_controllers.yaml'].
> ^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
> [dashboard_client-2] [INFO] [1686232556.574169573] [rclcpp]: signal_handler(signum=2)
> [robot_state_publisher-4] [INFO] [1686232556.574116679] [rclcpp]: signal_handler(signum=2)
> [controller_stopper_node-3] [INFO] [1686232556.575697036] [rclcpp]: signal_handler(signum=2)
> [controller_stopper_node-3] [INFO] [1686232556.578755051] [Controller stopper]: Service available
> [controller_stopper_node-3] [INFO] [1686232556.579320660] [Controller stopper]: Waiting for list controllers service to come up on controller_manager/list_controllers
> [controller_stopper_node-3] [INFO] [1686232556.579343578] [Controller stopper]: Service available
> [controller_stopper_node-3] 
[... cut ...]
> [ur_ros2_control_node-1] [INFO] [1686233129.955884918] [URPositionHardwareInterface]: Initializing driver...
> [ur_ros2_control_node-1] [WARN] [1686233129.966285196] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
> [ur_ros2_control_node-1] [INFO] [1686233129.966468618] [UR_Client_Library]: Negotiated RTDE protocol version to 2.
> [ur_ros2_control_node-1] [INFO] [1686233129.966687546] [UR_Client_Library]: Setting up RTDE communication with frequency 500.000000

gr3atchoi, 2023-06-09:
I've done all the procedures in document, but experiencing the same situation now. 

Ubuntu 22.04 
ROS Humble
UR3e hardware. 

> ➜  ros_ur_driver ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e robot_ip:=192.168.0.51 launch_rviz:=true     
[INFO] [launch]: All log files can be found below /home/gr3atchoi/.ros/log/2023-06-09-09-52-04-675137-gr3atchoi-ws-55577
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [55586]
[INFO] [controller_stopper_node-3]: process started with pid [55588]
[INFO] [ur_ros2_control_node-1]: process started with pid [55584]
[INFO] [robot_state_publisher-4]: process started with pid [55590]
[INFO] [rviz2-5]: process started with pid [55592]
[INFO] [spawner-6]: process started with pid [55594]
[INFO] [spawner-7]: process started with pid [55596]
[INFO] [spawner-8]: process started with pid [55598]
[INFO] [spawner-9]: process started with pid [55600]
[INFO] [spawner-10]: process started with pid [55602]
[INFO] [spawner-11]: process started with pid [55604]
[controller_stopper_node-3] [INFO] [1686271925.121223953] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[dashboard_client-2] [INFO] [1686271925.121391542] [UR_Client_Library]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[ur_ros2_control_node-1] terminate called after throwing an instance of 'std::runtime_error'
[ur_ros2_control_node-1]   what():  text not specified in the tf_prefix tag
[robot_state_publisher-4] [INFO] [1686271925.123852624] [robot_state_publisher]: got segment base
[robot_state_publisher-4] [INFO] [1686271925.123914582] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1686271925.123919732] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-4] [INFO] [1686271925.123923068] [robot_state_publisher]: got segment flange
[robot_state_publisher-4] [INFO] [1686271925.123925903] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-4] [INFO] [1686271925.123928619] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-4] [INFO] [1686271925.123931314] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-4] [INFO] [1686271925.123933979] [robot_state_publisher]: got segment tool0
[robot_state_publisher-4] [INFO] [1686271925.123936684] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-4] [INFO] [1686271925.123939399] [robot_state_publisher]: got segment world
[robot_state_publisher-4] [INFO] [1686271925.123941974] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-4] [INFO] [1686271925.123944559] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-4] [INFO] [1686271925.123947164] [robot_state_publisher]: got segment wrist_3_link
[ERROR] [ur_ros2_control_node-1]: process has died [pid 55584, exit code -6, cmd '/home/gr3atchoi/Workspace/UniversalRobots/ros_ur_driver/install/ur_robot_driver/lib/ur_robot_driver/ur_ros2_control_node --ros-args --params-file /tmp/launch_params_djuv8jfl --params-file /home/gr3atchoi/Workspace/UniversalRobots/ros_ur_driver/install/ur_robot_driver/share/ur_robot_driver/config/ur3e_update_rate.yaml --params-file /home/gr3atchoi/Workspace/UniversalRobots/ros_ur_driver/install/ur_robot_driver/share/ur_robot_driver/config/ur_controllers.yaml'].
[rviz2-5] [INFO] [1686271925.581529485] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] [INFO] [1686271925.581645556] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-5] [INFO] [1686271925.609077191] [rviz2]: Stereo is NOT SUPPORTED
[spawner-8] [INFO] [1686271927.297561769] [spawner_io_and_status_controller]: Waiting for '/controller_manager' node to exist
[spawner-7] [INFO] [1686271927.299515559] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-11] [INFO] [1686271927.299514918] [spawner_forward_position_controller]: Waiting for '/controller_manager' node to exist
[spawner-10] [INFO] [1686271927.300432851] [spawner_force_torque_sensor_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-9] [INFO] [1686271927.300847087] [spawner_speed_scaling_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1686271927.300847117] [spawner_scaled_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
[spawner-8] [INFO] [1686271929.308230917] [spawner_io_and_status_controller]: Waiting for '/controller_manager' node to exist
[spawner-11] [INFO] [1686271929.309923241] [spawner_forward_position_controller]: Waiting for '/controller_manager' node to exist
[spawner-7] [INFO] [1686271929.309920756] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-10] [INFO] [1686271929.310042658] [spawner_force_torque_sensor_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-9] [INFO] [1686271929.310572534] [spawner_speed_scaling_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1686271929.310674788] [spawner_scaled_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
[spawner-8] [INFO] [1686271931.317408633] [spawner_io_and_status_controller]: Waiting for '/controller_manager' node to exist
[spawner-10] [INFO] [1686271931.318583484] [spawner_force_torque_sensor_broadcaster]: Waiting for '/controller_manager' node to exist
[... cut ...]
[ERROR] [spawner-7]: process has died [pid 55596, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_state_broadcaster --controller-manager /controller_manager --controller-manager-timeout 10 --ros-args'].
[ERROR] [spawner-6]: process has died [pid 55594, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner scaled_joint_trajectory_controller -c /controller_manager --controller-manager-timeout 10 --ros-args'].
[ERROR] [spawner-8]: process has died [pid 55598, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner io_and_status_controller --controller-manager /controller_manager --controller-manager-timeout 10 --ros-args'].

destogl, 2023-06-09:
This is known issue. @fmauch provided work around in another issue (#677). There is one regression in current Humble release. I this this is the issue. If so please close this one, so we have only one opened.

closed 2023-06-12 by Luc90