UniversalRobots/Universal_Robots_ROS2_Driver #714: scaled_joint_trajectory_controller delayed execution of trajectory

SuperDiodo, 2023-06-09:
**Configuration**:
- Ubuntu 22.04 (RT kernel)
- ROS2 Humble
- ur_driver cloned from github Humble branch
- externalcontrol-1.0.5.urcap

___
Hi,
I'm trying to plan trajectories with Moveit and send them to a real UR10e through a C++ script. The problem is that the execution of the trajectories is randomly delayed (sometimes they are executed instantly and sometimes it can takes up to minutes). However, If I try to plan a trajectory from Rviz and click the "execute" button the trajectory is always instantly executed.

Adding some prints in `scaled_joint_trajectory_controller.cpp` I discovered that the internal timestamp of the SJTC (`traj_time`) is lower than the timestamp of the ros node when the trajectory has been sent (let's call it `ros_time`). The controller isn't able to start executing the trajectory until `traj_time` == `ros_time`, probably because of this part of the script:

```
const bool valid_point =
        (*traj_point_active_ptr_)
            ->sample(traj_time,
                     joint_trajectory_controller::interpolation_methods::InterpolationMethod::VARIABLE_DEGREE_SPLINE,
                     state_desired, start_segment_itr, end_segment_itr);
```

I have noticed that:
- After a trajectory execution (called from the C++ script or Rviz) the driver is synchronized and the following trajectories will be executed instantly
- Each time I restart the external control urcap the driver desynchronizes

Thanks in advance for any suggestion :smile:

destogl, 2023-06-11:
@SuperDiodo to clarify the issue. Does this mean then Scaled JTC also prolongs the first time-stamp when trajectory should be started?

This might be the issue (without detailed) code-check. And this should be quite well solvable by not adjusting trajectory start time (which controller should do as I remember).

SuperDiodo, 2023-06-12:
The SJTC starts the execution when `traj_time` reaches the ros time-stamp related to the trajectory reception. The problem is that `traj_time` in some cases is lower than the expected timestamp and the execution is delayed. 

To give a practical example, I report below the results of a small test (I'm sorry for the amount of code lines): I planned a Cartesian Path and sent it to the robot 3 times with different speed scaling (98%, 99%, 100%). 
Before sending a trajectory to the robot I restarted the urcap program on the teach pendant + I added a line to print `traj_time` until the trajectory execution starts (ideally, the execution should starts instantly and no prints should appear).

---
Speed scaling 98%:
```
[ur_ros2_control_node-2] [INFO] [1686578418.807146267] [scaled_joint_trajectory_controller]: Received new action goal
[ur_ros2_control_node-2] [INFO] [1686578418.807181674] [scaled_joint_trajectory_controller]: Accepted new action goal
[ur_ros2_control_node-2] [INFO] [1686578418.810122798] [scaled_joint_trajectory_controller]: traj_time = 1686578418646041936
[ur_ros2_control_node-2] [INFO] [1686578418.812121022] [scaled_joint_trajectory_controller]: traj_time = 1686578418648005583
[ur_ros2_control_node-2] [INFO] [1686578418.814171895] [scaled_joint_trajectory_controller]: traj_time = 1686578418650023235
[ur_ros2_control_node-2] [INFO] [1686578418.816141985] [scaled_joint_trajectory_controller]: traj_time = 1686578418651962600
[ur_ros2_control_node-2] [INFO] [1686578418.818163623] [scaled_joint_trajectory_controller]: traj_time = 1686578418653933556
[ur_ros2_control_node-2] [INFO] [1686578418.820239032] [scaled_joint_trajectory_controller]: traj_time = 1686578418655976889
[ur_ros2_control_node-2] [INFO] [1686578418.822138890] [scaled_joint_trajectory_controller]: traj_time = 1686578418657825247
[ur_ros2_control_node-2] [INFO] [1686578418.824110683] [scaled_joint_trajectory_controller]: traj_time = 1686578418659766984
[ur_ros2_control_node-2] [INFO] [1686578418.826134004] [scaled_joint_trajectory_controller]: traj_time = 1686578418661731798
[ur_ros2_control_node-2] [INFO] [1686578418.828107170] [scaled_joint_trajectory_controller]: traj_time = 1686578418663685534
[ur_ros2_control_node-2] [INFO] [1686578418.830102348] [scaled_joint_trajectory_controller]: traj_time = 1686578418665641476
[ur_ros2_control_node-2] [INFO] [1686578418.832176966] [scaled_joint_trajectory_controller]: traj_time = 1686578418667663964
[ur_ros2_control_node-2] [INFO] [1686578418.834083226] [scaled_joint_trajectory_controller]: traj_time = 1686578418669543733
[ur_ros2_control_node-2] [INFO] [1686578418.836136864] [scaled_joint_trajectory_controller]: traj_time = 1686578418671555363
[ur_ros2_control_node-2] [INFO] [1686578418.838124537] [scaled_joint_trajectory_controller]: traj_time = 1686578418673496705
[ur_ros2_control_node-2] [INFO] [1686578418.840112021] [scaled_joint_trajectory_controller]: traj_time = 1686578418675436075
[ur_ros2_control_node-2] [INFO] [1686578418.842089154] [scaled_joint_trajectory_controller]: traj_time = 1686578418677389367
[ur_ros2_control_node-2] [INFO] [1686578418.844096565] [scaled_joint_trajectory_controller]: traj_time = 1686578418679352101
[ur_ros2_control_node-2] [INFO] [1686578418.846083267] [scaled_joint_trajectory_controller]: traj_time = 1686578418681302868
[ur_ros2_control_node-2] [INFO] [1686578418.848079206] [scaled_joint_trajectory_controller]: traj_time = 1686578418683258840
[ur_ros2_control_node-2] [INFO] [1686578418.850075957] [scaled_joint_trajectory_controller]: traj_time = 1686578418685214202
[ur_ros2_control_node-2] [INFO] [1686578418.852122471] [scaled_joint_trajectory_controller]: traj_time = 1686578418687221552
[ur_ros2_control_node-2] [INFO] [1686578418.854118871] [scaled_joint_trajectory_controller]: traj_time = 1686578418689177647
[ur_ros2_control_node-2] [INFO] [1686578418.856119439] [scaled_joint_trajectory_controller]: traj_time = 1686578418691140208
[ur_ros2_control_node-2] [INFO] [1686578418.858130818] [scaled_joint_trajectory_controller]: traj_time = 1686578418693096637
[ur_ros2_control_node-2] [INFO] [1686578418.860124152] [scaled_joint_trajectory_controller]: traj_time = 1686578418695056844
[ur_ros2_control_node-2] [INFO] [1686578418.862105834] [scaled_joint_trajectory_controller]: traj_time = 1686578418697002992
[ur_ros2_control_node-2] [INFO] [1686578418.864102034] [scaled_joint_trajectory_controller]: traj_time = 1686578418698960450
[ur_ros2_control_node-2] [INFO] [1686578418.866106259] [scaled_joint_trajectory_controller]: traj_time = 1686578418700923478
[ur_ros2_control_node-2] [INFO] [1686578418.868104051] [scaled_joint_trajectory_controller]: traj_time = 1686578418702881852
[ur_ros2_control_node-2] [INFO] [1686578418.870091805] [scaled_joint_trajectory_controller]: traj_time = 1686578418704832722
[ur_ros2_control_node-2] [INFO] [1686578418.872102923] [scaled_joint_trajectory_controller]: traj_time = 1686578418706795587
[ur_ros2_control_node-2] [INFO] [1686578418.874084906] [scaled_joint_trajectory_controller]: traj_time = 1686578418708746503
[ur_ros2_control_node-2] [INFO] [1686578418.876091284] [scaled_joint_trajectory_controller]: traj_time = 1686578418710708203
[ur_ros2_control_node-2] [INFO] [1686578418.878084829] [scaled_joint_trajectory_controller]: traj_time = 1686578418712662688
[ur_ros2_control_node-2] [INFO] [1686578418.880081299] [scaled_joint_trajectory_controller]: traj_time = 1686578418714619157
[... cut ...]
[ur_ros2_control_node-2] [INFO] [1686578317.781871010] [scaled_joint_trajectory_controller]: Received new action goal
[ur_ros2_control_node-2] [INFO] [1686578317.781909442] [scaled_joint_trajectory_controller]: Accepted new action goal
```
---

As it is possible to notice, with a speed scaling < 100%, `traj_time` is lower than the ros timestamp when the goal is accepted and the execution is delayed. Conversely, when the speed scaling is 100%, no prints occurs.

For now I "solved" this problem forcing `scaling_factor_` to be equal to 100% until the trajectory execution is started.

SuperDiodo, 2023-06-28:
Hi,
I've cloned the driver package from the repo and modified the `scaled_joint_trajectory_controller` cpp/hpp code (attached txt files).

[scaled_joint_trajectory_controller.txt](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/files/11892400/scaled_joint_trajectory_controller.txt)

[scaled_joint_trajectory_controller.hpp.txt](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/files/11892459/scaled_joint_trajectory_controller.hpp.txt)

I don't know if it's the best solution, but it's working for me.
Basically:  I've added a boolean variable `executing_trajectory_` that tracks the execution of a trajectory. Until the trajectory has not started the variable is `false` and the speed scaling factor is by default 1. If the variable is `true` the speed scaling becomes equal to the value set in the teach pendant.

Let me know If it works for you.

SuperDiodo, 2023-06-28:
Hi,
I've cloned the driver package from the repo and modified the scaled_joint_trajectory_controller cpp/hpp code (attached txt files).

[scaled_joint_trajectory_controller.hpp.txt](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/files/11892508/scaled_joint_trajectory_controller.hpp.txt)

[scaled_joint_trajectory_controller.cpp.txt](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/files/11892510/scaled_joint_trajectory_controller.cpp.txt)

I don't know if it's the best solution, but it's working for me.
Basically: I've added a boolean variable executing_trajectory_ that tracks the execution of a trajectory. Until the trajectory has not started the variable is false and the speed scaling factor is by default 1. If the variable is true the speed scaling becomes equal to the value set in the teach pendant.

Let me know If it works for you.

fmauch, 2023-06-28:
@SuperDiodo tank you for putting that work into this. I'll try looking into the scaled JTC the next couple of days. Please keep this open!

JaisonJose241, 2023-08-23:
@SuperDiodo it worked like charm !! thx

urfeex, 2025-02-07:
Tried reproducing the issue but it seems to no longer exist. I know that we have been touching that part of the code in the meantime. Closing.

closed 2025-02-07 by urfeex