UniversalRobots/Universal_Robots_ROS2_Driver #716: Warning: Invalid frame ID "wrist_1_link" passed to canTransform argument source_frame - frame does not exist

AlbinEV, 2023-06-11:
Hi all, I am new to ros2 (also to ros). I am trying to use ROS2 Humbel with UR, specifically the e-series. When I try to use the following commands: 
`ros2 run ur_robot_driver start_ursim.sh -m ur5e`
and
`ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.56.101 launch_rviz:=true`
the result is the following image

[image]

in the terminal appears a series of errors like this:
```
[rviz2-5]          at line 93 in ./src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "wrist_1_link" passed to canTransform argument source_frame - frame does not exist

```

destogl, 2023-06-11:
Please check if you re not having issue described in #677.

What you see is "the last" consequence of that issue

AlbinEV, 2023-06-11:
Yes, it looks the same. Thank you for the reference. Unfortunately, nothing works...

AlbinEV, 2023-06-12:
Ok, following @WaReStu 's instructions in #677 now this problem seems to be solved but I have a new one, or maybe the same one but with different consequences. 

drim@drim:~$ ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.56.101 launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/drim/.ros/log/2023-06-12-11-11-48-835612-drim-8365
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [8372]
[INFO] [controller_stopper_node-3]: process started with pid [8374]
[INFO] [ur_ros2_control_node-1]: process started with pid [8370]
[INFO] [robot_state_publisher-4]: process started with pid [8376]
[INFO] [rviz2-5]: process started with pid [8378]
[INFO] [spawner-6]: process started with pid [8381]
[INFO] [spawner-7]: process started with pid [8391]
[INFO] [spawner-8]: process started with pid [8402]
[INFO] [spawner-9]: process started with pid [8409]
[INFO] [spawner-10]: process started with pid [8421]
[INFO] [spawner-11]: process started with pid [8425]
[dashboard_client-2] [INFO] [1686561110.424688994] [UR_Client_Library]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[controller_stopper_node-3] [INFO] [1686561110.553765725] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[robot_state_publisher-4] [INFO] [1686561110.628654153] [robot_state_publisher]: got segment base
[robot_state_publisher-4] [INFO] [1686561110.645818192] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1686561110.647312789] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-4] [INFO] [1686561110.648588860] [robot_state_publisher]: got segment flange
[robot_state_publisher-4] [INFO] [1686561110.649903867] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-4] [INFO] [1686561110.651152881] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-4] [INFO] [1686561110.652345588] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-4] [INFO] [1686561110.653591365] [robot_state_publisher]: got segment tool0
[robot_state_publisher-4] [INFO] [1686561110.655158652] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-4] [INFO] [1686561110.655195775] [robot_state_publisher]: got segment world
[robot_state_publisher-4] [INFO] [1686561110.655215304] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-4] [INFO] [1686561110.655234864] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-4] [INFO] [1686561110.655253268] [robot_state_publisher]: got segment wrist_3_link
[ur_ros2_control_node-1] text not specified in the tf_prefix tag
[ur_ros2_control_node-1] [INFO] [1686561110.835351850] [resource_manager]: Loading hardware 'ur5e' 
[ur_ros2_control_node-1] [INFO] [1686561110.848527356] [resource_manager]: Initialize hardware 'ur5e' 
[ur_ros2_control_node-1] [INFO] [1686561110.859946583] [resource_manager]: Successful initialization of hardware 'ur5e'
[ur_ros2_control_node-1] [INFO] [1686561110.861669917] [resource_manager]: 'configure' hardware 'ur5e' 
[ur_ros2_control_node-1] [INFO] [1686561110.863458720] [resource_manager]: Successful 'configure' of hardware 'ur5e'
[ur_ros2_control_node-1] [INFO] [1686561110.864516280] [resource_manager]: 'activate' hardware 'ur5e' 
[ur_ros2_control_node-1] [INFO] [1686561110.865413143] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1686561110.866685248] [URPositionHardwareInterface]: Initializing driver...
[ur_ros2_control_node-1] [WARN] [1686561110.876178475] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[ur_ros2_control_node-1] [INFO] [1686561111.224093554] [UR_Client_Library]: Negotiated RTDE protocol version to 2.
[ur_ros2_control_node-1] [INFO] [1686561111.225093493] [UR_Client_Library]: Setting up RTDE communication with frequency 500.000000
[ur_ros2_control_node-1] [INFO] [1686561112.240543237] [URPositionHardwareInterface]: Calibration checksum: 'calib_12788084448423163542'.
[ur_ros2_control_node-1] [WARN] [1686561112.259829065] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[rviz2-5] [INFO] [1686561113.139528187] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] [INFO] [1686561113.139760670] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-5] [INFO] [1686561113.186139821] [rviz2]: Stereo is NOT SUPPORTED
[ur_ros2_control_node-1] [INFO] [1686561113.294633107] [URPositionHardwareInterface]: Calibration checked successfully.
[ur_ros2_control_node-1] [INFO] [1686561113.294724236] [URPositionHardwareInterface]: System successfully started!
[ur_ros2_control_node-1] [INFO] [1686561113.294746972] [resource_manager]: Successful 'activate' of hardware 'ur5e'
[controller_stopper_node-3] [INFO] [1686561113.327927125] [Controller stopper]: Service available
[controller_stopper_node-3] [INFO] [1686561113.327984551] [Controller stopper]: Waiting for list controllers service to come up on controller_manager/list_controllers
[controller_stopper_node-3] [INFO] [1686561113.328017969] [Controller stopper]: Service available
[ur_ros2_control_node-1] [WARN] [1686561113.355031596] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[... cut ...]
[ERROR] [ur_ros2_control_node-1]: process has died [pid 8370, exit code -11, cmd '/home/drim/workspace/ros_ur_driver/install/ur_robot_driver/lib/ur_robot_driver/ur_ros2_control_node --ros-args --params-file /tmp/launch_params_jm60a0k9 --params-file /home/drim/workspace/ros_ur_driver/install/ur_robot_driver/share/ur_robot_driver/config/ur5e_update_rate.yaml --params-file /home/drim/workspace/ros_ur_driver/install/ur_robot_driver/share/ur_robot_driver/config/ur_controllers.yaml'].
[INFO] [rviz2-5]: process has finished cleanly [pid 8378]

Myabe is the same problem?

TZECHIN6, 2023-06-25:
https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/724

I am having the exact same issue with you bro...

AlbinEV, 2023-07-02:
With new release the problem is solved.

closed 2023-07-02 by AlbinEV