UniversalRobots/Universal_Robots_ROS2_Driver #718: The controller_manager node does not appear to be running.

gr3atchoi, 2023-06-19:
It seems that controller_manager is not running and this is causing timeouts on nodes.

At first I think it was the same problem with #712 or #677 .
However, only the error related with `tf_prefix`  disappeared and an issue occurs where controller_manager is terminated as shown in the following log.

My Environment : 
* UR3e  hardware
* Ubuntu 22.04 
* ROS Humble 
* Semi-binary source build with ros2 driver humble (2.2.7) 

command : ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.56.2 launch_rviz:=false
(followed document) 

* error 
>[ERROR] [spawner-7]: process has died [pid 246663, exit code 1, cmd '/home/gr3atchoi/Workspace/UniversalRobots/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner joint_state_broadcaster --controller-manager /controller_manager --controller-manager-timeout 10 --ros-args'].
[ERROR] [spawner-11]: process has died [pid 246671, exit code 1, cmd '/home/gr3atchoi/Workspace/UniversalRobots/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner forward_position_controller --controller-manager /controller_manager --controller-manager-timeout 10 --inactive --ros-args'].
[ERROR] [spawner-9]: process has died [pid 246667, exit code 1, cmd '/home/gr3atchoi/Workspace/UniversalRobots/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner speed_scaling_state_broadcaster --controller-manager /controller_manager --controller-manager-timeout 10 --ros-args'].
[ERROR] [spawner-10]: process has died [pid 246669, exit code 1, cmd '/home/gr3atchoi/Workspace/UniversalRobots/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner force_torque_sensor_broadcaster --controller-manager /controller_manager --controller-manager-timeout 10 --ros-args'].
[ERROR] [spawner-6]: process has died [pid 246661, exit code 1, cmd '/home/gr3atchoi/Workspace/UniversalRobots/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner scaled_joint_trajectory_controller -c /controller_manager --controller-manager-timeout 10 --ros-args'].
[ERROR] [spawner-8]: process has died [pid 246665, exit code 1, cmd '/home/gr3atchoi/Workspace/UniversalRobots/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner io_and_status_controller --controller-manager /controller_manager --controller-manager-timeout 10 --ros-args'].

*full log  

>[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [246653]
[INFO] [controller_stopper_node-3]: process started with pid [246655]
[INFO] [ur_ros2_control_node-1]: process started with pid [246651]
[INFO] [robot_state_publisher-4]: process started with pid [246657]
[INFO] [rviz2-5]: process started with pid [246659]
[INFO] [spawner-6]: process started with pid [246661]
[INFO] [spawner-7]: process started with pid [246663]
[INFO] [spawner-8]: process started with pid [246665]
[INFO] [spawner-9]: process started with pid [246667]
[INFO] [spawner-10]: process started with pid [246669]
[INFO] [spawner-11]: process started with pid [246671]
[dashboard_client-2] [INFO] [1687161342.070661211] [UR_Client_Library]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[controller_stopper_node-3] [INFO] [1687161342.070785146] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[robot_state_publisher-4] [INFO] [1687161342.074414081] [robot_state_publisher]: got segment base
[robot_state_publisher-4] [INFO] [1687161342.074483232] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1687161342.074489143] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-4] [INFO] [1687161342.074492580] [robot_state_publisher]: got segment flange
[robot_state_publisher-4] [INFO] [1687161342.074495576] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-4] [INFO] [1687161342.074498431] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-4] [INFO] [1687161342.074501206] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-4] [INFO] [1687161342.074503992] [robot_state_publisher]: got segment tool0
[robot_state_publisher-4] [INFO] [1687161342.074506697] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-4] [INFO] [1687161342.074509422] [robot_state_publisher]: got segment world
[robot_state_publisher-4] [INFO] [1687161342.074512017] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-4] [INFO] [1687161342.074514662] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-4] [INFO] [1687161342.074517247] [robot_state_publisher]: got segment wrist_3_link
[ur_ros2_control_node-1] text not specified in the tf_prefix tag
[ur_ros2_control_node-1] [INFO] [1687161342.076738474] [resource_manager]: Loading hardware 'ur3e' 
[ur_ros2_control_node-1] [INFO] [1687161342.078116525] [resource_manager]: Initialize hardware 'ur3e' 
[ur_ros2_control_node-1] [INFO] [1687161342.078157342] [resource_manager]: Successful initialization of hardware 'ur3e'
[ur_ros2_control_node-1] [INFO] [1687161342.078271879] [resource_manager]: 'configure' hardware 'ur3e' 
[ur_ros2_control_node-1] [INFO] [1687161342.078276308] [resource_manager]: Successful 'configure' of hardware 'ur3e'
[ur_ros2_control_node-1] [INFO] [1687161342.078292909] [resource_manager]: 'activate' hardware 'ur3e' 
[ur_ros2_control_node-1] [INFO] [1687161342.078297859] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1687161342.078308278] [URPositionHardwareInterface]: Initializing driver...
[ur_ros2_control_node-1] [WARN] [1687161342.078822743] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[ur_ros2_control_node-1] [INFO] [1687161342.391415312] [UR_Client_Library]: Negotiated RTDE protocol version to 2.
[ur_ros2_control_node-1] [INFO] [1687161342.391753343] [UR_Client_Library]: Setting up RTDE communication with frequency 500.000000
[rviz2-5] [INFO] [1687161342.634220987] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] [INFO] [1687161342.634343159] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-5] [INFO] [1687161342.702046477] [rviz2]: Stereo is NOT SUPPORTED
[ur_ros2_control_node-1] [INFO] [1687161343.405649903] [URPositionHardwareInterface]: Calibration checksum: 'calib_16756443741236045476'.
[ur_ros2_control_node-1] [WARN] [1687161343.406112840] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[spawner-9] [INFO] [1687161344.245963966] [spawner_speed_scaling_state_broadcaster]: Waiting for '/controller_manager' node to exist
[... 14 lines cut ...]
[spawner-8] [INFO] [1687161346.257453308] [spawner_io_and_status_controller]: Waiting for '/controller_manager' node to exist
[spawner-9] [INFO] [1687161348.261054534] [spawner_speed_scaling_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-7] [INFO] [1687161348.261535325] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-11] [INFO] [1687161348.261999945] [spawner_forward_position_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1687161348.262444748] [spawner_scaled_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
[spawner-10] [INFO] [1687161348.262680224] [spawner_force_torque_sensor_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-8] [INFO] [1687161348.265702910] [spawner_io_and_status_controller]: Waiting for '/controller_manager' node to exist
[spawner-9] [INFO] [1687161350.268641840] [spawner_speed_scaling_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-7] [INFO] [1687161350.269368256] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1687161350.269441575] [spawner_scaled_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
[spawner-11] [INFO] [1687161350.270001005] [spawner_forward_position_controller]: Waiting for '/controller_manager' node to exist
[spawner-10] [INFO] [1687161350.270859762] [spawner_force_torque_sensor_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-8] [INFO] [1687161350.274059063] [spawner_io_and_status_controller]: Waiting for '/controller_manager' node to exist
[... cut ...]
[ERROR] [spawner-6]: process has died [pid 246661, exit code 1, cmd '/home/gr3atchoi/Workspace/UniversalRobots/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner scaled_joint_trajectory_controller -c /controller_manager --controller-manager-timeout 10 --ros-args'].
[ERROR] [spawner-8]: process has died [pid 246665, exit code 1, cmd '/home/gr3atchoi/Workspace/UniversalRobots/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner io_and_status_controller --controller-manager /controller_manager --controller-manager-timeout 10 --ros-args'].

LasseBerg, 2023-06-28:
Any updates or workarounds on this?
I'm having the same issue with my ur5

fmauch, 2023-06-28:
This looks like a duplicate of #677 (in #677 being mentioned different errors resulting from the same change).

Please use the most recent version of the driver (being 2.2.8 at the time of writing) together with ros2_control built from sources / from the testing repo / wait until it gets synced tomorrow.

Since there is nothing to additionally add here, I'll close this.

closed 2023-06-28 by fmauch