UniversalRobots/Universal_Robots_ROS2_Driver #723: ur_moveit_config launch error

dcross23, 2023-06-21:
Im trying to launch ur_robot_driver with an UR3e robot. It works, but when i try to run it using moveit, no robot appears on it, neither any group to move or anything related.
Im using: 
- Ubuntu 22.04
- ROS Rolling

Im running the following commands:

- ros2 launch ur_calibration calibration_correction.launch.py robot_ip:=212.128.172.172 target_filename:="${HOME}/my_robot_calibration.yaml"
- ros2 launch ur_robot_driver ur3e.launch.py robot_ip:=212.128.172.172
- ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e 

With this, driver start correctly and robot connects, but when ir run moveit i get this after "You can start planning no" message:
```
[move_group-1] 
[move_group-1] You can start planning now!
[move_group-1] 
[rviz2-2] [INFO] [1687365269.023217120] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1687365269.023411311] [rviz2]: OpenGl version: 3.3 (GLSL 3.3)
[rviz2-2] [INFO] [1687365269.081645321] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[rviz2-2]          at line 321 in /opt/ros/rolling/include/class_loader/class_loader/class_loader_core.hpp
[rviz2-2] [ERROR] [1687365272.357837537] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-2] [INFO] [1687365272.397845311] [moveit_ros_visualization.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
[rviz2-2] [INFO] [1687365272.501214292] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00720844 seconds
[rviz2-2] [INFO] [1687365272.501495068] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[rviz2-2] [INFO] [1687365272.501543002] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rviz2-2] [ERROR] [1687365272.520429834] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'robot_description_planning.joint_limits.shoulder_pan_joint.max_acceleration' has invalid type: Wrong parameter type, parameter {robot_description_planning.joint_limits.shoulder_pan_joint.max_acceleration} is of type {double}, setting it to {string} is not allowed.
```

I also attach a photo of moveit in rviz.
[image]

 Is there any solution?

fmauch, 2023-07-03:
I'm afraid, I cannot reproduce this issue.

beckersven, 2023-08-08:
Hello,

I also experienced this issue. I fixed it by prepending "LC_NUMERIC=en_US.UTF-8" to the launch. So:
`LC_NUMERIC=en_US.UTF-8 ros2 launch ...`

Best of Luck

PS: My solution is from [this](https://github.com/ros-planning/moveit2/issues/1049#issuecomment-1047029751)

fmauch, 2024-04-10:
If setting `LC_NUMERIC` is the solution, I'd forward this issue to https://github.com/ros-planning/moveit2/issues/1882

closed 2024-04-10 by fmauch