UniversalRobots/Universal_Robots_ROS2_Driver #731: Compile error:name'tool_contact' is not defined

Howie-Ye, 2023-07-03:
I've gotten a new problem after fixing the previous one in #677 by using the repo for the semi-binary builds:
`vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.humble.repos`
It built without error or warning and worked fine. The robot model can show in Rviz.

[image]

But when I pressed the start button in PolyScope to boot the robot, an error occurred which has never been seen before.
I cannot even find something related to the `'tool_contact'`.

[image]

Once I press the button,the terminal gives this:
`
[ur_ros2_control_node-1] [INFO] [1688379381.067357722] [UR_Client_Library]: Robot requested program
[ur_ros2_control_node-1] [INFO] [1688379381.067614856] [UR_Client_Library]: Sent program to robot
`

`Humble` distro with package versions
 ` ur_robot_driver: 2.2.8`
 `ros2_control: 2.28.0`

Here is the log
 :~/ws$ ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5 robot_ip:=192.168.137.6 
[INFO] [launch]: All log files can be found below /home/fan/.ros/log/2023-07-03-18-15-54-566515-fan-33660
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [33665]
[INFO] [controller_stopper_node-3]: process started with pid [33667]
[INFO] [ur_ros2_control_node-1]: process started with pid [33663]
[INFO] [robot_state_publisher-4]: process started with pid [33669]
[INFO] [rviz2-5]: process started with pid [33671]
[INFO] [spawner-6]: process started with pid [33673]
[INFO] [spawner-7]: process started with pid [33675]
[INFO] [spawner-8]: process started with pid [33677]
[INFO] [spawner-9]: process started with pid [33679]
[INFO] [spawner-10]: process started with pid [33719]
[INFO] [spawner-11]: process started with pid [33723]
[rviz2-5] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[dashboard_client-2] [INFO] [1688379355.222054674] [UR_Client_Library]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[controller_stopper_node-3] [INFO] [1688379355.220882491] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[ur_ros2_control_node-1] text not specified in the tf_prefix tag
[ur_ros2_control_node-1] [INFO] [1688379355.256222119] [resource_manager]: Loading hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1688379355.261410358] [resource_manager]: Initialize hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1688379355.261616297] [resource_manager]: Successful initialization of hardware 'ur5'
[ur_ros2_control_node-1] [INFO] [1688379355.261999019] [resource_manager]: 'configure' hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1688379355.262017045] [resource_manager]: Successful 'configure' of hardware 'ur5'
[ur_ros2_control_node-1] [INFO] [1688379355.262073121] [resource_manager]: 'activate' hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1688379355.262107416] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1688379355.262133169] [URPositionHardwareInterface]: Initializing driver...
**[ur_ros2_control_node-1] [ERROR] [1688379355.268664283] [UR_Client_Library]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md for details.
[ur_ros2_control_node-1] [ERROR] [1688379355.268714625] [UR_Client_Library]: Scheduling is NOT SCHED_FIFO!**
[robot_state_publisher-4] [INFO] [1688379355.271597081] [robot_state_publisher]: got segment base
[robot_state_publisher-4] [INFO] [1688379355.271874373] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1688379355.271909108] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-4] [INFO] [1688379355.271920180] [robot_state_publisher]: got segment flange
[robot_state_publisher-4] [INFO] [1688379355.271930052] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-4] [INFO] [1688379355.271939882] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-4] [INFO] [1688379355.271949595] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-4] [INFO] [1688379355.271959548] [robot_state_publisher]: got segment tool0
[robot_state_publisher-4] [INFO] [1688379355.271968948] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-4] [INFO] [1688379355.271978412] [robot_state_publisher]: got segment world
[robot_state_publisher-4] [INFO] [1688379355.271987624] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-4] [INFO] [1688379355.271997207] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-4] [INFO] [1688379355.272006522] [robot_state_publisher]: got segment wrist_3_link
[ur_ros2_control_node-1] [INFO] [1688379355.482159905] [UR_Client_Library]: Negotiated RTDE protocol version to 2.
[ur_ros2_control_node-1] [INFO] [1688379355.483151289] [UR_Client_Library]: Setting up RTDE communication with frequency 125.000000
[rviz2-5] [INFO] [1688379355.698843654] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] [INFO] [1688379355.699104831] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-5] [INFO] [1688379355.730489913] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] Warning: Invalid frame ID "forearm_link" passed to canTransform argument source_frame - frame does not exist
[... 148 lines cut ...]
[ur_ros2_control_node-1] [INFO] [1688379357.827330694] [controller_manager]: Configuring controller 'force_torque_sensor_broadcaster'
[ur_ros2_control_node-1] [INFO] [1688379357.828883530] [io_and_status_controller]: Configure UR gpio controller with tf_prefix: 
[ur_ros2_control_node-1] [INFO] [1688379357.829006707] [io_and_status_controller]: Configure UR gpio controller with tf_prefix: 
[ur_ros2_control_node-1] [INFO] [1688379357.831305713] [io_and_status_controller]: Configure UR gpio controller with tf_prefix: 
[ur_ros2_control_node-1] [INFO] [1688379357.845542027] [io_and_status_controller]: Configure UR gpio controller with tf_prefix: 
[spawner-8] [INFO] [1688379357.849354471] [spawner_io_and_status_controller]: Configured and activated io_and_status_controller
[ur_ros2_control_node-1] [INFO] [1688379357.855900515] [io_and_status_controller]: Configure UR gpio controller with tf_prefix: 
[spawner-9] [INFO] [1688379357.864894351] [spawner_speed_scaling_state_broadcaster]: Configured and activated speed_scaling_state_broadcaster
[ur_ros2_control_node-1] [INFO] [1688379357.871574785] [io_and_status_controller]: Configure UR gpio controller with tf_prefix: 
[spawner-10] [INFO] [1688379357.881750373] [spawner_force_torque_sensor_broadcaster]: Configured and activated force_torque_sensor_broadcaster
[INFO] [spawner-6]: process has finished cleanly [pid 33673]
[INFO] [spawner-11]: process has finished cleanly [pid 33723]
[INFO] [spawner-8]: process has finished cleanly [pid 33677]
[INFO] [spawner-9]: process has finished cleanly [pid 33679]
[INFO] [spawner-10]: process has finished cleanly [pid 33719]
**[ur_ros2_control_node-1] [INFO] [1688379381.067357722] [UR_Client_Library]: Robot requested program
[ur_ros2_control_node-1] [INFO] [1688379381.067614856] [UR_Client_Library]: Sent program to robot
[ur_ros2_control_node-1] [INFO] [1688379705.721470628] [UR_Client_Library]: Robot requested program
[ur_ros2_control_node-1] [INFO] [1688379705.721676700] [UR_Client_Library]: Sent program to robot**

BTW here is also confusing
`[ur_ros2_control_node-1] [ERROR] [1688379355.268664283] [UR_Client_Library]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md for details.`
`[ur_ros2_control_node-1] [ERROR] [1688379355.268714625] [UR_Client_Library]: Scheduling is NOT SCHED_FIFO!`

Does anyone has meet this problem? Thanks for any help.

fmauch, 2023-07-03:
@Howie-Ye which verison of Polyscope do you use?

Howie-Ye, 2023-07-04:
> @Howie-Ye which verison of Polyscope do you use?

URSoftware 3.9.1.64192
URSafetyA (3.5.2)
URSafetyB (3.5.4)
URCap: External Control    Version:1.0.4

fmauch, 2023-07-04:
Could you try updating your robot software? I think 3.12 was required since a while and with the latest changes we will even require 3.14.

Howie-Ye, 2023-07-07:
> Could you try updating your robot software? I think 3.12 was required since a while and with the latest changes we will even require 3.14.

Yes,I updated the robot to the 3.12 version, then it worked. 
I also tried on ros-neotic, and updating the system to at least 3.12 is recommended in error message.  
Thanks for your help !!!

closed 2023-07-07 by Howie-Ye